rockchip-linux/mpp #130: 关于rk1808使用mpp解码h.264问题

wujialiang, 2020-04-17:
参考mpi_dec_test的代码进行改造，在第一次灌入h.264的数据结束后，调用mpp_packet_set_eos(packet) 后得到mpp_frame_get_info_change的事件。

然后再灌入新的h.264后，调用mpi->decode_get_frame(ctx, &frame)一直获取不到frame，函数返回的是MPP_OK，但是frame一直为NULL，然后导致一直处于死循环当中，请问是什么问题呢？

当中的操作是否有误？要注意什么事项？我这边是用ffmpeg实时获取rtsp的数据流，调用av_read_frame获取一帧的h.264数据后直接灌入到mpp中。

WainDing, 2020-04-17:
mpp_packet_set_eos(packet)  设置了这个，说明送码流结束了，后面的码流，解码器不再接收。该接口一般用于关闭播放器，告诉解码器释放最后一帧使用的。

wujialiang, 2020-04-17:
@WainDing 
> mpp_packet_set_eos(packet) 设置了这个，说明送码流结束了，后面的码流，解码器不再接收。该接口一般用于关闭播放器，告诉解码器释放最后一帧使用的。

请问如果做摄像头实时获取数据流的情况，应该的处理流程逻辑是怎么样的呢？
另外有qq或者微信或者其他联系方式可以直接跟您这边咨询吗？

WainDing, 2020-04-17:
参考test/mpi_dec_test.c
另外，如果需要支持，请走公司的redmine，谢谢！

wujialiang, 2020-04-17:
> redmine

@WainDing 

我这边也是参考mpi_dec_test.c的，我把mpp_packet_set_eos注释不调用的情况下是可以获取到frame，但是获取的frame一直都是前几帧的图来回循环一直播放

wujialiang, 2020-04-17:
> 参考test/mpi_dec_test.c
> 另外，如果需要支持，请走公司的redmine，谢谢！
@WainDing 

请问流程是怎么走？没有走过，谢谢

wujialiang, 2020-04-21:
@WainDing 可以指引一下？？谢谢

WainDing, 2020-04-21:
可以找供应商或者售后支持

wujialiang, 2020-04-21:
> 可以找供应商或者售后支持

@WainDing 

已经咨询过了，供应商售后不支持软件方面的问题，只支持硬件或者驱动问题。

MUZLATAN, 2020-04-26:
能否提供test_ff_rtsp_dec 这个demo的源码， 只看MPP开发手册， 有些难以理解

HermanChen, 2020-04-26:
http://dev.t-firefly.com/thread-100764-1-1.html
网上看是firefly做的，
apt update
apt install firefly-multi-rtsp

MUZLATAN, 2020-04-26:
@HermanChen 我看到过这个，我没有搞这个固件升级， 所以没有下载这个东西， 但是猜测这是一个二进制可执行文件， 并不是源码

MUZLATAN, 2020-04-27:
我是用ffmpeg遇到了相同的问题， 
-------------------------------
调用mpi->decode_get_frame(ctx, &frame)一直获取不到frame，函数返回的是MPP_OK，但是frame一直为NULL，然后导致一直处于死循环当中
-------------------------------
将   if (pkt_eos)
    {
        mpp_packet_set_eos(packet);
    }
注释掉，还是在死循环， 能否提供一下思路

HermanChen, 2020-04-27:
在 mpp 的 init 之前，通过 contorl 接口 配置一个 split_mode 为 1，然后通过 control 配置输出阻塞模式 为 1，然后再试下，具体做法参考下 mpi_dec_test

MUZLATAN, 2020-04-28:
感谢， @HermanChen  代码跑通了， 但是说实话， 关于mpp官方资料实在是太少了，对于初学者不太友好， 我将我的代码发到github上仅做参考  https://github.com/MUZLATAN/ffmpeg_rtsp_mpp

HermanChen, 2020-04-28:
好的，也感谢使用 mpp 代码，我们开发人员很少，文档来不及写……只有《mpp 开发参考》这个文档。
有空的话，可以帮我们一起完善下：）
也可以加入我们来一起搞，哈哈：）

MUZLATAN, 2020-04-28:
明白， 这个解码效率挺高的， cpu占用少， 说明你们很厉害！

HermanChen, 2020-04-28:
谢谢：）

closed 2020-04-28 by HermanChen

wujialiang, 2020-04-28:
我也跑通，但是我的做发跟demo有区别，目前同时跑两路的720p解码

wujialiang, 2020-04-28:
> 感谢， @HermanChen 代码跑通了， 但是说实话， 关于mpp官方资料实在是太少了，对于初学者不太友好， 我将我的代码发到github上仅做参考 https://github.com/MUZLATAN/ffmpeg_rtsp_mpp

@HermanChen 请问你的做法是怎么样的呢？

HermanChen, 2020-04-28:
@MUZLATAN 我能把你的地址加到 mpp 的 readme 说明里，让大家来参考么？

HermanChen, 2020-04-29:
> 
> 
> > 感谢， @HermanChen 代码跑通了， 但是说实话， 关于mpp官方资料实在是太少了，对于初学者不太友好， 我将我的代码发到github上仅做参考 https://github.com/MUZLATAN/ffmpeg_rtsp_mpp
> 
> @HermanChen 请问你的做法是怎么样的呢？

啊，我们是开发人员，主要做底层驱动，看你们用户是怎么用的，我们会做针对性改进
我们内部有用 rtp/rtsp 这些的对接，是放到产品 sdk 里的

wujialiang, 2020-04-29:
> > > 感谢， @HermanChen 代码跑通了， 但是说实话， 关于mpp官方资料实在是太少了，对于初学者不太友好， 我将我的代码发到github上仅做参考 https://github.com/MUZLATAN/ffmpeg_rtsp_mpp
> > 
> > 
> > @HermanChen 请问你的做法是怎么样的呢？
> 
> 啊，我们是开发人员，主要做底层驱动，看你们用户是怎么用的，我们会做针对性改进
> 我们内部有用 rtp/rtsp 这些的对接，是放到产品 sdk 里的

@HermanChen 

mpi->decode_get_frame(ctx, &frame)一直获取不到frame，函数返回的是MPP_OK，但是frame一直为NULL

像这种情况，返回值为MPP_OK，但是frame为null是否可以认为是一个bug呢？能否把返回值改为frame未null的真正原因吗？

HermanChen, 2020-04-29:
不是bug，mpi 的设计原则是函数返回用于判断流程是否正常，用于错误路径的处理，是否有得到结果，要通过函数的输出参数来判断。
不过，如果大家都觉得要用返回值判断是否得到图像是也可以的

wujialiang, 2020-05-05:
> 不是bug，mpi 的设计原则是函数返回用于判断流程是否正常，用于错误路径的处理，是否有得到结果，要通过函数的输出参数来判断。
> 不过，如果大家都觉得要用返回值判断是否得到图像是也可以的

希望文档方面可以更新一版比较详细的说明。

shanchao0906, 2024-05-29:
h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480 参考以上的教程长时间就会有这个问题