rockchip-linux/mpp #132: rk1808，使用mpp调用decode_put_packet灌入h.264码流是否有限制？

wujialiang, 2020-04-22:
目前使用mpp，从ffmpeg不断拉去h.264码流，然后一直调用decode_put_packet灌入，但是会出现！MPP_OK，请问是否有灌入的速度限制或者容量限制？ 因为灌入的处理不及时，会导致ffmpeg出现掉帧的情况，下面是日志信息？请问这操作流程有什么问题？

[rtsp @ 0x7f7401e390] max delay reached. need to consume packet
[rtsp @ 0x7f7401e390] RTP: missed 25 packets
mpp_log: decoder_get_frame get err info:1 discard:0.
[ 5470.753204] rk-vcodec ffb80000.vpu_service: resetting...
[ 5470.753316] rk-vcodec ffb80000.vpu_service: reset done
[ 5470.753453] rk-vcodec ffb80000.vpu_service: reset done
mpp_log: decoder_get_frame get err info:1 discard:0.
mpp_log: decoder_get_frame get err info:1 discard:0.
mpp_log: decoder_get_frame get err info:1 discard:0.
mpp_log: decoder_get_frame get err info:1 discard:0.
mpp_log: decoder_get_frame get err info:1 discard:0.
mpp_log: decoder_get_frame get err info:1 discard:0.
mpp_log: decoder_get_frame get err info:1 discard:0.
mpp_log: decoder_get_frame get err info:1 discard:0.
mpp_log: decoder_get_frame get err info:1 discard:0.
mpp_log: decoded        297 frame   24.87 fps
mpp_log: decoded        345 frame   46.86 fps

wujialiang, 2020-04-24:
decoder_get_frame get err info:1  代表什么意思？

@WainDing

MUZLATAN, 2020-04-26:
我之前也出现这个问题， 我换了一个最新版本，然后加了编译选项就没了这个问题   err info : 1, 你可以试一下。

另外， 能否请教， 在拿到rtsp流的buf  后怎么传给mpp解码， 就你现在的代码错误信息来看，应该是解码的地方出现错误了， 方便的话 把你的代码分享给我吗

wujialiang, 2020-04-28:
@MUZLATAN 我也是直接用demo的来改的，只是不知道如何判断h.264码流是否有错误。我直接写在qt上，有联系方式吗？

MUZLATAN, 2020-05-07:
你可以将264文件直接保存下来， 网上有对应的例子， 用vlc播放器打开， 能打开看就说明没问题， 然后再用mpi_dec_test测试， 保存为yuv文件，  可以从网上下载yuvplayer.exe的文件查看，
确保每一步都走通就可以了

MUZLATAN, 2020-05-08:
我的qq:769795480 @wujialiang

wujialiang, 2020-05-15:
> 我的qq:769795480 @wujialiang

@MUZLATAN 加你啦

zhaochengwei, 2020-05-25:
kernel:
```
root@douding-pack:~/mpp/test# lsb_release  -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.6 LTS
Release:        16.04
Codename:       xenial
root@douding-pack:~/mpp/test# uname -a
Linux douding-pack 4.4.52 #14 SMP PREEMPT Fri Jun 14 17:18:50 CST 2019 aarch64 aarch64 aarch64 GNU/Linux
```

xiaqing10, 2020-12-02:
我出现的问题是 decode_get_frame 返回MPP_OK,　但是frame 是null.
貌似是解码的问题，但是喂的h264数据我又能正确的打开。

HermanChen, 2020-12-02:
decode_get_frame 返回 OK 指是流程正常，需要检查输出的 frame
给 mpi_dec_test 的输入需要是祼码流，不带封装的

xiaqing10, 2020-12-03:
> decode_get_frame 返回 OK 指是流程正常，需要检查输出的 frame
> 给 mpi_dec_test 的输入需要是祼码流，不带封装的

是的，我的用法是利用ffmpeg解码rtsp来获取h264流，然后利用mpp来解h264得到yuv，不过decode_get_frame得到frame是null.
一直还没定位到问题。

wxlzyz, 2022-09-08:
> > decode_get_frame 返回 OK 指是流程正常，需要检查输出的 frame
> > 给 mpi_dec_test 的输入需要是祼码流，不带封装的
> 
> 是的，我的用法是利用ffmpeg解码rtsp来获取h264流，然后利用mpp来解h264得到yuv，不过decode_get_frame得到frame是null. 一直还没定位到问题。

你好，请问后边解决这个问题了吗？能否分享一下解决方法，我也遇到这个问题了

sxj731533730, 2023-02-10:
我也遇到了，很奇怪，我遇到的是在cmakelists.txt增加opencv选项　就会出现这个问题　奇怪
https://github.com/yijiesun/mpp_drm_rga_demo.git

/home/firefly/h264_mpp_yuv/cmake-build-debug/h264_mpp_yuv
decode_get_frame get info changed found.
decoder require buffer w:h [1920:1080] stride [1920:1088]decoder_get_frame get err info:1 discard:0.
decode_get_frame ID:0 get srcFrm 0.
decoder_get_frame get err info:1 discard:0.
decode_get_frame ID:0 get srcFrm 1.
decoder_get_frame get err info:1 discard:0.
decode_get_frame ID:0 get srcFrm 2.
decoder_get_frame get err info:1 discard:0.
decode_get_frame ID:0 get srcFrm 3.
decoder_get_frame get err info:1 discard:0.
decode_get_frame ID:0 get srcFrm 4.
decoder_get_frame get err info:1 discard:0.
decode_get_frame ID:0 get srcFrm 5.
decoder_get_frame get err info:1 discard:0.
decode_get_frame ID:0 get srcFrm 6.
decoder_get_frame get err info:1 discard:0.
decode_get_frame ID:0 get srcFrm 7.
decoder_get_frame get err info:1 discard:0.
decode_get_frame ID:0 get srcFrm 8.
decoder_get_frame get err info:1 discard:0.
decode_get_frame ID:0 get srcFrm 9.
decoder_get_frame get err info:1 discard:0.
decode_get_frame ID:0 get srcFrm 10.
decoder_get_frame get err info:1 discard:0.
decode_get_frame ID:0 get srcFrm 11.
decoder_get_frame get err info:1 discard:0.
decode_get_frame ID:0 get srcFrm 12.
decoder_get_frame get err info:1 discard:0.
decode_get_frame ID:0 get srcFrm 13.

sxj731533730, 2023-02-10:
我解决一部分问题，更新了官方最新的库文件和使用超级权限和执行代码　可以显示画面和解码

xiaogao22, 2024-01-19:
> > decode_get_frame 返回 OK 指是流程正常，需要检查输出的 frame
> > 给 mpi_dec_test 的输入需要是祼码流，不带封装的
> 
> 是的，我的用法是利用ffmpeg解码rtsp来获取h264流，然后利用mpp来解h264得到yuv，不过decode_get_frame得到frame是null. 一直还没定位到问题。

我这边用ffmpeg保存了下我的摄像头的h264裸流，用ffmpeg和VLC都不能播，ffmpeg提示non-existing PPS 0 referenced，可能是mpp分帧模式没有pps sps没法正常解码？

HermanChen, 2024-01-19:
264 里 slice 头里的 pps_id 要有对应的信息才能正常解码啊，协议如此，没有 pps 让解码器怎么解……
如果码流是从非 I 帧开始的，会有一段码流是无法解码的，到后面带着 sps/pps 的 I 帧来的时候，就能开始正常解码