rockchip-linux/mpp #135: Why  decoder_get_frame get err info:1 discard:0.

zhaochengwei, 2020-05-25:
## branch：
release 

## test.264:
https://whale-firmware.oss-cn-shanghai.aliyuncs.com/tmp/test.264

## board:
firefly-rk3399
## dmesg
```
[956695.405992] rk-vcodec ff660000.rkvdec: resetting...
[956695.406001] rk-vcodec ff660000.rkvdec: for 3288/3368...
[956695.406005] rk-vcodec ff660000.rkvdec: vpu reset in
[956695.406044] rk-vcodec ff660000.rkvdec: reset done
[956695.409858] rk-vcodec ff660000.rkvdec: can not find 16 buffer in list
[956696.490088] rk-vcodec ff660000.rkvdec: closed
```
## result
```
root@douding-pack:~/mpp/test# ./mpi_dec_test -i ./test.264 -t 7 -o ./test.yuv
mpi_dec_test: cmd parse result:
mpi_dec_test: input  file name: ./test.264
mpi_dec_test: output file name: ./test.yuv
mpi_dec_test: config file name: 
mpi_dec_test: width      :    0
mpi_dec_test: height     :    0
mpi_dec_test: type       : 7
mpi_dec_test: debug flag : 0
mpi_dec_test: max frames : 0
mpi_dec_test: mpi_dec_test start
mpi_dec_test: input file size 1068248
mpi_dec_test: mpi_dec_test decoder test start w 0 h 0 type 7
mpi: mpp version: 06a6014 author: Johnson Ding [build]: Update Android build script
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpp_log: can not found match soc name: rockchip,android 9tripod,x3399-development-board rockchip,rk339
mpi_dec_test: decode_get_frame get info changed found
mpi_dec_test: decoder require buffer w:h [704:576] stride [704:576] buf_size 811008
mpi_dec_test: decode_get_frame get frame 1
mpi_dec_test: decode_get_frame get frame 2
mpi_dec_test: decode_get_frame get frame 3
mpi_dec_test: decode_get_frame get frame 4
mpi_dec_test: decode_get_frame get frame 5
mpi_dec_test: decode_get_frame get frame 6
mpi_dec_test: decode_get_frame get frame 7
mpi_dec_test: decode_get_frame get frame 8
mpi_dec_test: decode_get_frame get frame 9
mpi_dec_test: decode_get_frame get frame 10
mpi_dec_test: decode_get_frame get frame 11
mpi_dec_test: decode_get_frame get frame 12
mpi_dec_test: decode_get_frame get frame 13
mpi_dec_test: decode_get_frame get frame 14
mpi_dec_test: decode_get_frame get frame 15
mpi_dec_test: decode_get_frame get frame 16
mpi_dec_test: decode_get_frame get frame 17
mpi_dec_test: decode_get_frame get frame 18
mpi_dec_test: decode_get_frame get frame 19
mpi_dec_test: decode_get_frame get frame 20
mpi_dec_test: decode_get_frame get frame 21
mpi_dec_test: decode_get_frame get frame 22
mpi_dec_test: decode_get_frame get frame 23
mpi_dec_test: decode_get_frame get frame 24
mpi_dec_test: decode_get_frame get frame 25
mpi_dec_test: decode_get_frame get frame 26
mpi_dec_test: decode_get_frame get frame 27
mpi_dec_test: decode_get_frame get frame 28
mpi_dec_test: decode_get_frame get frame 29
mpi_dec_test: decode_get_frame get frame 30
mpi_dec_test: decode_get_frame get frame 31
mpi_dec_test: decode_get_frame get frame 32
[... 155 lines cut ...]
mpi_dec_test: decode_get_frame get frame 188
mpi_dec_test: decode_get_frame get frame 189
mpi_dec_test: decode_get_frame get frame 190
mpi_dec_test: decode_get_frame get frame 191
mpi_dec_test: decode_get_frame get frame 192
mpi_dec_test: decode_get_frame get frame 193
mpi_dec_test: decode_get_frame get frame 194
mpi_dec_test: decode_get_frame get frame 195
mpi_dec_test: decode_get_frame get frame 196
mpi_dec_test: decode_get_frame get frame 197
mpi_dec_test: decode_get_frame get frame 198
mpi_dec_test: decode_get_frame get frame 199
mpi_dec_test: decode_get_frame get frame 200
**mpi_dec_test: get error and check frame_num
mpi_dec_test: found last packet**
mpi_dec_test: decode_get_frame get frame 201
mpi_dec_test: decode_get_frame get frame 202
mpi_dec_test: decode_get_frame get frame 203
mpi_dec_test: decode_get_frame get frame 204
mpi_dec_test: decode_get_frame get frame 205
**mpi_dec_test: decoder_get_frame get err info:1 discard:0.**
mpi_dec_test: decode_get_frame get frame 206
mpi_dec_test: found last frame
mpi_dec_test: test success max memory 4.64 MB
```

zhaochengwei, 2020-05-25:
root@douding-pack:~/mpp/test# lsb_release  -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.6 LTS
Release:        16.04
Codename:       xenial
root@douding-pack:~/mpp/test# uname -a
Linux douding-pack 4.4.52 #14 SMP PREEMPT Fri Jun 14 17:18:50 CST 2019 aarch64 aarch64 aarch64 GNU/Linux

programcj, 2021-05-11:
mpp_buffer_group_unused(data.frm_grp)=23
mpi_dec_test: decode_get_frame get frame 47
MPP_FMT_YUV420SP
				 use: 22 ms
mpp_buffer_group_unused(data.frm_grp)=23
mpi_dec_test: decode_get_frame get frame 48
MPP_FMT_YUV420SP
				 use: 36 ms
mpp_buffer_group_unused(data.frm_grp)=23
mpi_dec_test: decode_get_frame get frame 49
MPP_FMT_YUV420SP
				 use: 39 ms
mpp_buffer_group_unused(data.frm_grp)=23
mpi_dec_test: decode_get_frame get frame 50
MPP_FMT_YUV420SP
				 use: 23 ms
mpp_buffer_group_unused(data.frm_grp)=23
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 51
mpp_buffer_group_unused(data.frm_grp)=20
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 52
mpp_buffer_group_unused(data.frm_grp)=21
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 53
mpp_buffer_group_unused(data.frm_grp)=21
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 54
mpp_buffer_group_unused(data.frm_grp)=22
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 55
mpp_buffer_group_unused(data.frm_grp)=23
mpi_dec_test: decoder_get_frame get err info:1 discard:0.
mpi_dec_test: decode_get_frame get frame 56

我这也出现很多的这种消息，硬件解码失败，丢失帧

我在其他网站上找到了一个说法：

硬解码是不会给你合并帧的，丢进去硬件一帧要么解的出来，要么就解不出来。
如果送mpp的一帧不完整，你要开SPLITE_MODE，这样mpp软件部分会去合并，但是建议自己合并完完整帧再送mpp。

但是不知道怎么开启splite_mode，

HermanChen, 2021-05-11:
> ```
> root@douding-pack:~/mpp/test# lsb_release  -a
> No LSB modules are available.
> Distributor ID: Ubuntu
> Description:    Ubuntu 16.04.6 LTS
> Release:        16.04
> Codename:       xenial [...]

这个最后的解码报错估计是最后一帧的最后一包不完整导致出错的，没什么问题的。

HermanChen, 2021-05-11:
> mpp_buffer_group_unused(data.frm_grp)=23
> mpi_dec_test: decode_get_frame get frame 47
> MPP_FMT_YUV420SP
> use: 22 ms
> mpp_buffer_group_unused(data.frm_grp)=23
> mpi_dec_test: decode_get_frame get frame 48 [...]

看 mpi_dec_test 里的 need_split 标志

programcj, 2021-05-14:
@HermanChen
已经解决了 mpp可以屏蔽 解码错误信息的功能，不注意看文档确实很难发现
最近github国内太难访问了，什么时候也弄一个比较好的国内的镜像呢？

HermanChen, 2021-05-15:
https://gitee.com/hermanchen82/mpp/
这里有镜像