rockchip-linux/mpp #140: rk3399 mpp h.264 decoder with 4096 pixel width

prefergui, 2020-06-16:
I use RockPro64 (rk3399) with almost minimal Linux 4.4
MPP h.264 decoder (with DRM) works well, except with 4096 pixel width video data.
I tested some 24fps video data. 
With 4080 or less pixel width, there is no problem.
With 4096 pixel width, performance degradation occur.
The decoder output is not in time and not smooth.
Is this a problem on my system only?
Regards.

HermanChen, 2020-06-16:
I am not sure about the root cause without any knowledge about the boad information.

It mainly about cache issue when the image width come to 4096 which is just the page size.
Please check the hardware working time by the cmd:
echo 0x100 > /sys/module/rk_vcodec/parameters/debug
And compare and check the hardware timing for each frame.

I guess the linux drm display required buffer is just equal to size of decoding image. But normally we do optimization on Android to avoid 4K aligned buffer width. Then the performance drop will not happen.

prefergui, 2020-06-16:
Thank you.

This is 4096x2160 24fps (encoded on macOS 10.5.5 AV Foundation)
[  514.136142] rk_vcodec: rkvdec task: 73 ms
[  514.158706] rk_vcodec: rkvdec task: 18 ms
[  514.256342] rk_vcodec: rkvdec task: 92 ms
[  514.279756] rk_vcodec: rkvdec task: 18 ms
[  514.352231] rk_vcodec: rkvdec task: 67 ms
[  514.374831] rk_vcodec: rkvdec task: 18 ms
[  514.458699] rk_vcodec: rkvdec task: 80 ms
[  514.481597] rk_vcodec: rkvdec task: 19 ms
[  514.545251] rk_vcodec: rkvdec task: 60 ms
[  514.569023] rk_vcodec: rkvdec task: 18 ms
[  514.644558] rk_vcodec: rkvdec task: 70 ms
[  514.667091] rk_vcodec: rkvdec task: 19 ms
[  514.758735] rk_vcodec: rkvdec task: 86 ms
[  514.781133] rk_vcodec: rkvdec task: 17 ms
[  514.876184] rk_vcodec: rkvdec task: 90 ms
[  514.899704] rk_vcodec: rkvdec task: 18 ms
[  514.970561] rk_vcodec: rkvdec task: 66 ms
[  514.992373] rk_vcodec: rkvdec task: 18 ms
[  515.078038] rk_vcodec: rkvdec task: 80 ms
[  515.102382] rk_vcodec: rkvdec task: 19 ms
[  515.171904] rk_vcodec: rkvdec task: 65 ms
[  515.194868] rk_vcodec: rkvdec task: 15 ms
[  515.229754] rk_vcodec: rkvdec task: 31 ms
[  515.259172] rk_vcodec: rkvdec task: 17 ms
[  515.343055] rk_vcodec: rkvdec task: 79 ms
[  515.360651] rk_vcodec: rkvdec task: 12 ms
[  515.457182] rk_vcodec: rkvdec task: 90 ms
[  515.473141] rk_vcodec: rkvdec task: 12 ms
[  515.548079] rk_vcodec: rkvdec task: 70 ms
[  515.571774] rk_vcodec: rkvdec task: 18 ms
[  515.661845] rk_vcodec: rkvdec task: 84 ms
[  515.684836] rk_vcodec: rkvdec task: 19 ms
[  515.754541] rk_vcodec: rkvdec task: 64 ms
[  515.775992] rk_vcodec: rkvdec task: 18 ms
[  515.856150] rk_vcodec: rkvdec task: 76 ms
[  515.879690] rk_vcodec: rkvdec task: 18 ms
[  515.962796] rk_vcodec: rkvdec task: 78 ms
[  515.983758] rk_vcodec: rkvdec task: 17 ms
[  516.074986] rk_vcodec: rkvdec task: 88 ms
[  516.096573] rk_vcodec: rkvdec task: 18 ms
[  516.164203] rk_vcodec: rkvdec task: 65 ms
[  516.187236] rk_vcodec: rkvdec task: 18 ms
[  516.269294] rk_vcodec: rkvdec task: 78 ms
[  516.293685] rk_vcodec: rkvdec task: 19 ms
[  516.357365] rk_vcodec: rkvdec task: 58 ms
[  516.380178] rk_vcodec: rkvdec task: 18 ms
[  516.462667] rk_vcodec: rkvdec task: 77 ms
[  516.487549] rk_vcodec: rkvdec task: 20 ms
[  516.514746] rk_vcodec: rkvdec task: 22 ms
[  516.532183] rk_vcodec: rkvdec task: 12 ms
[  516.620153] rk_vcodec: rkvdec task: 83 ms
[  516.638036] rk_vcodec: rkvdec task: 13 ms
[  516.700539] rk_vcodec: rkvdec task: 57 ms
[  516.718749] rk_vcodec: rkvdec task: 13 ms
[  516.802123] rk_vcodec: rkvdec task: 78 ms
[  516.821009] rk_vcodec: rkvdec task: 14 ms
[  516.884369] rk_vcodec: rkvdec task: 58 ms
[  516.904200] rk_vcodec: rkvdec task: 15 ms
[  516.973770] rk_vcodec: rkvdec task: 64 ms
[  516.992289] rk_vcodec: rkvdec task: 13 ms
[  517.080126] rk_vcodec: rkvdec task: 84 ms
[  517.101886] rk_vcodec: rkvdec task: 18 ms
[  517.188943] rk_vcodec: rkvdec task: 84 ms
[  517.212815] rk_vcodec: rkvdec task: 19 ms
[  517.278667] rk_vcodec: rkvdec task: 61 ms
[  517.300389] rk_vcodec: rkvdec task: 18 ms
[  517.385291] rk_vcodec: rkvdec task: 80 ms
[... 45 lines cut ...]
[  529.342947] rk_vcodec: rkvdec task: 23 ms
[  529.382288] rk_vcodec: rkvdec task: 18 ms
[  529.409404] rk_vcodec: rkvdec task: 21 ms
[  529.437985] rk_vcodec: rkvdec task: 18 ms
[  529.463173] rk_vcodec: rkvdec task: 22 ms
[  529.494679] rk_vcodec: rkvdec task: 16 ms
[  529.525326] rk_vcodec: rkvdec task: 28 ms
[  529.556983] rk_vcodec: rkvdec task: 16 ms
[  529.582805] rk_vcodec: rkvdec task: 22 ms
[  529.610414] rk_vcodec: rkvdec task: 12 ms
[  529.635382] rk_vcodec: rkvdec task: 20 ms
[  529.667798] rk_vcodec: rkvdec task: 12 ms
[  529.690678] rk_vcodec: rkvdec task: 19 ms
[  529.724062] rk_vcodec: rkvdec task: 19 ms
[  529.749011] rk_vcodec: rkvdec task: 20 ms
[  529.781712] rk_vcodec: rkvdec task: 19 ms
[  529.805307] rk_vcodec: rkvdec task: 19 ms
[  529.840028] rk_vcodec: rkvdec task: 19 ms
[  529.863884] rk_vcodec: rkvdec task: 20 ms
[  529.896930] rk_vcodec: rkvdec task: 19 ms
[  529.952322] rk_vcodec: rkvdec task: 20 ms
[  529.989529] rk_vcodec: rkvdec task: 12 ms
[  530.020336] rk_vcodec: rkvdec task: 22 ms

Regards.

HermanChen, 2020-06-16:
The time below 33ms is just ok.
The time over 33ms is not ok.
So it is the stride issue in mpp H.264 decoder buffer stride alignment issue.
You need to check the mpp library version with the RockPro64 vendor.

prefergui, 2020-06-16:
I downloaded from this site mpp-release on June 11

HermanChen, 2020-06-16:
Try update again?

prefergui, 2020-06-17:
I have just tried, but no improvement.
However, I found a solution.
As your hint, in rkv_h264d_init(), changing rkv_hor_align to rkv_hor_align_256_odds for 
SLOTS_HOR_ALIGN solved the problem.
Thank you.
I also found that extra 16 (rather than 256) is also works.
Then I found MPP_DEC_SET_FRAME_INFO in rkv_h264d_control() does this.
However, it seems to me that it is called from VpuApiLegacy::init() only.
Any way, thanks for nice software and hardware.

HermanChen, 2020-06-17:
Great thanks for feedback and thanks for using our product : )

HermanChen, 2020-06-17:
On rk3399 there are two hardware for decoding H.264:
vpu: low power 1080p@30fps decoder which must has fix 16 aligned stride
rkv (RKVDEC) : high performance 4K@60fps decoder which has flexible stride.
If rkv decoded small picture like 320x240 then the 256_odds aligned is not necessary.

closed 2020-06-22 by HermanChen