rockchip-linux/mpp #142: USB

clilye, 2020-06-22:


clilye, 2020-06-22:
dmesg 可以看到大量的这样的信息输出：
[990545.035287] rk-vcodec ff360000.rkvdec: closed
[990628.350941] rk-vcodec ff360000.rkvdec: closed
[990711.659365] rk-vcodec ff360000.rkvdec: closed
[990794.983128] rk-vcodec ff360000.rkvdec: closed
[990878.426486] rk-vcodec ff360000.rkvdec: closed
[990961.716942] rk-vcodec ff360000.rkvdec: closed
[991045.015930] rk-vcodec ff360000.rkvdec: closed
[991128.207350] rk-vcodec ff360000.rkvdec: closed
[991211.475624] rk-vcodec ff360000.rkvdec: closed
[991294.781434] rk-vcodec ff360000.rkvdec: closed
[991378.089390] rk-vcodec ff360000.rkvdec: closed
[991461.453547] rk-vcodec ff360000.rkvdec: closed
[991544.678290] rk-vcodec ff360000.rkvdec: closed
[991627.993417] rk-vcodec ff360000.rkvdec: closed

HermanChen, 2020-06-22:
估计是jpeg解码代码没写好，处理一帧就开关一次解码器

clilye, 2020-06-22:
@HermanChen 您说的jpeg解码代码是mpp模块的吗？请问有什么规避的办法吗？我们上层代码是调用usb摄像头来录制视频的。谢谢

HermanChen, 2020-06-22:
那个这需要找上层 sdk 的开发人员了，单独在 MPP 模块解决不了，最好去 redmine 提问

closed 2020-06-22 by HermanChen