rockchip-linux/mpp #143: RK3399,多线程解码，同时多线程编码

programcj, 2020-06-22:
使用4个线程读取RTSP视频并解码没什么问题；
但是，当加入另外4个线程对视频的每一帧都编码成JPG，发现解码速度慢了很多，使用netstat -ant也看到网络中还有很多数据未接收；
不知道这是什么问题导致的？

HermanChen, 2020-06-22:
jpeg 编码使用 VPU 硬件，是与小分辨率 264 码流解码的硬件是复用的
setprop use_mpp_mode 1 之后再试一下

programcj, 2020-06-22:
我是在Ubuntu下测试的`export use_mpp_mode=1`
但是结果依旧

programcj, 2020-06-23:
是因为 mpp_buffer_group_limit_config(data->frm_grp, buf_size, 24) 导致的吗?
因为我将 MPP_Frame缓存起来,造成解码线程不能分配倒新的MPP_Buffer?
@HermanChen

HermanChen, 2020-06-23:
echo 0x100 > /sys/modules/rk_vcodec/parameters/debug
看一下 log

programcj, 2020-06-23:
我感觉已经可以了,应该是buffer问题

RKMPPDecodeContext *rk_context = pCodecCtx->priv_data;
RKMPPDecoder *decoder = (RKMPPDecoder *)rk_context->decoder_ref->data;
ret = mpp_buffer_group_limit_config(decoder->frame_group, 0, 100);

wxlzyz, 2022-11-05:
> 使用4个线程读取RTSP视频并解码没什么问题； 但是，当加入另外4个线程对视频的每一帧都编码成JPG，发现解码速度慢了很多，使用netstat -ant也看到网络中还有很多数据未接收； 不知道这是什么问题导致的？

请问多线程是怎么处理的？我这边一使用多线程解码多路视频mpp_frame_get_errinfo就返回1

HermanChen, 2022-11-06:
是这样的，3399 的 264 编码器和 jpeg 编解码是同一个器件时分复用的，所以跑 jpeg 的时候会让 264 编码性能大大下降

HermanChen, 2025-07-07:
编码是不是有去拷贝图像？还有码流 buffer 用的是 uncache 的？

lxzlife, 2025-07-07:
@HermanChen 我这种操作有什么问题吗？

lxzlife, 2025-07-07:
@HermanChen 
系统里的mpp frame看起来只有通过mpp buffer的方式绑定，如下：

[image]

如下是mpp_buffer的操作：

[image]

[image]

HermanChen, 2025-07-07:
参考下 mpi_rc2_test