rockchip-linux/mpp #147: H265解码与H264解码后的MppFrame编码成JPG的问题

programcj, 2020-07-10:
我读取H264格式RTSP流1920*1080分辨率，解码后得到Mpp_Frame，从中读取Mpp_Buffer，参考例子编码成JPG没问题

但是当用H265格式RTSP流，对解码后的Mpp_Frame中的Mpp_Buffer 编码成Jpg就出现问题，
H264格式解码 1920*1080 的 hor_stride与ver_stride为1920*1088
H265格式(hevc_rkmpp)解码 1920*1080 的 hor_stride与ver_stride为2304x1080

然后使用RGA转换 H265变成 hv变成16对齐（1920*1088），再编码成jpg照片就不会花；

不知道为什么使用 H265解码的Mpp_Frame 1920*1080 (2304x1080) 的Mpp_Buffer编码成Jpg就不正常？
我使用的是ubuntu16.04 MPP版本是：
```
mpi: mpp version: 1babbc9c author: Ding Wei [h264d]: fix bug: fast output frame
```

qvoid, 2020-07-13:
写出来的stride是怎么样的，就需要按照那样的stride去读取YUV，JPEG编码应该是默认hor_stride 是16对齐。所以读取YUV时错乱了。

programcj, 2020-07-14:
@qvoid JPG编码必须是16对齐的数据吗？ 但是H265解码后的Mpp_Frame 1920x1080 的 stride为 2304x1080了， 编码器 MPP_ENC_SET_PREP_CFG我也是按照解码后的 stride设置的
```
prep_cfg->width = 1920;
prep_cfg->height = 1080;
prep_cfg->hor_stride = 2304;
prep_cfg->ver_stride = 1080;
```
在编码的时候
```
mpp_frame_set_width(frame, 1920);
mpp_frame_set_height(frame, 1080);
mpp_frame_set_hor_stride(frame, 2304);
mpp_frame_set_ver_stride(frame, 1080);
encode_put_frame(ctx, frame);
```
当我讲 1920的width写成2304是可以出图片，只是宽度的右边会多一块绿布
原始的1920又不会出正常图片

qvoid, 2020-07-16:
宽度2304的编码出一块绿，是因为右边的YUV数据全是0，这是因为265解码出来的Buffer是奇数256对齐。你之前用RGA去裁剪的做法是正确的。

qvoid, 2020-07-16:
JPEG编码输入的YUV宽度至少是要8对齐才可以

programcj, 2020-07-16:
@qvoid 这是硬件解码决定的吗？奇怪为什么要对H265解码的做基数对齐

qvoid, 2020-07-16:
硬件决定，265的输出也可以配成16对齐，但是效率会低一些

haithink, 2021-09-15:
> 硬件决定，265的输出也可以配成16对齐，但是效率会低一些

请问怎么配成16对齐呢？谢谢！

HermanChen, 2021-09-16:
https://github.com/rockchip-linux/mpp/blob/02058ce67a244ad50e589104a1fa52e724c303e5/mpp/hal/rkdec/h265d/hal_h265d_com.c#L89
这个 hevc_hor_align 函数里定义了 hevc 在 rkvdec 上的对齐方式