rockchip-linux/mpp #148: h264 编码idr帧模糊

caryluck, 2020-07-13:
我用mpi_enc_test例子循环编码一张yuv图片，编码出来的每个idr帧图像下半部分出现模糊。368*640的yuv第一帧idr大小是为10110字节，后面的p真1024左右，但是后面的每个idr帧为5000字节左右，后面连续的几个p帧3000-5000左右，要经过3个p左右画面才会变清晰，编出来的视频效果就是出现闪烁。如果第一帧idr后面全是p帧，有概率一小段时候后画面出现马赛克，这个该怎么解决？

qvoid, 2020-07-13:
编码参数是什么样的？可以尝试增大bitrate。

caryluck, 2020-07-13:
先感谢@qvoid, 用的例子中的设置rc_mode CBR固定码率,大小是是w*h/8*30大小。我调整到6倍大小后确实好很多了，但还是能看出来idr帧前后帧的细微变化。
还有一个问题，同一张图片第一帧idr gop设置成2000一直编码，后续会有出现马赛克。

qvoid, 2020-07-14:
固定码率是 w*h/830 ?这个有问题。码率应该是太小了。是不是其实想设置成 ｗ * h* 30 / 8?

caryluck, 2020-07-14:
是设置成 w * h/ 8 * 30， 我现在码率改成 w * h * 30  gop设置30，画面变化情况还是会闪，而且再设大几倍也没什么效果

qvoid, 2020-07-14:
试试看打上下面这个补丁是否有改善
```
--- a/mpp/hal/vpu/h264e/hal_h264e_vepu_v2.c
+++ b/mpp/hal/vpu/h264e/hal_h264e_vepu_v2.c
@@ -599,8 +599,12 @@ MPP_RET h264e_vepu_mbrc_prepare(HalH264eVepuMbRcCtx ctx, HalH264eVepuMbRc *mbrc,
 
     srcPrm = axb_div_c(mbrc->out_strm_size * 8, 256, mbrc->rlc_count);
     /* Disable Mb Rc for Intra Slices, because coeffTarget will be wrong */
-    if (frm->is_intra || srcPrm == 0)
-        return 0;
+    if (frm->is_intra || srcPrm == 0) {
+        mbrc->mad_qp_change = 0;
+        mbrc->mad_threshold = 0;
+        mbrc->cp_distance_mbs = 0;
+        return MPP_OK;
+    }

```

caryluck, 2020-07-14:
打上这个patch后，之前出现画面持续脏乱的问题没有出现了，画面闪的问题看起来也好了。但是还有一个画面问题，一直在画面变动的场景，不定期的时间10多秒或者更长时间偶现画面闪电一样的裂纹有时候出现从中心爆炸一样的画面，现在设置依然是gop 60, CBR w * h/ 8 * 30的码率

caryluck, 2020-07-14:
正常画面
[image]
异常画面
[image]

HermanChen, 2020-07-15:
抓一段码流下来看下，这种的情况可能是码流有错或者有丢，导致参考帧乱了

caryluck, 2020-07-16:
gop设置成60时会偶尔闪现错乱画面，设置大的值没有发现，猜测还是idr帧引起的。抓码流我可以在编码时候保存yuv数据和h264数据比较吧？

还有一个发现宽 720 高1080图像编码比368*640的编码流输出到远端解码延迟要明显，720*1080和1080*720在编码上会有性能上差别吧？

HermanChen, 2020-07-16:
对，同样面积的图像，高度多，行数多的情况下编码性能会下降一些

caryluck, 2020-07-16:
我现在做视频实时传输，实时性要求较高，同时画质也要有保证。在降低延时方面，编码端和解码端有哪些参数可以设置或者有什么建议？

HermanChen, 2020-07-16:
硬件编码器一般都没有 B 帧，解码器也是 MPP 的话，可以加个 IMMEDIATE_OUT 的 control，输出会快一些

YouRancestor, 2020-07-17:
> 是设置成 w * h/ 8 * 30， 我现在码率改成 w * h * 30 gop设置30，画面变化情况还是会闪，而且再设大几倍也没什么效果

请问修改码率是怎么设置的？有效果吗？我测试CBR不管bps_target填多少都对输出没有影响 #152

caryluck, 2020-07-30:
> 试试看打上下面这个补丁是否有改善
>
> ```
> --- a/mpp/hal/vpu/h264e/hal_h264e_vepu_v2.c
> +++ b/mpp/hal/vpu/h264e/hal_h264e_vepu_v2.c
> @@ -599,8 +599,12 @@ MPP_RET h264e_vepu_mbrc_prepare(HalH264eVepuMbRcCtx ctx, [...]

这个patch能解决idr模糊抖动问题，但是测试下下来上面图像爆炸也是这个patch引起的

JeffyCN, 2020-08-26:
try to modify qp limits:
https://github.com/rockchip-linux/gstreamer-rockchip/commit/3bcc77fe9619a340ae3d06ef036511e5f7b7544d#diff-09f79b5ecd71013a58a5adb6c7759e55L91