rockchip-linux/mpp #150: 感觉rga.h中定义的RgaFormat有问题

YouRancestor, 2020-07-15:
经测试rga.h中定义的format名称和程序实际的行为不一致，也无法解释通。
我认为`RGA_FMT_RGBA_8888`和`RGA_FMT_BGRA_8888`应调换位置，即`RGA_FMT_BGRA_8888=0x0`，`RGA_FMT_RGBA_8888=0x3`。原始rga.h代码：
```c
typedef enum RgaFormat_e {
    RGA_FMT_RGBA_8888       = 0x0,
    RGA_FMT_RGBX_8888       = 0x1,
    RGA_FMT_RGB_888         = 0x2,
    RGA_FMT_BGRA_8888       = 0x3,
    RGA_FMT_RGB_565         = 0x4,
    RGA_FMT_RGBA_5551       = 0x5,
    RGA_FMT_RGBA_4444       = 0x6,
    RGA_FMT_BGR_888         = 0x7,

    RGA_FMT_YCbCr_422_SP    = 0x8,
    RGA_FMT_YCbCr_422_P     = 0x9,
    RGA_FMT_YCbCr_420_SP    = 0xa,
    RGA_FMT_YCbCr_420_P     = 0xb,

    RGA_FMT_YCrCb_422_SP    = 0xc,
    RGA_FMT_YCrCb_422_P     = 0xd,
    RGA_FMT_YCrCb_420_SP    = 0xe,
    RGA_FMT_YCrCb_420_P     = 0xf,

    RGA_FMT_BPP1            = 0x10,
    RGA_FMT_BPP2            = 0x11,
    RGA_FMT_BPP4            = 0x12,
    RGA_FMT_BPP8            = 0x13,
    RGA_FMT_BUTT,
} RgaFormat;
```
另外rga.cpp中实现的Mpp格式到RGA格式的映射也让人摸不着头脑
```c
static RgaFormat rga_fmt_map(MppFrameFormat fmt)
{
    RgaFormat ret;

    switch (fmt) {
    case MPP_FMT_YUV420P:
        ret = RGA_FMT_YCbCr_420_P;
        break;
    case MPP_FMT_YUV420SP:
        ret = RGA_FMT_YCbCr_420_SP;
        break;
    case MPP_FMT_YUV422P:
        ret = RGA_FMT_YCbCr_422_P;
        break;
    case MPP_FMT_YUV422SP:
        ret = RGA_FMT_YCrCb_422_SP;
        break;
    case MPP_FMT_RGB565:
        ret = RGA_FMT_RGB_565;
        break;
    case MPP_FMT_RGB888:
        ret = RGA_FMT_RGB_888;
        break;
    case MPP_FMT_ARGB8888:
        ret = RGA_FMT_RGBA_8888;  // 为什么ARGB要映射为RGBA？这怎么讲都说不通
        break;
    default:
        ret = RGA_FMT_BUTT;
        mpp_err("unsupport mpp fmt %d found\n", fmt);
        break;
    }

    return ret;
}
```
我现在将这个映射修改为
```c
    case MPP_FMT_BGRA8888:
        ret = RGA_FMT_RGBA_8888;
        break;
    case MPP_FMT_RGBA8888:
        ret = RGA_FMT_BGRA_8888;
        break;
```
输入数据的排布跟MPP定义的名称一致，输出符合预期，但MPP格式映射到RGA格式的时候RGBA和BGRA是反的
所以如果再将rga.h中的两个名称定义调换就能全部一致了

注1：
测试程序使用的是本项目中的rga_test.cpp，输入是RGBA或BGRA的图像数据，输出格式是YUV420P，命令：
```shell
./rga_test -i test.rgba -o test.yuv -w 1920 -h 1080 -f 65549 -dst_w 1920 -dst_h 1080 -dst_fmt 4
```
其中输入格式代码65549对应的是MPP_FMT_RGBA8888，在rga.cpp中映射为RGA_FMT_BGRA_8888(0x3)是我按上述代码修改过的，总之要把rgba的数据对应到rga的0x3才正确
test.rgba文件可使用ffmpeg命令由任意图像文件test.bmp得到：
```shell
ffmpeg -i test.bmp -vcodec rawvideo -pix_fmt rgba test.rgba.yuv // 输出rgba格式的裸数据（文件后缀须指定为yuv，否则ffmpeg报错）
mv test.rgba.yuv test.rgba // 去掉后缀
```
rga_test转换完成后，使用ffplay观看
```shell
ffplay -f rawvideo -video_size 1920x1080 -pix_fmt yuv420p  test.yuv
```
注2：
rga.cpp中有bug，原始版本无法完成rgb转yuv，修改后的代码（不含映射部分）已提交pull request。
注3：
修改映射后的完整[rga.cpp](https://github.com/YouRancestor/mpp/blob/egret/mpp/mpp.cpp)

HermanChen, 2020-07-15:
多谢，主要是对颜色格式的大小端解释问题……
这个问题本身比较繁杂，还有驱动写法有关 @qvoid

YouRancestor, 2020-07-15:
> 多谢，主要是对颜色格式的大小端解释问题……
> 这个问题本身比较繁杂，还有驱动写法有关 @qvoid

我想到过大小端的因素，但大端RGBA应对应小端ABGR不是吗？顺序倒过来。而现有代码里的定义完全无法理解。
而且驱动内部的问题可以通过定义和映射让外部不感知，例如使用我提出的将api中两个定义调换的方法。

YouRancestor, 2020-07-15:
另外8888的四字节数据格式应该都是指大端的，ffmpeg中4字节的rgba格式都不区分大小端，而565，444的格式会明确指出大小端，如rgb565le和rgb565be。

qvoid, 2020-07-15:
可以提一个 pull request。RGA这块的定义确实有些问题。RGBA的格式对应的各个分量的排布，不同的软件和平台定义会有些差别除了大小端，还需要考虑是字序还是字节序，可以参考[Wikipedia](https://en.wikipedia.org/wiki/RGBA_color_model)的描述。ffmpeg的定义是按照字节序列，默认大端。我们内部也有个关于RGB定义的补丁，在review中。你的改动我们会一并验证后整合

YouRancestor, 2020-07-15:
> 可以提一个 pull request。RGA这块的定义确实有些问题。RGBA的格式对应的各个分量的排布，不同的软件和平台定义会有些差别除了大小端，还需要考虑是字序还是字节序，可以参考[Wikipedia](https://en.wikipedia.org/wiki/RGBA_color_model)的描述。ffmpeg的定义是按照字节序列，默认大端。我们内部也有个关于RGB定义的补丁，在review中。你的改动我们会一并验证后整合

pull request我就不提了，因为我不知道内部的实现细节，所以我没敢动rga.h里的定义，我目前就反着映射RGBA和BGRA够用了，我也没用过565等非4字节的rgb格式，也没测过不知道有没有类似的问题。所以我觉得这个问题还是得你们自己来修复，而且mpp引用了rga.h里的定义来比大小比如如下代码，都得跟着一起改
```c
static int is_rgb_format(int fmt)
{
    if (fmt >= RGA_FMT_RGBA_8888 && fmt <= RGA_FMT_BGR_888) {
        return 1;
    }

    return 0;
}
```
建议多定义一个`RGA_FMT_RGB_FIRST`和一个`RGA_FMT_RGBA_LAST`这样的东西来比大小用。
```c
typedef enum RgaFormat_e {
    RGA_FMT_RGB_FIRST = 0x0,
    RGA_FMT_RGBA_8888       = 0x0,
    RGA_FMT_RGBX_8888       = 0x1,
    RGA_FMT_RGB_888         = 0x2,
    RGA_FMT_BGRA_8888       = 0x3,
    RGA_FMT_RGB_565         = 0x4,
    RGA_FMT_RGBA_5551       = 0x5,
    RGA_FMT_RGBA_4444       = 0x6,
    RGA_FMT_BGR_888         = 0x7,
    RGA_FMT_RGBA_LAST = 0x7,

    RGA_FMT_YUV_FIRST = 0x8,
    RGA_FMT_YCbCr_422_SP    = 0x8,
    RGA_FMT_YCbCr_422_P     = 0x9,
    RGA_FMT_YCbCr_420_SP    = 0xa,
    RGA_FMT_YCbCr_420_P     = 0xb,

    RGA_FMT_YCrCb_422_SP    = 0xc,
    RGA_FMT_YCrCb_422_P     = 0xd,
    RGA_FMT_YCrCb_420_SP    = 0xe,
    RGA_FMT_YCrCb_420_P     = 0xf,
    RGA_FMT_YUV_LAST = 0xf,

    RGA_FMT_BPP1            = 0x10,
    RGA_FMT_BPP2            = 0x11,
    RGA_FMT_BPP4            = 0x12,
    RGA_FMT_BPP8            = 0x13,
    RGA_FMT_BUTT,
} RgaFormat;

YouRancestor, 2020-07-15:
@qvoid 对了 有没有ioctl的rga参数文档？

qvoid, 2020-07-15:
> @qvoid 对了 有没有ioctl的rga参数文档？

参考这里 https://github.com/AngerLee/librga

JeffyCN, 2020-07-23:
rga formats are following opengl formats.

closed 2020-07-23 by JeffyCN