rockchip-linux/mpp #151: 能够将opencv bgr Mat 转为yuv 数据，再采用mpp编码h264

MUZLATAN, 2020-07-15:
目前，使用mpi_enc_test 可以将前面的yuv文件转换为h264文件，但是现在项目需求需要保存生产过程中opencv mat的数据， 当我使用
```  
 cap.read(rgbImg);
 cv::cvtColor(rgbImg, yuvImg, CV_BGR2YUV_YV12);
 if (count++ > 100)
    p->frm_eos = 1;

 memcpy(buf, yuvImg.data, yuvImg.cols*yuvImg.rows);
```
保存的h264文件通过ffplay 查看会出现花屏，不知道是什么问题

有两点疑问：
1. put_frame 中frame 是什么样的数据格式   yuv420p  还是 yuv420sp
2. 单独的写一帧yuv数据是否可行，需不需要添加头部信息，通过read_image()函数发现，里面有头部数据信息

HermanChen, 2020-07-15:
1. 硬件支持 420SP 和 420P，送入的 MppFrame 正确标记即可
2. 不行，H.264 一定要有 sps/pps 这些信息才能解码

MUZLATAN, 2020-07-15:
你们有没有对应的使用cv::mat 编码的demo, 我自己使用opencv Mat 的yuv数据，编码失败， mpi_enc_test 是从文件读取的， 我想研究下如何从cv::mat 编码

MUZLATAN, 2020-07-15:
在写文件之前有H.264的 sps/pps  的头部信息

```
    cv::VideoCapture cap;
    cap.open("rtsp://admin:Buzhongyao123@192.168.200.228:554/h264/ch39/main/av_stream");
    cv::Mat rgbImg;
    cv::Mat yuvImg;
    int count = 0;

    while (!p->pkt_eos) {
        MppFrame frame = NULL;
        MppPacket packet = NULL;
        void *buf = mpp_buffer_get_ptr(p->frm_buf);

//        if (p->fp_input) {
//            ret = read_image((RK_U8*)buf, p->fp_input, p->width, p->height,
//                             p->hor_stride, p->ver_stride, p->fmt);
//            if (ret == MPP_NOK || feof(p->fp_input)) {
//                p->frm_eos = 1;
//
//                if (p->num_frames < 0 || p->frame_count < p->num_frames) {
//                    clearerr(p->fp_input);
//                    rewind(p->fp_input);
//                    p->frm_eos = 0;
//                    mpp_log("loop times %d\n", ++p->loop_times);
//                    continue;
//                }
//                mpp_log("found last frame. feof %d\n", feof(p->fp_input));
//            } else if (ret == MPP_ERR_VALUE)
//                return ret;
//        } else {
//            ret = fill_image((RK_U8*)buf, p->width, p->height, p->hor_stride,
//                             p->ver_stride, p->fmt, p->frame_count);
//            if (ret)
//                return ret;
//        }
        cap.read(rgbImg);
        cv::cvtColor(rgbImg, yuvImg, CV_BGR2YUV_YV12);
        if (count++ > 100)
            p->frm_eos = 1;

        memcpy(buf, yuvImg.data, yuvImg.cols*yuvImg.rows);

//省略后续操作
........

}
```
我将mpi_enc_test 里面的从文件读数据改为从cv::mat 转换数据再编码， 但是最终编码数据不正常，会是什么问题
你说的  "送入MppFrame 正确标记即可" 是在哪里标记

MUZLATAN, 2020-07-16:
问题已解决 ，缓冲区大小设置过小导致不能编码2560x1440的图片

closed 2020-07-16 by MUZLATAN