rockchip-linux/mpp #152: rate control has no effect

YouRancestor, 2020-07-16:
码率控制无效（CBR）
```c    
    MppEncRcCfg rcCfg;
    memset(&rcCfg, 0, sizeof(rcCfg));
    rcCfg.change = MPP_ENC_RC_CFG_CHANGE_ALL ^ MPP_ENC_RC_CFG_CHANGE_QUALITY;
    rcCfg.rc_mode = MPP_ENC_RC_MODE_CBR;
    rcCfg.bps_target = 0; // 这里填0,40000,400000,4000000效果都一样
    rcCfg.gop = 30; // 这里填30,90也是对输出没有影响
    ret = mpi->control(ctx, MPP_ENC_SET_RC_CFG, &rcCfg);
```
测试方法是对一个固定的yuv文件编码，打印编码后每一个nal的长度，修改上述参数对输出完全没有影响。
另外试了一下mode改为VBR，并设置了bps_min和bps_max，崩溃了。

YouRancestor, 2020-07-28:
找到问题了：文档说bps_target管cbr，而bps_min和bps_max管vbr，实际上这仨都要填；文档说fps_*一系列参数填0有默认值，实际上没有，memset为0是不行的，需要挨个赋值。

closed 2020-07-28 by YouRancestor

Tristin-9527, 2021-01-14:
> 找到问题了：文档说bps_target管cbr，而bps_min和bps_max管vbr，实际上这仨都要填；文档说fps_*一系列参数填0有默认值，实际上没有，memset为0是不行的，需要挨个赋值。

我怎么调都没法让码率稳定下来， 静态视频和动态视频码率总是会飘， 范围挺大，知道什么原因吗？