rockchip-linux/mpp #155: [ question ] Kernel 4.19

avafinger, 2020-08-01:
Will this work with the Kernel 4.19?

avafinger, 2020-08-03:
I compiled mpp and linux-rga and ffmpeg on kernel 4.19.111 and i got problems with RGA colorspace while decoding.

Here is the ffmpeg output and the rga device error:

```
ls -la /dev/rga
crw------- 1 root root 10, 59 Jan 28  2018 /dev/rga

```
```
[AVHWDeviceContext @ 0x5577e72b60] Opened DRM device /dev/dri/card0: driver rockchip version 2.0.0.
[hevc_rkmpp @ 0x5577e764d0] Format nv12 chosen by get_format().
[hevc_rkmpp @ 0x5577e764d0] Initializing RKMPP decoder.
mpp[629]: mpp_info: mpp version: 4a61fda3 author: sayon.chen    2020-07-31 [vepu_api_v2]: Fix rotation issue
mpp[629]: mpp_rt: NOT found ion allocator
mpp[629]: mpp_rt: found drm allocator
mpp[629]: mpp_log: got the /dev/mpp_service
mpp[629]: mpp: deprecated block control, use timeout control instead
mpp[629]: mpp: deprecated block control, use timeout control instead
[hevc_rkmpp @ 0x5577e764d0] RKMPP decoder initialized successfully.
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (hevc_rkmpp) -> rawvideo (native))
Press [q] to stop, [?] for help
cur_dts is invalid (this is harmless if it occurs once at the start per stream)
[hevc_rkmpp @ 0x5577e764d0] Wrote 111 bytes to decoder
mpp[629]: H265D_PARSER: No start code is found.
[hevc_rkmpp @ 0x5577e764d0] Wrote 237527 bytes to decoder
[hevc_rkmpp @ 0x5577e764d0] Decoder noticed an info change (1920x1080), format=1
[AVHWFramesContext @ 0x5577e901d0] Unsupported format: (null).
Error while decoding stream #0:0: Invalid argument
cur_dts is invalid (this is harmless if it occurs once at the start per stream)
[hevc_rkmpp @ 0x5577e764d0] Wrote 154149 bytes to decoder
[hevc_rkmpp @ 0x5577e764d0] Received a frame.
fd-vir-phy-hnd-format[16, (nil), (nil), 0, 0] 
rect[0, 0, 1920, 1080, 2816, 1080, 10, 0] 
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1] 
fd-vir-phy-hnd-format[-1, 0x7f84ee4010, (nil), 0, 0] 
rect[0, 0, 1920, 1080, 1920, -546, 10, 0] 
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1] 
This output the user patamaters when rga call blit fail 
Failed to do rga blit
detected 6 logical cores
[graph 0 input from stream 0:0 @ 0x5577e858e0] Setting 'video_size' to value '1920x1080'
[graph 0 input from stream 0:0 @ 0x5577e858e0] Setting 'pix_fmt' to value '23'
[graph 0 input from stream 0:0 @ 0x5577e858e0] Setting 'time_base' to value '1/1000'
[graph 0 input from stream 0:0 @ 0x5577e858e0] Setting 'pixel_aspect' to value '1/1'
[graph 0 input from stream 0:0 @ 0x5577e858e0] Setting 'sws_param' to value 'flags=2'
[graph 0 input from stream 0:0 @ 0x5577e858e0] Setting 'frame_rate' to value '2997/100'
[graph 0 input from stream 0:0 @ 0x5577e858e0] w:1920 h:1080 pixfmt:nv12 tb:1/1000 fr:2997/100 sar:1/1 sws_param:flags=2
[format @ 0x5577e85df0] Setting 'pix_fmts' to value 'bgra'
[auto_scaler_0 @ 0x5577e87080] Setting 'flags' to value 'bicubic'
[auto_scaler_0 @ 0x5577e87080] w:iw h:ih flags:'bicubic' interl:0
[format @ 0x5577e85df0] auto-inserting filter 'auto_scaler_0' between the filter 'Parsed_null_0' and the filter 'format'
[AVFilterGraph @ 0x5577e84400] query_formats: 4 queried, 2 merged, 1 already done, 0 delayed
[auto_scaler_0 @ 0x5577e87080] w:1920 h:1080 fmt:nv12 sar:1/1 -> w:1920 h:1080 fmt:bgra sar:1/1 flags:0x4

```

Without the **/dev/rga** i can get the output image but with RGB issues.

kernel config for RGA:

```
#
# RGA
#
# CONFIG_ROCKCHIP_RGA is not set

#
# RGA2
#
CONFIG_ROCKCHIP_RGA2=y

```

How could i fix this?

HermanChen, 2020-08-03:
mpp can work with 4.19 kernel.
The ffmpeg support need to be updated. We will check it.

JeffyCN, 2020-08-03:
please use gstreamer.

lihuang1111, 2020-08-03:
@avafinger 

you pass invalid patamater(-546) to rga. Please check hstride on dst.

`rect[0, 0, 1920, 1080, 1920, -546, 10, 0] `

JeffyCN, 2020-08-03:
> @avafinger
> 
> you pass invalid patamater(-546) to rga. Please check hstride on dst.
> 
> `rect[0, 0, 1920, 1080, 1920, -546, 10, 0] `

that is because the caller of ffmpeg is using it's custom buffer allocator, which provided discontig buffers and cause wrong hstride(calculated from y/uv offsets).

avafinger, 2020-08-03:
Thank you all for your prompt response!

If i understand it right FFmpeg is to blame.
Just for the record, i built the rockchip/release/4.1 branch.

avafinger, 2020-08-08:
> > @avafinger
> > you pass invalid patamater(-546) to rga. Please check hstride on dst.
> > `rect[0, 0, 1920, 1080, 1920, -546, 10, 0] `
> 
> that is because the caller of ffmpeg is using it's custom buffer allocator, which provided discontig buffers and cause wrong hstride(calculated from y/uv offsets).

Hi @JeffyCN 
Do you have any update on this?
Can you point me to the sources you think i should look at? Although Not sure i could fix it.

BR

JeffyCN, 2020-08-10:
please use gstreamer or mpp(directly).

if you want to know more about this issue, there are multiple ways to make it work:
1/ modify your get_buffer2 to alloc contig buffers(keep yuv planes configuous), since the librga currently only support passing single buffer address now.
2/ or use avcodec_default_get_buffer2:
https://github.com/rockchip-linux/ffmpeg/blob/rockchip/release/4.1/libavcodec/rkmppdec.c#L589
3/ or implement get_format to select AV_PIX_FMT_DRM_PRIME(sw format NV12), and handle it:
http://ixia.jkqxz.net/~mrt/kms_va_nv12.c
https://github.com/rockchip-linux/mpv/blob/master/video/out/vo_xv.c#L874
4/ or fallback to a software slow conversion:
https://github.com/JeffyCN/FFmpeg/blob/master/libavcodec/rkmppdec.c#L413

avafinger, 2020-08-10:
@JeffyCN 

Thank you for the info!

_please use gstreamer or mpp(directly)._
I was building for KODI. i will study your code and try to do something or if i can't fix it will wait for your implementation.

Thanks
Regard

mrfixit2001, 2022-03-08:
Sorry to necromance this post a bit - but I'm also investigating into using librga with kodi to allow for screen rotation. Looking for any feedback and advice. Using BSP kernel at the moment.