rockchip-linux/mpp #156: Qt+gstreamer+mpp  play video file long time will be stuck at mpp get_frame

jasonwxj2009, 2020-08-01:
Hi,
I use Qt + gstreamer+mpp to play video continuously， about 4 hour later  （sometimes less than ），video will stuck，code at:
`
 if (mOutputTimeout < 0) {
                /* block wait */
                mFrames->wait();
`
Qt application is dead lock , should be restart！
Is there any solution？

HermanChen, 2020-08-03:
What the version of mpp and gst?

jasonwxj2009, 2020-08-03:
thanks for replying!
gstreamer version 1.14 

mpp_info_test: mpp revision is -1
mpp_info_test: mpp info all:
1614c5d4 author: Caesar Wang debian: add rules for mpp with 20191031
mpp_info_test: mpp info revision: 1614c5d472dad37e81597d16bf7def2a7a429312
mpp_info_test: mpp info date    :
mpp_info_test: mpp info author  : Caesar Wang

and I have also compile the libs  with gstreamer-rockchip  and mpp use the newest version  
mpp[25815]: mpp_info_test: normal version log:
mpp[25815]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[25815]: mpp_info_test: history version log:
mpp[25815]: mpp_info: mpp version history 0:

The problem is still  repeated.

below is bt info:
#0  0x0000007f77e0522c in pthread_cond_wait@@GLIBC_2.17 () at /lib/aarch64-linux-gnu/libpthread.so.0
#1  0x0000007f5c287524 in Mpp::get_frame(void**) () at /usr/lib/aarch64-linux-gnu/librockchip_mpp.so.1
#2  0x0000007f5c28ab6c in mpi_decode_get_frame(void*, void**) () at /usr/lib/aarch64-linux-gnu/librockchip_mpp.so.1
#3  0x0000007f5c46e2ac in gst_mpp_dec_buffer_pool_acquire_buffer (bpool=0x7f2c0126f0, buffer=0x7f357417b0, params=<optimized out>) at gstmppdecbufferpool.c:241
#4  0x0000007f5f490600 in gst_buffer_pool_acquire_buffer () at /usr/lib/aarch64-linux-gnu/libgstreamer-1.0.so.0
#5  0x0000007f5c471254 in gst_mpp_video_dec_loop (decoder=0x7f2c03a420) at gstmppvideodec.c:509
#6  0x0000007f5f502eb4 in  () at /usr/lib/aarch64-linux-gnu/libgstreamer-1.0.so.0
#7  0x0000007f77a04000 in  () at /usr/lib/aarch64-linux-gnu/libglib-2.0.so.0

jasonwxj2009, 2020-08-07:
I have used qt+mpv+mpp instead of gstreamer

closed 2020-08-07 by jasonwxj2009