rockchip-linux/mpp #159: mjpeg解码 循环读jpg文件 运行不到一分钟 出现错误 参考demo mpi_dec_test

Archer112, 2020-08-26:
pi_dec_test start
input file size 960000
mpi_dec_test decoder test start w 1280 h 720 type 8
mpi: mpp version: 0b701fe author: Lawrence-Tang DEBIAN: prepare for release_20171218-3
mpp_device: mpp_device_init failed to open device /dev/vpu_service, errno 24, error msg: Too many open files
HAL_JPEG_VDPU2: hal_jpegd_vdpu2_init get vpu_socket(-1) <= 0, failed. 
mpp_hal: mpp_hal_init hal jpegd init failed ret -2
mpp_hal: mpp_hal_init could not found coding type 8
mpp_dec: mpp_dec_init could not init hal
mpp: error found on mpp initialization
mpp_dec: mpp_dec_control found NULL input dec (nil)
mpp: command 31000a param 0x7fce15fc04 ret -3
mpp input poll failed
found last packet
mpp input poll failed
mpi->reset failed

HermanChen, 2020-08-27:
Too many open files 这里应该是有 fd 泄漏

Archer112, 2020-08-27:
> Too many open files 这里应该是有 fd 泄漏

我单独测试了这个demo 每50ms解一次图片 类似下面这样 是会出现这个错误 您看看是因为我使用方法不对？
while(1)
{
    for(uint8_t  i = 1; i < 26; i++)
    {
        mpi_dec_test_decode();
        usleep(50*1000);
    }
}

HermanChen, 2020-08-27:
20171218 这个版本感觉比较旧了，最好更新下代码

Archer112, 2020-08-27:
> 20171218 这个版本感觉比较旧了，最好更新下代码

好的 我更新试试 谢谢

Archer112, 2020-08-27:
> 20171218 这个版本感觉比较旧了，最好更新下代码

使用新的版本解决了 谢谢

alexanderdumas, 2023-11-23:
能给个源码吗？你的jpg文件是BGR的吗？BGR，经过jpeg压缩的吗？