rockchip-linux/mpp #161: Support for kernel 5.x

gusarg81, 2020-08-26:
Hi,

I would like to know is there a support for Kernel 5.7.6, which is the one that comes with Armbian (for rock64 hardware, Armbian Focal)?

I know that same distro version includes legacy kernel (4.x), But I just want to know.

Thanks.

keveryang, 2020-08-28:
Hi, 
Rockchip don't have plan to support 5.7.6. we have 4.4 and 4.19 now, and may support next new LTS kernel to be release this year.
Thanks.

gusarg81, 2020-08-28:
Damn Rockchip... Thanks for the info.

gusarg81, 2020-08-28:
> Hi,
> Rockchip don't have plan to support 5.7.6. we have 4.4 and 4.19 now, and may support next new LTS kernel to be release this year.
> Thanks.

closed 2020-08-28 by gusarg81

bakwc, 2021-08-25:
Hi! Any chance to see it on 5.x kernel?