rockchip-linux/mpp #162: YUYV broken after commit 1edcff8a6b5d9a0c558c8793643a8d15da2649a4

jpiat, 2020-09-04:
YUYV work at commit 1edcff8a6b5d9a0c558c8793643a8d15da2649a4 but result in the following error for earlier commits

(https://github.com/rockchip-linux/mpp/commit/1edcff8a6b5d9a0c558c8793643a8d15da2649a4)

hal_h264e_vepu_v2: warning: input buffer size 0x3fc000 is smaller then required size 0x7f8000

seems like the encoder expects a buffer of size 1920*1088*4 for a 1080p input while YUYV encoding result in 1920*1088*2 bytes

jpiat, 2020-09-04:
issue was actually introduced with 86aaf61837c14ae6c09656875f3a18836e8a877d

HermanChen, 2020-09-04:
Yes, the hardware requires 16 aligned buffer to encode. Otherwise hardware will generate error if buffer is too small.

jpiat, 2020-09-04:
The buffer is 16bits aligned (it was a requirement before this commit already), the error i get (copied in my post) is about the size not the alignment.

jpiat, 2020-09-04:
to get 16 bits alignment, my buffer is 1088*1920

jpiat, 2020-09-04:
I think that the erro comes from the following

RK_S32 hor_stride = MPP_ALIGN(cfg->width, 16);

was replaced with 

RK_S32 hor_stride = cfg->hor_stride;

for YUYV cfg->hor_stride equals (1088*2) wile MPP_ALIGN(cfg->width, 16) equals 1088. This creates an issue with the computation of the bufer size that is

prep->size_y = hor_stride * 2 * MPP_ALIGN(prep->src_h, 16);

for MPP_FMT_YUV422_UYVY. This expected buffer size thuse doubled. Does it means that for now, cfg->hor_stride should be 1088 instead of being 1088*2 ?

jpiat, 2020-09-07:
how can i test mpi_enc_test with format  MPP_FMT_YUV422_YUYV and h264 encoding ? It seem that there is no fill function for this format.

HermanChen, 2020-09-08:
I updated the whole branch by fixing the patch with error.
Could you force update and have a try?

jpiat, 2020-09-08:
Hi,

the fix worked

Thanks

closed 2020-09-08 by jpiat