rockchip-linux/mpp #164: mpi_enc_test编码出来的H.264码流解码花屏

0x79696e67, 2020-11-16:
平台：RK3399

MPP版本：Android_9_industry_20190904_1.00

修改mpi_enc_test.c中fps为25，gop为25，bps为4000000（附件中已经提供修改好的mpi_enc_test.txt）
[mpi_enc_test.txt](https://github.com/rockchip-linux/mpp/files/5545819/mpi_enc_test.txt)

mpi_enc_test运行参数：-i frames_1920x1080_rgba8888  -o frames.h264 -width 1920 -height 1080 -t 7 -f 65546

数据帧：见附件frames_1920x1080_rgba8888.gz压缩包。格式：RGBA8888，分辨率：1920x1080  [frames_1920x1080_rgba8888.gz](https://github.com/rockchip-linux/mpp/files/5545815/frames_1920x1080_rgba8888.gz)

对数据帧进行编码，发现最后一帧是“破碎”的。ffmpeg、StreamEye和MPP都无法解码编码出来的H.264数据。

HermanChen, 2020-11-16:
最后一帧可能是写文件的时候没做 fflush？

0x79696e67, 2020-11-17:
> 最后一帧可能是写文件的时候没做 fflush？

不是，只是方便展示问题把错误帧放在最后一帧。继续编码下去写入文件或者flush，出错的那一帧还是一样。

0x79696e67, 2020-11-20:
当编码CBR并且用RGB565色彩空间时，在[最新](https://github.com/rockchip-linux/mpp/commit/d1e47f8137c8c568fe47c42fa9723b88aa016d8f)的开发版中也存在相似的问题。
运行命令参数：mpi_enc_test -i frames_rga565_1920x1080 -o frames.h264 -width 1920 -height 1080 -t 7 -f 0x1010000 -b 4000000:0:0:1 -r 25:1:0/25:1:0 -g 0:25:0
编码时使用的数据帧：[frames_rga565_1920x1080.7z.gz](https://github.com/rockchip-linux/mpp/files/5572089/frames_rga565_1920x1080.7z.gz) 格式为：RGB565 大小为1920x1080 附件大小限制用了gz和7z压缩，需要解压两次。