rockchip-linux/mpp #165: 如何将cv::Mat包装成mpp_buffer?

xiayip, 2020-11-24:
你好, 我尝试直接将一个cv::Mat中的内容包装成, mpp_buffer, 但是这样的做法:

``` C
         MppBuffer inputbuf;
         MppBufferInfo commit;
         commit.type  = MPP_BUFFER_TYPE_NORMAL;
         commit.size  = 640 * 480 * 4;
         commit.ptr   = img_data.data;  // img_data is an exist 'BGRA' cv::Mat
         commit.fd    = 0;
         commit.index = 0;
         commit.hnd   = 0;
         mpp_buffer_import(&inputbuf, &commit);
```

 但是这样的做法会导致encode出的图像是雪花图, 请问是哪里做错了吗?

另外我encoder的设置如下
```
    cmd->width      = 640;
    cmd->height     = 480;
    cmd->type       = MPP_VIDEO_CodingMJPEG;
    cmd->format     = MPP_FMT_BGRA8888;
    cmd->num_frames = 1;
    cmd->rc_mode    = MPP_ENC_RC_MODE_FIXQP;
```

HermanChen, 2020-11-30:
MppBufferInfo 里的 fd 0 是有问题的，需要一个从 ion 或者 drm 分配器分配出来的 dmafuf 才行。

xiayip, 2020-11-30:
谢谢您, 
也就是说外部申请的内存块是没办法转成MppBuffer的是吗?
必须得通过ion或是drm分配器分配, 然后再拷贝过去.

HermanChen, 2020-11-30:
不是，是说这种 cpu malloc 出来的 buffer 没法直接给硬件编码器使用，需要做一次拷贝。

xiayip, 2020-11-30:
明白了, 谢谢