rockchip-linux/mpp #166: decode_get_frame - frame is NULL

Nian8113, 2020-11-25:
您好，RK3399 Android MediaCodec本身沒有支持MJPEG格式，所以我想要用MPP來做。
build好android arm後，參考RKMediaCodecDemo把MPP包成Android library，
[https://github.com/c-xh/RKMediaCodecDemo](url)

我試著先做H.264格式，
因為Android 用MediaCodec是正常的，
但把MediaCodec換成MPP後，
decode_get_frame返回是成功的，
可是frame一直是空的，
有確定decode_put_packet是正確完成的。
請問是我有漏掉什麼或是哪邊出問題呢?

Thanks.

HermanChen, 2020-11-25:
需要确认下输入是否是 H.264 祼码流，即没有外部封装的码流。
可以使用 mpi_dec_test 在平台上跑一下看看有没有输出。

Nian8113, 2020-11-26:
Android怎麼把mpi_dec_test放進去執行?
因為我用adb push進去要執行都返回"can't execute: Permission denied"。
還是新手不太了解...

HermanChen, 2020-11-26:
adb root
adb remount
还有找一下 chmod 的使用

Nian8113, 2020-11-26:
可以執行了但是都沒有回饋訊息，
連我執行mpp_info_test都沒有回饋，
我把libvpu, libmpp放到/system/lib
mpi_dec_test, mpp_info_test放到/system/bin

HermanChen, 2020-11-26:
logcat -c && logcat -v time &
然后再运行这些测试用例

Nian8113, 2020-11-26:
執行後出現錯誤:

> CP-B:/system/bin # ./mpi_dec_test -i ../output_h264.mp4 -o ../out_h264_test.NV12 -w 1920 -h 1080 -t 4 -d 1
> 11-26 13:23:11.351 I/mpi_dec_test( 7381): cmd parse result:
> 11-26 13:23:11.351 I/mpi_dec_test( 7381): input  file name: ../output_h264.mp4
> 11-26 13:23:11.351 I/mpi_dec_test( 7381): output file name: ../out_h264_test.NV12
> 11-26 13:23:11.351 I/mpi_dec_test( 7381): config file name:
> 11-26 13:23:11.351 I/mpi_dec_test( 7381): width      : 1920
> 11-26 13:23:11.351 I/mpi_dec_test( 7381): height     : 1080
> 11-26 13:23:11.351 I/mpi_dec_test( 7381): type       : 4
> 11-26 13:23:11.351 I/mpi_dec_test( 7381): debug flag : 1
> 11-26 13:23:11.351 I/mpi_dec_test( 7381): max frames : 0
> 11-26 13:23:11.352 I/mpi_dec_test( 7381): mpi_dec_test start
> 11-26 13:23:11.352 I/mpi_dec_test( 7381): input file size 2486407
> 11-26 13:23:11.352 I/mpi     ( 7381): mpp_create enter ctx 0xffe65798 mpi 0xffe6579c
> 11-26 13:23:11.352 I/mpp_info( 7381): mpp version: 998f9a29 author: Herman Chen   2020-08-31 [mpp_enc_v2]: Add more log for pts debug
> 11-26 13:23:11.352 I/mpi     ( 7381): mpp_create leave ret 0 ctx 0xecf19000 mpi 0xed1ff004
> 11-26 13:23:11.352 I/mpi_dec_test( 7381): 0xecf19000 mpi_dec_test decoder test start w 1920 h 1080 type 4
> 11-26 13:23:11.352 I/mpi     ( 7381): mpi_control enter ctx 0xecf19000 cmd 310005 parm 0xffe657a8
> 11-26 13:23:11.352 I/mpi     ( 7381): mpi_control leave ret 0
> 11-26 13:23:11.352 I/mpi     ( 7381): mpp_init enter ctx 0xecf19000 type 0 coding 4
> 11-26 13:23:11.352 I/mpp_rt  ( 7381): NOT found ion allocator
> 11-26 13:23:11.352 I/mpp_rt  ( 7381): found drm allocator
> 11-26 13:23:11.353 I/mpi     ( 7381): mpp_init leave ret 0
> 11-26 13:23:11.353 I/mpi     ( 7381): mpi_decode_put_packet enter ctx 0xecf19000 packet 0xecf17040
> 11-26 13:23:11.353 I/mpi     ( 7381): mpi_decode_put_packet leave ret 0
> 11-26 13:23:11.353 I/mpi     ( 7381): mpi_decode_get_frame enter ctx 0xecf19000 frame 0xffe657b8
> 11-26 13:23:11.354 I/mpi     ( 7381): mpi_decode_get_frame leave ret 0
> 11-26 13:23:11.354 I/mpi     ( 7381): mpi_decode_put_packet enter ctx 0xecf19000 packet 0xecf17040
> 11-26 13:23:11.354 I/mpi     ( 7381): mpi_decode_put_packet leave ret 0
> 11-26 13:23:11.354 I/mpi     ( 7381): mpi_decode_get_frame enter ctx 0xecf19000 frame 0xffe657b8
> 11-26 13:23:11.355 I/mpi     ( 7381): mpi_decode_get_frame leave ret 0
> ...
> 11-26 13:23:12.078 I/mpi     ( 7381): mpi_decode_put_packet enter ctx 0xecf19000 packet 0xecf17040
> 11-26 13:23:12.078 I/mpi     ( 7381): mpi_decode_put_packet leave ret 0
> 11-26 13:23:12.078 I/mpi     ( 7381): mpi_decode_get_frame enter ctx 0xecf19000 frame 0xffe657b8
> 11-26 13:23:12.079 I/mpi     ( 7381): mpi_decode_get_frame leave ret 0
> 11-26 13:23:12.079 I/mpi_dec_test( 7381): 0xecf19000 get error and check frame_num
> 11-26 13:23:12.079 I/mpi_dec_test( 7381): 0xecf19000 found last packet
> 11-26 13:23:12.079 I/mpi     ( 7381): mpi_decode_put_packet enter ctx 0xecf19000 packet 0xecf17040
> 11-26 13:23:12.079 I/mpi     ( 7381): mpi_decode_put_packet leave ret 0
> 11-26 13:23:12.079 I/mpi     ( 7381): mpi_decode_get_frame enter ctx 0xecf19000 frame 0xffe657b8
> 11-26 13:23:12.080 I/mpi     ( 7381): mpi_decode_get_frame leave ret 0
> 11-26 13:23:12.090 I/mpi     ( 7381): mpi_decode_get_frame enter ctx 0xecf19000 frame 0xffe657b8
> 11-26 13:23:12.091 I/mpi     ( 7381): mpi_decode_get_frame leave ret 0
> 11-26 13:23:12.101 I/mpi     ( 7381): mpi_decode_get_frame enter ctx 0xecf19000 frame 0xffe657b8
> 11-26 13:23:12.102 I/mpi     ( 7381): mpi_decode_get_frame leave ret 0
> 11-26 13:23:12.112 E/mpg4d_parser( 7381): GMC is not supported
> 11-26 13:23:12.113 I/mpi     ( 7381): mpi_decode_get_frame enter ctx 0xecf19000 frame 0xffe657b8
> 11-26 13:23:12.113 I/mpi     ( 7381): mpi_decode_get_frame leave ret 0
> 11-26 13:23:12.113 I/mpi_dec_test( 7381): 0xecf19000 decode get frame 0
> 11-26 13:23:12.113 I/mpi_dec_test( 7381): 0xecf19000 found last packet
> 11-26 13:23:12.113 I/mpi     ( 7381): mpi_reset enter ctx 0xecf19000
> 11-26 13:23:12.113 I/mpi     ( 7381): mpi_reset leave ret 0
> 11-26 13:23:12.113 I/mpi     ( 7381): mpp_destroy enter ctx 0xecf19000
> 11-26 13:23:12.117 I/mpi     ( 7381): mpp_destroy leave ret 0
> 11-26 13:23:12.117 I/mpi_dec_test( 7381): test success max memory 0.00 MB

請問會是哪邊弄錯?

HermanChen, 2020-11-26:
不能输入 output_h264.mp4，这个是带封装的，需要解封装之后的 264 祼码流，就是带 00 00 00 01 NAL 头的数据包。

Pony23333, 2020-11-26:
用ffmpeg可以提取裸流
ffmpeg -i video.mp4 -codec copy -f h264 video.h264

Nian8113, 2020-11-26:
@Pony23333 謝謝提供的方法

@HermanChen  
有把輸入改成解封裝的264裸碼流了，但依舊是一樣的錯誤

> CP-B:/system/bin # ./mpi_dec_test -i ../output_h264.h264 -o ../out_h264_test.NV12 -w 1920 -h 1080 -t 4 -d 1
> 11-26 16:29:20.064 I/mpi_dec_test(13278): cmd parse result:
> 11-26 16:29:20.064 I/mpi_dec_test(13278): input  file name: ../output_h264.h264
> 11-26 16:29:20.064 I/mpi_dec_test(13278): output file name: ../out_h264_test.NV12
> 11-26 16:29:20.064 I/mpi_dec_test(13278): config file name:
> 11-26 16:29:20.064 I/mpi_dec_test(13278): width      : 1920
> 11-26 16:29:20.064 I/mpi_dec_test(13278): height     : 1080
> 11-26 16:29:20.064 I/mpi_dec_test(13278): type       : 4
> 11-26 16:29:20.064 I/mpi_dec_test(13278): debug flag : 1
> 11-26 16:29:20.064 I/mpi_dec_test(13278): max frames : 0
> 11-26 16:29:20.065 I/mpi_dec_test(13278): mpi_dec_test start
> 11-26 16:29:20.065 I/mpi_dec_test(13278): input file size 2481479
> 11-26 16:29:20.065 I/mpi     (13278): mpp_create enter ctx 0xffc2a4a8 mpi 0xffc2a4ac
> 11-26 16:29:20.065 I/mpp_info(13278): mpp version: 998f9a29 author: Herman Chen   2020-08-31 [mpp_enc_v2]: Add more log for pts debug
> 11-26 16:29:20.065 I/mpi     (13278): mpp_create leave ret 0 ctx 0xf6f19000 mpi 0xf724c004
> 11-26 16:29:20.065 I/mpi_dec_test(13278): 0xf6f19000 mpi_dec_test decoder test start w 1920 h 1080 type 4
> 11-26 16:29:20.065 I/mpi     (13278): mpi_control enter ctx 0xf6f19000 cmd 310005 parm 0xffc2a4b8
> 11-26 16:29:20.065 I/mpi     (13278): mpi_control leave ret 0
> 11-26 16:29:20.065 I/mpi     (13278): mpp_init enter ctx 0xf6f19000 type 0 coding 4
> 11-26 16:29:20.065 I/mpp_rt  (13278): NOT found ion allocator
> 11-26 16:29:20.065 I/mpp_rt  (13278): found drm allocator
> 11-26 16:29:20.066 I/mpi     (13278): mpp_init leave ret 0
> 11-26 16:29:20.066 I/mpi     (13278): mpi_decode_put_packet enter ctx 0xf6f19000 packet 0xf6f17040
> 11-26 16:29:20.066 I/mpi     (13278): mpi_decode_put_packet leave ret 0
> 11-26 16:29:20.066 I/mpi     (13278): mpi_decode_get_frame enter ctx 0xf6f19000 frame 0xffc2a4c8
> 11-26 16:29:20.067 I/mpi     (13278): mpi_decode_get_frame leave ret 0
> ...
> 11-26 16:29:20.791 I/mpi     (13278): mpi_decode_put_packet enter ctx 0xf6f19000 packet 0xf6f17040
> 11-26 16:29:20.791 I/mpi     (13278): mpi_decode_put_packet leave ret 0
> 11-26 16:29:20.791 I/mpi     (13278): mpi_decode_get_frame enter ctx 0xf6f19000 frame 0xffc2a4c8
> 11-26 16:29:20.792 I/mpi     (13278): mpi_decode_get_frame leave ret 0
> 11-26 16:29:20.792 I/mpi_dec_test(13278): 0xf6f19000 get error and check frame_num
> 11-26 16:29:20.792 I/mpi_dec_test(13278): 0xf6f19000 found last packet
> 11-26 16:29:20.792 I/mpi     (13278): mpi_decode_put_packet enter ctx 0xf6f19000 packet 0xf6f17040
> 11-26 16:29:20.792 I/mpi     (13278): mpi_decode_put_packet leave ret 0
> 11-26 16:29:20.792 I/mpi     (13278): mpi_decode_get_frame enter ctx 0xf6f19000 frame 0xffc2a4c8
> 11-26 16:29:20.793 I/mpi     (13278): mpi_decode_get_frame leave ret 0
> 11-26 16:29:20.804 I/mpi     (13278): mpi_decode_get_frame enter ctx 0xf6f19000 frame 0xffc2a4c8
> 11-26 16:29:20.805 I/mpi     (13278): mpi_decode_get_frame leave ret 0
> 11-26 16:29:20.815 I/mpi     (13278): mpi_decode_get_frame enter ctx 0xf6f19000 frame 0xffc2a4c8
> 11-26 16:29:20.816 I/mpi     (13278): mpi_decode_get_frame leave ret 0
> 11-26 16:29:20.826 I/mpi     (13278): mpi_decode_get_frame enter ctx 0xf6f19000 frame 0xffc2a4c8
> 11-26 16:29:20.827 I/mpi     (13278): mpi_decode_get_frame leave ret 0
> 11-26 16:29:20.838 I/mpi     (13278): mpi_decode_get_frame enter ctx 0xf6f19000 frame 0xffc2a4c8
> 11-26 16:29:20.839 I/mpi     (13278): mpi_decode_get_frame leave ret 0
> 11-26 16:29:20.849 I/mpi     (13278): mpi_decode_get_frame enter ctx 0xf6f19000 frame 0xffc2a4c8
> 11-26 16:29:20.850 I/mpi     (13278): mpi_decode_get_frame leave ret 0
> 11-26 16:29:20.860 I/mpi     (13278): mpi_decode_get_frame enter ctx 0xf6f19000 frame 0xffc2a4c8
> 11-26 16:29:20.861 I/mpi     (13278): mpi_decode_get_frame leave ret 0
> 11-26 16:29:20.872 I/mpi     (13278): mpi_decode_get_frame enter ctx 0xf6f19000 frame 0xffc2a4c8
> 11-26 16:29:20.872 I/mpi     (13278): mpi_decode_get_frame leave ret 0
> 11-26 16:29:20.872 I/mpi_dec_test(13278): 0xf6f19000 decode get frame 0
> 11-26 16:29:20.872 I/mpi_dec_test(13278): 0xf6f19000 found last packet
> 11-26 16:29:20.872 I/mpi     (13278): mpi_reset enter ctx 0xf6f19000
> 11-26 16:29:20.872 I/mpi     (13278): mpi_reset leave ret 0
> 11-26 16:29:20.872 I/mpi     (13278): mpp_destroy enter ctx 0xf6f19000
> 11-26 16:29:20.876 I/mpi     (13278): mpp_destroy leave ret 0
> 11-26 16:29:20.876 I/mpi_dec_test(13278): test success max memory 0.00 MB

還有可能是我哪邊弄錯?

HermanChen, 2020-11-26:
-t 4 参数错误，H.264 解码的类型是 -t 7，或者不要 -t 参数，让 mpi_dec_test 自动识别

Nian8113, 2020-11-26:
有了!! 但還是出現get error and check frame_num。
回到最一開始的問題，
decode_get_frame後的fream NULL，
是因為輸入不是H.264裸碼流嗎?

> CP-B:/system/bin # ./mpi_dec_test -i ../output_h264.h264 -o ../out_h264_test.NV12 -w 1920 -h 1080 -t 7
> 11-26 17:00:28.093 I/mpi_dec_test(14052): cmd parse result:
> 11-26 17:00:28.093 I/mpi_dec_test(14052): input  file name: ../output_h264.h264
> 11-26 17:00:28.093 I/mpi_dec_test(14052): output file name: ../out_h264_test.NV12
> 11-26 17:00:28.093 I/mpi_dec_test(14052): config file name:
> 11-26 17:00:28.093 I/mpi_dec_test(14052): width      : 1920
> 11-26 17:00:28.093 I/mpi_dec_test(14052): height     : 1080
> 11-26 17:00:28.093 I/mpi_dec_test(14052): type       : 7
> 11-26 17:00:28.093 I/mpi_dec_test(14052): debug flag : 0
> 11-26 17:00:28.093 I/mpi_dec_test(14052): max frames : 0
> 11-26 17:00:28.094 I/mpi_dec_test(14052): mpi_dec_test start
> 11-26 17:00:28.094 I/mpi_dec_test(14052): input file size 2481479
> 11-26 17:00:28.347 I/mpp_info(14052): mpp version: 998f9a29 author: Herman Chen   2020-08-31 [mpp_enc_v2]: Add more log for pts debug
> 11-26 17:00:28.347 I/mpi_dec_test(14052): 0xf5199000 mpi_dec_test decoder test start w 1920 h 1080 type 7
> 11-26 17:00:28.347 I/mpp_rt  (14052): NOT found ion allocator
> 11-26 17:00:28.347 I/mpp_rt  (14052): found drm allocator
> 11-26 17:00:28.384 I/mpi_dec_test(14052): 0xf5199000 decode_get_frame get info changed found
> 11-26 17:00:28.384 I/mpi_dec_test(14052): 0xf5199000 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 4177920
> 11-26 17:00:28.396 I/mpi_dec_test(14052): 0xf5199000 decode get frame 0
> 11-26 17:00:28.415 I/mpi_dec_test(14052): 0xf5199000 decode get frame 1
> ...
> 11-26 17:00:34.414 I/mpi_dec_test(14052): 0xf5199000 decode get frame 292
> 11-26 17:00:34.426 I/mpi_dec_test(14052): 0xf5199000 decode get frame 293
> 11-26 17:00:34.496 I/mpi_dec_test(14052): 0xf5199000 get error and check frame_num
> 11-26 17:00:34.496 I/mpi_dec_test(14052): 0xf5199000 found last packet
> 11-26 17:00:34.508 I/mpi_dec_test(14052): 0xf5199000 decode get frame 294
> 11-26 17:00:34.557 I/mpi_dec_test(14052): 0xf5199000 decode get frame 295
> 11-26 17:00:34.579 I/mpi_dec_test(14052): 0xf5199000 decode get frame 296
> 11-26 17:00:34.601 I/mpi_dec_test(14052): 0xf5199000 decode get frame 297
> 11-26 17:00:34.623 I/mpi_dec_test(14052): 0xf5199000 decode get frame 298 err 1 discard 0
> 11-26 17:00:34.623 I/mpi_dec_test(14052): 0xf5199000 found last packet
> 11-26 17:00:34.646 I/mpi_dec_test(14052): test success max memory 23.91 MB

Pony23333, 2020-11-26:
我的理解是，设置-t的化就按照你设置的格式去解码了，喂的数据格式不对得到的frame是NULL很正常，不过得看源码了。get error and check frame_num 是不是拼接这一帧的时候出错了啊，源码是按照split_mode以4k大小去读buffer然后拼接的。但是这个实际应该还是可以接受的。

HermanChen, 2020-11-26:
mpi_dec_test -i ../output_h264.h264 -o ../out_h264_test.NV12
这样即可，不需要给宽高，解码器会解析图像大小
只有 jpeg 解码时，需要预先分配空间，才需要在解码时配置宽高信息

Nian8113, 2020-11-26:
我使用的影片來源: https://test-videos.co.uk/bigbuckbunny/mp4-h264
用yuvplayer播放發現第296張有破圖的感覺

> CP-B:/system/bin # ./mpi_dec_test -i ../out_test.h264 -o ../out_test.NV12
> 11-26 17:45:52.892 I/mpi_dec_test(15558): cmd parse result:
> 11-26 17:45:52.892 I/mpi_dec_test(15558): input  file name: ../out_test.h264
> 11-26 17:45:52.892 I/mpi_dec_test(15558): output file name: ../out_test.NV12
> 11-26 17:45:52.892 I/mpi_dec_test(15558): config file name:
> 11-26 17:45:52.892 I/mpi_dec_test(15558): width      :    0
> 11-26 17:45:52.893 I/mpi_dec_test(15558): height     :    0
> 11-26 17:45:52.893 I/mpi_dec_test(15558): type       : 7
> 11-26 17:45:52.893 I/mpi_dec_test(15558): debug flag : 0
> 11-26 17:45:52.893 I/mpi_dec_test(15558): max frames : 0
> 11-26 17:45:52.893 I/mpi_dec_test(15558): mpi_dec_test start
> 11-26 17:45:52.893 I/mpi_dec_test(15558): input file size 1957852
> 11-26 17:45:52.893 I/mpp_info(15558): mpp version: 998f9a29 author: Herman Chen   2020-08-31 [mpp_enc_v2]: Add more log for pts debug
> 11-26 17:45:52.893 I/mpi_dec_test(15558): 0xee099000 mpi_dec_test decoder test start w 0 h 0 type 7
> 11-26 17:45:52.893 I/mpp_rt  (15558): NOT found ion allocator
> 11-26 17:45:52.893 I/mpp_rt  (15558): found drm allocator
> 11-26 17:45:52.945 I/mpi_dec_test(15558): 0xee099000 decode_get_frame get info changed found
> 11-26 17:45:52.945 I/mpi_dec_test(15558): 0xee099000 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 4177920
> 11-26 17:45:52.959 I/mpi_dec_test(15558): 0xee099000 decode get frame 0
> 11-26 17:45:53.011 I/mpi_dec_test(15558): 0xee099000 decode get frame 1
> ...
> 11-26 17:45:59.331 I/mpi_dec_test(15558): 0xee099000 decode get frame 295
> 11-26 17:45:59.366 I/mpi_dec_test(15558): 0xee099000 get error and check frame_num
> 11-26 17:45:59.366 I/mpi_dec_test(15558): 0xee099000 found last packet
> 11-26 17:45:59.389 I/mpi_dec_test(15558): 0xee099000 decode get frame 296 err 0 discard 1
> 11-26 17:45:59.411 I/mpi_dec_test(15558): 0xee099000 decode get frame 297
> 11-26 17:45:59.439 I/mpi_dec_test(15558): 0xee099000 decode get frame 298
> 11-26 17:45:59.463 I/mpi_dec_test(15558): 0xee099000 found last packet
> 11-26 17:45:59.520 I/mpi_dec_test(15558): test success max memory 75.70 MB

HermanChen, 2020-11-26:
试用 ffmpeg 解码成 nv12 格式，然后对比下区别，也观察下 ffmpeg 解码是否有报错，有可能码流本身在这一帧就有错误

Nian8113, 2020-11-30:
Hi HermanChen,
試著把我所使用影像來源(TCP傳輸H264)存下來，
用mpi_dec_test 轉成NV12是成功的。
但是參考RKMediaCodecDemo把MPP包成Android library，
[https://github.com/c-xh/RKMediaCodecDemo](url)

卻依舊是相同問題，
> decode_get_frame返回是成功的，
> 可是frame一直是空的，

還有可能是什麼原因造成的呢?
謝謝...

HermanChen, 2020-11-30:
估计你保存的 264 码流里没有 sps 和 pps

Nian8113, 2020-12-01:
Hi Herman,
請教一下如果要測試mpp解mjpeg該怎麼下，
我嘗試使用如下:

> CP-B:/system/bin # ./mpi_dec_test -i ../output_mjpeg.mp4 -o ../output_mjpeg.NV12 -w 1920 -h 1080 -t 8
> 12-01 14:02:57.174 I/mpi_dec_test( 2703): cmd parse result:
> 12-01 14:02:57.174 I/mpi_dec_test( 2703): input  file name: ../output_mjpeg.mp4
> 12-01 14:02:57.174 I/mpi_dec_test( 2703): output file name: ../output_mjpeg.NV12
> 12-01 14:02:57.174 I/mpi_dec_test( 2703): config file name:
> 12-01 14:02:57.174 I/mpi_dec_test( 2703): width      : 1920
> 12-01 14:02:57.174 I/mpi_dec_test( 2703): height     : 1080
> 12-01 14:02:57.174 I/mpi_dec_test( 2703): type       : 8
> 12-01 14:02:57.174 I/mpi_dec_test( 2703): debug flag : 0
> 12-01 14:02:57.174 I/mpi_dec_test( 2703): max frames : 0
> 12-01 14:02:57.174 I/mpi_dec_test( 2703): mpi_dec_test start
> 12-01 14:02:57.174 I/mpi_dec_test( 2703): input file size 20703083
> 12-01 14:02:57.174 I/mpp_rt  ( 2703): NOT found ion allocator
> 12-01 14:02:57.174 I/mpp_rt  ( 2703): found drm allocator
> 12-01 14:02:57.202 I/mpp_info( 2703): mpp version: 998f9a29 author: Herman Chen   2020-08-31 [mpp_enc_v2]: Add more log for pts debug
> 12-01 14:02:57.202 I/mpi_dec_test( 2703): 0xee899420 mpi_dec_test decoder test start w 1920 h 1080 type 8
> 12-01 14:02:57.270 E/mpp_dec ( 2703): mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 12-01 14:02:57.270 I/mpi_dec_test( 2703): 0xee899420 decoded frame 0
> 12-01 14:02:57.271 I/mpi_dec_test( 2703): 0xee899420 found last packet
> 12-01 14:02:57.271 I/mpi_dec_test( 2703): 0xee899420 decoded frame 1
> 12-01 14:02:57.271 I/mpi_dec_test( 2703): 0xee899420 found eos frame
> 12-01 14:02:57.282 I/mpi_dec_test( 2703): test success max memory 0.00 MB

這會是哪邊出問題?
謝謝

HermanChen, 2020-12-01:
-i ../output_mjpeg.mp4 ？还是用 mp4 文件解码？

Nian8113, 2020-12-01:
還是是要用 output_mjpeg.mjpeg ?

HermanChen, 2020-12-01:
mpi_dec_test 的输入是祼码流，不是 mp4 文件……