rockchip-linux/mpp #167: mpp硬解码mjpeg和jpg文件出现timeout

Pony23333, 2020-11-26:
所用的mpp版本为github上最新的。测试的mjpeg为ffmpeg提取的裸流，debug模式下的输出可以解码jpg文件时input 的pkt数量为0，mpi_poll一直timeout。而解码mjpeg文件的时候除了timeout还有一个mpp_parser_parse的问题。请问是advanced模式下的哪里的问题？还是我设置的参数或者输入的文件可能有问题。

祝好

toybrick@debian10:~/Desktop/mpp/mpp/build/linux/aarch64/test$ ./mpi_dec_test -i ~/Desktop/test_videos/factory.mjpeg -t 8 -w 544 -h 960 -d 1
mpp[7595]: mpi_dec_test: cmd parse result:
mpp[7595]: mpi_dec_test: input  file name: /home/toybrick/Desktop/test_videos/factory.mjpeg
mpp[7595]: mpi_dec_test: output file name: 
mpp[7595]: mpi_dec_test: config file name: 
mpp[7595]: mpi_dec_test: width      :  544
mpp[7595]: mpi_dec_test: height     :  960
mpp[7595]: mpi_dec_test: type       : 8
mpp[7595]: mpi_dec_test: debug flag : 1
mpp[7595]: mpi_dec_test: max frames : 0
mpp[7595]: mpi_dec_test: mpi_dec_test start
mpp[7595]: mpi_dec_test: input file size 2522594
mpp[7595]: mpp_rt: NOT found ion allocator
mpp[7595]: mpp_rt: found drm allocator
mpp[7595]: mpi: mpp_create enter ctx 0x7ff8a49588 mpi 0x7ff8a49590
mpp[7595]: mpp_info: mpp version: b845d774 author: Herman Chen   2020-10-28 [mpp_device]: Fix cmd_butt check error
mpp[7595]: mpi: mpp_create leave ret 0 ctx 0x559ad95ec0 mpi 0x7fad4e7a38
mpp[7595]: mpi_dec_test: 0x559ad95ec0 mpi_dec_test decoder test start w 544 h 960 type 8
mpp[7595]: mpi: mpi_control enter ctx 0x559ad95ec0 cmd 310005 parm 0x7ff8a49580
mpp[7595]: mpi: mpi_control leave ret 0
mpp[7595]: mpi: mpp_init enter ctx 0x559ad95ec0 type 0 coding 8
mpp[7595]: mpi: mpp_init leave ret 0
mpp[7595]: mpi: mpi_poll enter ctx 0x559ad95ec0 type 0 timeout -1
mpp[7595]: mpi: mpi_poll leave ret 0
mpp[7595]: mpi: mpi_dequeue enter ctx 0x559ad95ec0 type 0 task 0x7ff8a495c0
mpp[7595]: mpi: mpi_dequeue leave ret 0
mpp[7595]: mpi: mpi_enqueue enter ctx 0x559ad95ec0 type 0 task 0x559ad945e0
mpp[7595]: mpi: mpi_enqueue leave ret 0
mpp[7595]: mpi: mpi_poll enter ctx 0x559ad95ec0 type 1 timeout -1
mpp[7595]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
mpp[7595]: mpi: mpi_poll leave ret 0
mpp[7595]: mpi: mpi_dequeue enter ctx 0x559ad95ec0 type 1 task 0x7ff8a495c0
mpp[7595]: mpi: mpi_dequeue leave ret 0
mpp[7595]: mpi_dec_test: 0x559ad95ec0 decoded frame 0
mpp[7595]: mpi: mpi_enqueue enter ctx 0x559ad95ec0 type 1 task 0x559ad94740
mpp[7595]: mpi: mpi_enqueue leave ret 0
mpp[7595]: mpi_dec_test: 0x559ad95ec0 found last packet
mpp[7595]: mpi: mpi_poll enter ctx 0x559ad95ec0 type 0 timeout -1
mpp[7595]: mpi: mpi_poll leave ret 0
mpp[7595]: mpi: mpi_dequeue enter ctx 0x559ad95ec0 type 0 task 0x7ff8a495c0
mpp[7595]: mpi: mpi_dequeue leave ret 0
mpp[7595]: mpi: mpi_enqueue enter ctx 0x559ad95ec0 type 0 task 0x559ad945e0
mpp[7595]: mpi: mpi_enqueue leave ret 0
mpp[7595]: mpi: mpi_poll enter ctx 0x559ad95ec0 type 1 timeout -1
mpp[7595]: mpi: mpi_poll leave ret 0
mpp[7595]: mpi: mpi_dequeue enter ctx 0x559ad95ec0 type 1 task 0x7ff8a495c0
mpp[7595]: mpi: mpi_dequeue leave ret 0
mpp[7595]: mpi_dec_test: 0x559ad95ec0 decoded frame 1
mpp[7595]: mpi_dec_test: 0x559ad95ec0 found eos frame
mpp[7595]: mpi: mpi_enqueue enter ctx 0x559ad95ec0 type 1 task 0x559ad94740
mpp[7595]: mpi: mpi_enqueue leave ret 0
mpp[7595]: mpi: mpi_control enter ctx 0x559ad95ec0 cmd 310101 parm 0x7ff8a49650
mpp[7595]: mpi: mpi_control leave ret 0
mpp[7595]: mpi_dec_test: 0x559ad95ec0 input 0 pkt output 0 frm decode 0 frames
mpp[7595]: mpi: mpi_reset enter ctx 0x559ad95ec0
mpp[7595]: mpi: mpi_reset leave ret 0
mpp[7595]: mpi: mpp_destroy enter ctx 0x559ad95ec0
mpp[7595]: mpi: mpp_destroy leave ret 0
mpp[7595]: mpi_dec_test: test success max memory 0.00 MB

toybrick@debian10:~/Desktop/mpp/mpp/build/linux/aarch64/test$ ./mpi_dec_test -i ~/Desktop/test_videos/001.jpg -t 8 -w 1280 -h 720 -f 2  -d 1
mpp[7669]: mpi_dec_test: cmd parse result:
mpp[7669]: mpi_dec_test: input  file name: /home/toybrick/Desktop/test_videos/001.jpg
mpp[7669]: mpi_dec_test: output file name: 
mpp[7669]: mpi_dec_test: config file name: 
mpp[7669]: mpi_dec_test: width      : 1280
mpp[7669]: mpi_dec_test: height     :  720
mpp[7669]: mpi_dec_test: type       : 8
mpp[7669]: mpi_dec_test: debug flag : 1
mpp[7669]: mpi_dec_test: max frames : 0
mpp[7669]: mpi_dec_test: mpi_dec_test start
mpp[7669]: mpi_dec_test: input file size 56040
mpp[7669]: mpp_rt: NOT found ion allocator
mpp[7669]: mpp_rt: found drm allocator
mpp[7669]: mpi: mpp_create enter ctx 0x7fd4587a48 mpi 0x7fd4587a50
mpp[7669]: mpp_info: mpp version: b845d774 author: Herman Chen   2020-10-28 [mpp_device]: Fix cmd_butt check error
mpp[7669]: mpi: mpp_create leave ret 0 ctx 0x559da0bec0 mpi 0x7fb7ab4a38
mpp[7669]: mpi_dec_test: 0x559da0bec0 mpi_dec_test decoder test start w 1280 h 720 type 8
mpp[7669]: mpi: mpi_control enter ctx 0x559da0bec0 cmd 310005 parm 0x7fd4587a40
mpp[7669]: mpi: mpi_control leave ret 0
mpp[7669]: mpi: mpp_init enter ctx 0x559da0bec0 type 0 coding 8
mpp[7669]: mpi: mpp_init leave ret 0
mpp[7669]: mpi: mpi_control enter ctx 0x559da0bec0 cmd 31000a parm 0x7fd4587f74
mpp[7669]: mpi: mpi_control leave ret 0
mpp[7669]: mpi: mpi_poll enter ctx 0x559da0bec0 type 0 timeout -1
mpp[7669]: mpi: mpi_poll leave ret 0
mpp[7669]: mpi: mpi_dequeue enter ctx 0x559da0bec0 type 0 task 0x7fd4587a80
mpp[7669]: mpi: mpi_dequeue leave ret 0
mpp[7669]: mpi: mpi_enqueue enter ctx 0x559da0bec0 type 0 task 0x559da0a5e0
mpp[7669]: mpi: mpi_enqueue leave ret 0
mpp[7669]: mpi: mpi_poll enter ctx 0x559da0bec0 type 1 timeout -1
mpp[7669]: mpi: mpi_poll leave ret 0
mpp[7669]: mpi: mpi_dequeue enter ctx 0x559da0bec0 type 1 task 0x7fd4587a80
mpp[7669]: mpi: mpi_dequeue leave ret 0
mpp[7669]: mpi_dec_test: 0x559da0bec0 decoded frame 0
mpp[7669]: mpi: mpi_enqueue enter ctx 0x559da0bec0 type 1 task 0x559da0a740
mpp[7669]: mpi: mpi_enqueue leave ret 0
mpp[7669]: mpi_dec_test: 0x559da0bec0 found last packet
mpp[7669]: mpi: mpi_poll enter ctx 0x559da0bec0 type 0 timeout -1
mpp[7669]: mpi: mpi_poll leave ret 0
mpp[7669]: mpi: mpi_dequeue enter ctx 0x559da0bec0 type 0 task 0x7fd4587a80
mpp[7669]: mpi: mpi_dequeue leave ret 0
mpp[7669]: mpi: mpi_enqueue enter ctx 0x559da0bec0 type 0 task 0x559da0a5e0
mpp[7669]: mpi: mpi_enqueue leave ret 0
mpp[7669]: mpi: mpi_poll enter ctx 0x559da0bec0 type 1 timeout -1
mpp[7669]: mpi: mpi_poll leave ret 0
mpp[7669]: mpi: mpi_dequeue enter ctx 0x559da0bec0 type 1 task 0x7fd4587a80
mpp[7669]: mpi: mpi_dequeue leave ret 0
mpp[7669]: mpi_dec_test: 0x559da0bec0 decoded frame 1
mpp[7669]: mpi_dec_test: 0x559da0bec0 found eos frame
mpp[7669]: mpi: mpi_enqueue enter ctx 0x559da0bec0 type 1 task 0x559da0a740
mpp[7669]: mpi: mpi_enqueue leave ret 0
mpp[7669]: mpi: mpi_control enter ctx 0x559da0bec0 cmd 310101 parm 0x7fd4587b10
mpp[7669]: mpi: mpi_control leave ret 0
mpp[7669]: mpi_dec_test: 0x559da0bec0 input 0 pkt output 0 frm decode 0 frames
mpp[7669]: mpi: mpi_reset enter ctx 0x559da0bec0
mpp[7669]: mpi: mpi_reset leave ret 0
mpp[7669]: mpi: mpp_destroy enter ctx 0x559da0bec0
mpp[7669]: mpi: mpp_destroy leave ret 0
mpp[7669]: mpi_dec_test: test success max memory 0.00 MB

HermanChen, 2020-11-26:
是什么芯片平台？

Pony23333, 2020-11-26:
芯片是rk3399pro, 系统debian10, 板子是toybrick-3300proX

HermanChen, 2020-11-26:
麻烦把测试源提供下

Pony23333, 2020-11-26:
我看到了 https://github.com/rockchip-linux/mpp/issues/93 这个issue里的讨论，尝试了添加参数-f 0， 但是与上述帖子不同，我产生了新的错误：

toybrick@debian10:~/Desktop/mpp-develop/build/linux/aarch64/test$ ./mpi_dec_test -i ~/Desktop/test_videos/try_everything.jpg -t 8 -w 1920 -h 1080 -o frame.nv12 -f 0
mpp[3466]: mpi_dec_test: cmd parse result:
mpp[3466]: mpi_dec_test: input  file name: /home/toybrick/Desktop/test_videos/try_everything.jpg
mpp[3466]: mpi_dec_test: output file name: frame.nv12
mpp[3466]: mpi_dec_test: config file name: 
mpp[3466]: mpi_dec_test: width      : 1920
mpp[3466]: mpi_dec_test: height     : 1080
mpp[3466]: mpi_dec_test: type       : 8
mpp[3466]: mpi_dec_test: debug flag : 0
mpp[3466]: mpi_dec_test: max frames : 0
mpp[3466]: mpi_dec_test: mpi_dec_test start
mpp[3466]: mpi_dec_test: input file size 58823
mpp[3466]: mpp_rt: NOT found ion allocator
mpp[3466]: mpp_rt: found drm allocator
mpp[3466]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[3466]: mpi_dec_test: 0x55c03110a0 mpi_dec_test decoder test start w 1920 h 1080 type 8
mpp[3466]: mpp_device: mpp_dev_init found unsupported client type 3 in platform 40202
mpp[3466]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_gen_regs mpp_dev_init failed. ret: -6
mpp[3466]: mpp_device: mpp_dev_ioctl found NULL input ctx
mpp[3466]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start set register write failed -3
mpp[3466]: mpp_device: mpp_dev_ioctl found NULL input ctx
mpp[3466]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait poll cmd failed -3
mpp[3466]: mpi_dec_test: 0x55c03110a0 decoded frame 0
mpp[3466]: mpi_dec_test: 0x55c03110a0 found last packet
mpp[3466]: mpi_dec_test: 0x55c03110a0 decoded frame 1
mpp[3466]: mpi_dec_test: 0x55c03110a0 found eos frame
mpp[3466]: mpi_dec_test: 0x55c03110a0 input 0 pkt output 0 frm decode 0 frames
mpp[3466]: mpi_dec_test: test success max memory 0.00 MB

附上的两个文件分别是测试的jpg文件和提取的mjpeg裸流。
[D.zip](https://github.com/rockchip-linux/mpp/files/5601456/D.zip)

Pony23333, 2020-11-27:
测试用的jpg文件和mjpeg文件在上面的回复里贴了，我在这里再赋值一下吧
[test_files.zip](https://github.com/rockchip-linux/mpp/files/5606220/test_files.zip)

MinZhaoLin, 2020-12-08:
我这边使用的RK3399平台，用的release分支最新代码，也同样出错，如果编译的是以前很久的版本的话能正常解码：
/mpp-20201202/mpp-release/build/linux/aarch64/test$ ./mpi_dec_test -i test.jpg -w 1280 -h 960 -t 8 -n 1 -f 0 -o dec.yuv -d 1
mpp[2827]: mpi_dec_test: cmd parse result:
mpp[2827]: mpi_dec_test: input  file name: test.jpg
mpp[2827]: mpi_dec_test: output file name: dec.yuv
mpp[2827]: mpi_dec_test: config file name: 
mpp[2827]: mpi_dec_test: width      : 1280
mpp[2827]: mpi_dec_test: height     :  960
mpp[2827]: mpi_dec_test: type       : 8
mpp[2827]: mpi_dec_test: debug flag : 1
mpp[2827]: mpi_dec_test: max frames : 1
mpp[2827]: mpi_dec_test: mpi_dec_test start
mpp[2827]: mpi_dec_test: input file size 321065
mpp[2827]: mpp_rt: NOT found ion allocator
mpp[2827]: mpp_rt: NOT found drm allocator
mpp[2827]: mpi: mpp_create enter ctx 0x7fc99bdf38 mpi 0x7fc99bdf40
mpp[2827]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[2827]: mpi: mpp_create leave ret 0 ctx 0x558889bca0 mpi 0x7f8cf78ae8
mpp[2827]: mpi_dec_test: 0x558889bca0 mpi_dec_test decoder test start w 1280 h 960 type 8
mpp[2827]: mpi: mpi_control enter ctx 0x558889bca0 cmd 310005 parm 0x7fc99bdf30
mpp[2827]: mpi: mpi_control leave ret 0
mpp[2827]: mpi: mpp_init enter ctx 0x558889bca0 type 0 coding 8
mpp[2827]: mpi: mpp_init leave ret 0
mpp[2827]: mpi: mpi_control enter ctx 0x558889bca0 cmd 31000a parm 0x7fc99be464
mpp[2827]: mpi: mpi_control leave ret 0
mpp[2827]: mpi: mpi_poll enter ctx 0x558889bca0 type 0 timeout -1
mpp[2827]: mpi: mpi_poll leave ret 0
mpp[2827]: mpi: mpi_dequeue enter ctx 0x558889bca0 type 0 task 0x7fc99bdf70
mpp[2827]: mpi: mpi_dequeue leave ret 0
mpp[2827]: mpi: mpi_enqueue enter ctx 0x558889bca0 type 0 task 0x558889a340
mpp[2827]: mpi: mpi_enqueue leave ret 0
mpp[2827]: mpi: mpi_poll enter ctx 0x558889bca0 type 1 timeout -1
mpp[2827]: mpp_device: mpp_dev_init found unsupported client type 3 in platform 40202
mpp[2827]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_gen_regs mpp_dev_init failed. ret: -6
mpp[2827]: mpp_device: mpp_dev_ioctl found NULL input ctx
mpp[2827]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start set register write failed -3
mpp[2827]: mpp_device: mpp_dev_ioctl found NULL input ctx
mpp[2827]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait poll cmd failed -3
mpp[2827]: mpi: mpi_poll leave ret 0
mpp[2827]: mpi: mpi_dequeue enter ctx 0x558889bca0 type 1 task 0x7fc99bdf70
mpp[2827]: mpi: mpi_dequeue leave ret 0
mpp[2827]: mpi_dec_test: 0x558889bca0 decoded frame 0
mpp[2827]: mpi: mpi_enqueue enter ctx 0x558889bca0 type 1 task 0x558889a4a0
mpp[2827]: mpi: mpi_enqueue leave ret 0
mpp[2827]: mpi_dec_test: 0x558889bca0 found last packet
mpp[2827]: mpi: mpi_poll enter ctx 0x558889bca0 type 0 timeout -1
mpp[2827]: mpi: mpi_poll leave ret 0
mpp[2827]: mpi: mpi_dequeue enter ctx 0x558889bca0 type 0 task 0x7fc99bdf70
mpp[2827]: mpi: mpi_dequeue leave ret 0
mpp[2827]: mpi: mpi_enqueue enter ctx 0x558889bca0 type 0 task 0x558889a340
mpp[2827]: mpi: mpi_enqueue leave ret 0
mpp[2827]: mpi: mpi_poll enter ctx 0x558889bca0 type 1 timeout -1
mpp[2827]: mpi: mpi_poll leave ret 0
mpp[2827]: mpi: mpi_dequeue enter ctx 0x558889bca0 type 1 task 0x7fc99bdf70
mpp[2827]: mpi: mpi_dequeue leave ret 0
mpp[2827]: mpi_dec_test: 0x558889bca0 decoded frame 1
mpp[2827]: mpi_dec_test: 0x558889bca0 found eos frame
mpp[2827]: mpi: mpi_enqueue enter ctx 0x558889bca0 type 1 task 0x558889a4a0
mpp[2827]: mpi: mpi_enqueue leave ret 0
mpp[2827]: mpi: mpi_control enter ctx 0x558889bca0 cmd 310101 parm 0x7fc99be000
mpp[2827]: mpi: mpi_control leave ret 0
mpp[2827]: mpi_dec_test: 0x558889bca0 input 0 pkt output 0 frm decode 0 frames
mpp[2827]: mpi: mpi_reset enter ctx 0x558889bca0
mpp[2827]: mpi: mpi_reset leave ret 0
mpp[2827]: mpi: mpp_destroy enter ctx 0x558889bca0
mpp[2827]: mpi: mpp_destroy leave ret 0
mpp[2827]: mpi_dec_test: test success max memory 0.00 MB

虽然有输出，但是结果都是0

HermanChen, 2020-12-08:
测试时不要加 -d 1
运行 mpp_platform_test 看下 log
mpp_device: mpp_dev_init found unsupported client type 3 in platform 40202
看起来解码器 client type 的判断异常了

HermanChen, 2020-12-08:
另外可以测试下用 develop 分支的结果

MinZhaoLin, 2020-12-08:
mpp_platform_test的log如下：
mpp[8681]: mpp_plat_test: chip name: rockchip,rk3399-firefly-linux rockchip,rk3399
mpp[8681]: mpp_plat_test: 
mpp[8681]: mpp_plat_test: chip vcodec type 0004020a
mpp[8681]: mpp_plat_test: found vpu2 codec
mpp[8681]: mpp_plat_test: found rkvdec decoder
mpp[8681]: mpp_plat_test: found vpu2 encoder
mpp[8681]: mpp_plat_test: 
mpp[8681]: mpp_plat_test: start probing decoder device name:
mpp[8681]: mpp_plat_test: H.264 decoder: /dev/rkvdec
mpp[8681]: mpp_plat_test: H.265 decoder: /dev/rkvdec
mpp[8681]: mpp_plat_test: MJPEG decoder: /dev/vpu_service
mpp[8681]: mpp_plat_test: VP9   decoder: /dev/rkvdec
mpp[8681]: mpp_plat_test: avs   decoder: /dev/vpu_service
mpp[8681]: mpp_plat_test: 
mpp[8681]: mpp_plat_test: start probing encoder device name:
mpp[8681]: mpp_plat_test: H.264 encoder: /dev/vpu_service
mpp[8681]: mpp_plat_test: H.265 encoder: /dev/vpu_service
mpp[8681]: mpp_plat_test: MJPEG encoder: /dev/vpu_service
mpp[8681]: mpp_plat_test: mpp platform test done

使用了develop分支后，没有报client type的错，但是出现了其他错误：
mpp-develop/mpp-develop/build/linux/aarch64/test$ ./mpi_dec_test -i test.jpg -w 1280 -h 960 -t 8 -n 1 -f 0 -o dec.yuv 
mpp[8658]: mpi_dec_test: cmd parse result:
mpp[8658]: mpi_dec_test: input  file name: test.jpg
mpp[8658]: mpi_dec_test: output file name: dec.yuv
mpp[8658]: mpi_dec_test: config file name: 
mpp[8658]: mpi_dec_test: width      : 1280
mpp[8658]: mpi_dec_test: height     :  960
mpp[8658]: mpi_dec_test: type       : 8
mpp[8658]: mpi_dec_test: debug flag : 0
mpp[8658]: mpi_dec_test: max frames : 1
mpp[8658]: mpi_dec_test: mpi_dec_test start
mpp[8658]: mpi_dec_test: input file size 321065
mpp[8658]: mpp_rt: NOT found ion allocator
mpp[8658]: mpp_rt: NOT found drm allocator
mpp[8658]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[8658]: mpi_dec_test: 0x55ad0a6ca0 mpi_dec_test decoder test start w 1280 h 960 type 8
mpp[8658]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[8658]: vcodec_service: vcodec_service_cmd_send ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
mpp[8658]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start send cmd failed 14
mpp[8658]: vcodec_service: vcodec_service_cmd_poll ioctl VPU_IOC_GET_REG failed ret -1 errno 110 Connection timed out
mpp[8658]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait poll cmd failed 110
mpp[8658]: mpi_dec_test: 0x55ad0a6ca0 decoded frame 0
mpp[8658]: mpi_dec_test: 0x55ad0a6ca0 found last packet
mpp[8658]: mpi_dec_test: 0x55ad0a6ca0 decoded frame 1
mpp[8658]: mpi_dec_test: 0x55ad0a6ca0 found eos frame
mpp[8658]: mpi_dec_test: 0x55ad0a6ca0 input 0 pkt output 0 frm decode 0 frames
mpp[8658]: mpi_dec_test: test success max memory 0.00 MB
虽然有输出，但是结果都是0

Pony23333, 2020-12-09:
谢谢两位尝试@MinZhaoLin @HermanChen 
我试了解码jpg图像成功，但是对于MJPEG编码的视频怎么解决呢？逐帧解码吗

Pony23333, 2020-12-09:
另外补充一下，测试时带着-d 1的话还是会有timeout的问题