rockchip-linux/mpp #168: Configuration for 0x42e01f profile (as supported by default in browser)

jpiat, 2020-11-27:
Hi,

i'm trying to get the H264 output compatible with default supported profile in chrome/chromium but cannot get it work. I have set the  profile to 66 (for 0x42), but then whatever i change regarding h264:level  or other configuration the profile i get is 0x42c029.
Where can i affect the constrain flags ? How can i get a level lower than 40 for a 1080 stream ?

thanks

HermanChen, 2020-11-27:
https://github.com/rockchip-linux/mpp/blob/da9c6226940398246b05496e1ebad4bd9f46920c/mpp/codec/enc/h264/h264e_sps.c#L89-L109
Can you add mpp_log here and check your level_idc config for 1080p is correct or not.
It is better use new string config interface to setup level_idc.

jpiat, 2020-11-27:
ok, fomr what i see here, wahtever the lvel i set in the configuration, a minimum level is enforced using the frame size. The level is that last part of the profile, but what about the constraint flags ? Browser support 0x00 (no flags) and 0xe0 (all flags) but the mpp encoder default to 0xc0 and i can't seem to find how to affect this.

HermanChen, 2020-11-27:
https://github.com/rockchip-linux/mpp/blob/da9c6226940398246b05496e1ebad4bd9f46920c/mpp/codec/enc/h264/h264e_sps.c#L81-L87
You can change constraint flag here.