rockchip-linux/mpp #170: mpi_dec_test解码mjpeg视频文件只能解码一帧

Pony23333, 2020-12-11:
平台：Toybrick3399proX
系统：debian10

问题：mpi_dec_test解码mjpeg视频文件只能解码一帧。代码里面有对mjeg格式的设置，
`        // NOTE: for mjpeg decoding send the whole file
        if (type == MPP_VIDEO_CodingMJPEG) {
            packet_size = file_size;
        }
`
不是很理解为什么要将advanced_decode模式下的pkt_size设置成整个文件的大小，这样的化在下面的代码中直接将整个文件放入了packet中。
'
static int decode_advanced(MpiDecLoopData *data)
{
    RK_U32 pkt_eos  = 0;
    MPP_RET ret = MPP_OK;
    MppCtx ctx  = data->ctx;
    MppApi *mpi = data->mpi;
    char   *buf = data->buf;
    MppPacket packet = data->packet;
    MppFrame  frame  = data->frame;
    MppTask task = NULL;
    size_t read_size = fread(buf, 1, data->packet_size, data->fp_input);

    if (read_size != data->packet_size || feof(data->fp_input)) {
        mpp_log("%p found last packet\n", ctx);

        // setup eos flag
        data->eos = pkt_eos = 1;
    }
'
这样的化就有以下问题：
1. 如果按照以上方法解码mjpeg视频文件的时候，只能输出一帧，然后程序跳到文件末尾，解码结束。
2. 如果解码jpg图片的化，一次解码就结束了，为什么还在要设置循环呢？

所以，mpi_dec_test这个程序在设置的时候，-t 8 对应的就只是解码jpg文件吗？

WQuit, 2021-03-31:
你好问题有解决吗

HermanChen, 2021-04-01:
-t 是标记解码格式，-t 8 是 jpeg
可以用 -n -1 来循环解码

WQuit, 2021-04-01:
***@***.***:/home/code/mpp/build_mpp/test# ./mpi_dec_test -i /home/big_buck_bunny_480p.mp4 -t 8 -w 640 -h 480 -n -1
mpp[32172]: mpi_dec_utils: infinite loop decoding mode
mpp[32172]: mpi_dec_utils: cmd parse result:
mpp[32172]: mpi_dec_utils: input  file name: /home/big_buck_bunny_480p.mp4
mpp[32172]: mpi_dec_utils: output file name: 
mpp[32172]: mpi_dec_utils: config file name: 
mpp[32172]: mpi_dec_utils: width      :  640
mpp[32172]: mpi_dec_utils: height     :  480
mpp[32172]: mpi_dec_utils: type       : 8
mpp[32172]: mpi_dec_utils: debug flag : 0
mpp[32172]: mpi_dec_utils: max frames : -1
mpp[32172]: mpi_dec_test: mpi_dec_test start
mpp[32172]: mpi_dec_test: input file size 3992327
mpp[32172]: mpp_rt: NOT found ion allocator
mpp[32172]: mpp_rt: found drm allocator
mpp[32172]: mpp_info: mpp version: 4e240f0 author: Yandong Lin    2021-02-17 [m2vd_parser]: Fix the frame be output repeatly issue
mpp[32172]: mpi_dec_test: 0x8162aac0 mpi_dec_test decoder test start w 640 h 480 type 8
mpp[32172]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
mpp[32172]: mpi_dec_test: 0x8162aac0 decoded frame 0
mpp[32172]: mpi_dec_test: 0x8162aac0 input 0 pkt output 0 frm decode 0 frames
mpp[32172]: mpi_dec_test: test success max memory 0.00 MB

你好，目前测试输出如下，我不确认是不是我编译的问题。我用的gcc-8.4

WQuit, 2021-04-01:
> […](#)
> ------------------ 原始邮件 ------------------ 发件人: "Herman ***@***.***&gt;; 发送时间: 2021年4月1日(星期四) 上午9:20 收件人: ***@***.***&gt;; 抄送: ***@***.***&gt;; ***@***.***&gt;; 主题: Re: [rockchip-linux/mpp] mpi_dec_test解码mjpeg视频文件只能解码一帧 (#170)

问题已解决

HermanChen, 2021-04-01:
了解，问题解决麻烦关闭问题

zzzgno5, 2023-08-08:
-n 参数加了确实会多解帧数，但是都试解得同一张图，mjpeg是连续的应该，但是mpi_dec_test 只解第一帧，解-n次~~~~

是demo有问题是吗？

closed 2023-08-08 by HermanChen