rockchip-linux/mpp #174: rk18081平台下用mpi_enc_testj进行jpg编码出错。

liyungithub, 2020-12-25:
[image]

liyungithub, 2020-12-25:
[root@rk1808:/userdata/mpi_test]# ./mpi_enc_test -i gx3.nv12 -t 8 -f 0 -w 1920 -h 1080 -o nv12.jpg -n 1
mpp[946]: mpi_enc_utils: cmd parse result:
mpp[946]: mpi_enc_utils: input  file name: gx3.nv12
mpp[946]: mpi_enc_utils: output file name: nv12.jpg
mpp[946]: mpi_enc_utils: width      : 1920
mpp[946]: mpi_enc_utils: height     : 1080
mpp[946]: mpi_enc_utils: format     : 0
mpp[946]: mpi_enc_utils: type       : 8
mpp[946]: mpi_enc_test: mpi_enc_test start
mpp[946]: mpp_rt: NOT found ion allocator
mpp[946]: mpp_rt: found drm allocator
mpp[946]: mpp_info: mpp version: b5a5bb84 author: Herman Chen   2020-12-16 [vepu541]: Update vepu541 aq mode parameter
mpp[946]: mpi_enc_test: 0xdf36a00 mpi_enc_test encoder test start w 1920 h 1080 type 8
mpp[946]: vcodec_service: vcodec_service_cmd_send ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
mpp[946]: hal_jpege_vepu2: hal_jpege_vepu2_start send cmd failed 14
mpp[946]: mpp_enc: mpp 0xdf36a70 mpp_enc_hal_start:1042 failed return 14
mpp[946]: mpp_enc: mpp 0xdf36a70 mpp_enc_normal:1668 failed return 14
mpp[946]: mpi_enc_test: 0xdf36a00 encoded frame 0    size 20
mpp[946]: mpi_enc_test: 0xdf36a00 encode max 1 frames
mpp[946]: mpi_enc_test: 0xdf36a00 mpi_enc_test success total frame 1 bps 4800

closed 2020-12-26 by liyungithub

bakwc, 2021-08-08:
@liyungithub how did you solve it?