rockchip-linux/mpp #177: encoding with gstreamer-rockchiop causes invalid bitstream, help debugging settings

xandey, 2021-01-12:
Hi, I'm trying to get gstreamer-rockchip to work using the newer version of that package. I am preparing to make some patches and would prefer to patch against the current development.

The issues I see appears to be the some NAL output corruption as reported by h264parse in gstreamer 1.18.1, Maybe the output isn't byte-stream? I'm not sure. 

I see the problem using current versions of mpp and gstreamer-rockhip as of today:
gstreamer-rockchip: rockchip-linux/gstreamer-rockchip@a779505710996459166dc1ff4a003a2b7acc8292
mpp: 2e5ea77cb2969a1b240d8154fe42f33d898546f5

I have it working using the same mpp version but downgrading the gstreamer package to rockchip-linux/gstreamer-rockchip@490375216e0e88decf0752147504bc5dba00312e

I'm doing h264 encode using an RK3288 chip. My build system seems to break the VCS detection.

I'm using the following gstreamer pipeline:
`gst-launch-1.0 -ve videotestsrc ! mpph264enc ! h264parse ! fakesink silent=0`

I get a warning:
`0:00:00.087572161  2037  0x1353200 WARN               h264parse gsth264parse.c:1452:gst_h264_parse_handle_frame:<h264parse0> input stream is corrupt; it contains a NAL unit of length 0`

I suspect there is some configuration error in gstreamer-rockchip but I'm having a tough time getting the right debug output from mpp to determine what is wrong. I don't see any meaningful difference (a few different defaults but nothing really header related).

I posted here because it seems more active, and the issue may be api related.

I'm attaching the output of
`mpp_debug=0xfe GST_DEBUG=h264parse:4,mppvideoenc:5 gst-launch-1.0 -ve videotestsrc ! mpph264enc ! h264parse ! fakesink silent=0`
 * broken rockchip-linux/gstreamer-rockchip@a779505710996459166dc1ff4a003a2b7acc8292: 
[a779505.txt](https://github.com/rockchip-linux/mpp/files/5805027/a779505.txt)
 * working rockchip-linux/gstreamer-rockchip@490375216e0e88decf0752147504bc5dba00312e: 
[4903752.txt](https://github.com/rockchip-linux/mpp/files/5805026/4903752.txt)

I have a few patches in the old version to control bitrate and gop, but I got similar behavior using a newer commit without those patches, but it needed another patch to fix a bug rockchip-linux/gstreamer-rockchip@9440c06875f9a2491d092653beaf763aff1efcd0). I can repeat the capture without them if there are no clues.

xandey, 2021-01-15:
I did some more digging into this. I found from a bisect that rockchip-linux/gstreamer-rockchip@c796bc1e1db8d5fc28d4c456863175ec81126fae broke the stream. The commit message talks about switching "new encode config API".

I was able to turn on more debugging in the gst parser and with the broken code there is a start code followed by a zero word where there used to be the PPS NAL.

For the working one I get the output below, but the latest gstreamer-rockchip shows things breaking when it should have been parsing the PPS (type 8).

the line below is where things differ.
```
0:00:00.084202416  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:250:gst_h264_parse_nalu_header: Nal type 8, ref_idc 3
```

```
0:00:00.083959180  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:250:gst_h264_parse_nalu_header: Nal type 7, ref_idc 3
0:00:00.084035301  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:250:gst_h264_parse_nalu_header: Nal type 7, ref_idc 3
0:00:00.084060674  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:1529:gst_h264_parser_identify_nalu: Complete nal found. Off: 4, Size: 24
0:00:00.084100630  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:1907:gst_h264_parse_sps: parsing SPS
0:00:00.084134170  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:424:gst_h264_parse_vui_parameters: parsing "VUI Parameters"
0:00:00.084161002  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:1637:gst_h264_parser_parse_sps: adding sequence parameter set with id: 0 to array
0:00:00.084202416  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:250:gst_h264_parse_nalu_header: Nal type 8, ref_idc 3
0:00:00.084224873  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:1529:gst_h264_parser_identify_nalu: Complete nal found. Off: 32, Size: 4
0:00:00.084243538  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:2037:gst_h264_parse_pps: parsing PPS
0:00:00.084268037  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:2169:gst_h264_parser_parse_pps: adding picture parameter set with id: 0 to array
0:00:00.084303035  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:250:gst_h264_parse_nalu_header: Nal type 6, ref_idc 0
0:00:00.084325200  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:1529:gst_h264_parser_identify_nalu: Complete nal found. Off: 40, Size: 60
0:00:00.084344741  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:2486:gst_h264_parser_parse_sei: parsing SEI nal
0:00:00.084363698  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:1273:gst_h264_parser_parse_sei_message: parsing "SEI message"
0:00:00.084384697  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:1294:gst_h264_parser_parse_sei_message: SEI message received: payloadType  5, payloadSize = 448 bits
0:00:00.084420570  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:250:gst_h264_parse_nalu_header: Nal type 6, ref_idc 0
0:00:00.084443027  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:1529:gst_h264_parser_identify_nalu: Complete nal found. Off: 104, Size: 80
0:00:00.084461692  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:2486:gst_h264_parser_parse_sei: parsing SEI nal
0:00:00.084478608  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:1273:gst_h264_parser_parse_sei_message: parsing "SEI message"
0:00:00.084498732  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:1294:gst_h264_parser_parse_sei_message: SEI message received: payloadType  5, payloadSize = 616 bits
0:00:00.084531396  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:250:gst_h264_parse_nalu_header: Nal type 5, ref_idc 1
0:00:00.084557353  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:1514:gst_h264_parser_identify_nalu: Nal start 190, No end found
0:00:00.084580102  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:2232:gst_h264_parser_parse_slice_hdr: parsing "Slice header", slice type 2
0:00:00.084599642  2036   0xb77e00 DEBUG      codecparsers_h264 gsth264parser.c:709:gst_h264_slice_parse_dec_ref_pic_marking: parsing "Decoded reference picture marking"
```

xandey, 2021-01-15:
I'm able to work around this by setting `sei-mode=one-frame` and using the following pipeline:
```
gst-launch-1.0 -ve videotestsrc ! mpph264enc sei-mode=one-frame ! h264parse ! fakesink silent=0
```

However, when I change sei_mode to disable in mpi_enc_test.c I get a video with SPS and PPS, but when it's left disabled in the gstreamer-rockchip plugin it gives SPS but then the NAL which should contain PPS is just a start code followed by 4 bytes of zero.

I didn't find an easy way to dump the settings when the issue happens so I'm not sure what the remaining difference is between mpi_enc_test and mpph264enc defaults.

Is there a way to dump the MppEncCfg to console?

JeffyCN, 2021-05-17:
Hi xandey,

This issue should be fixed in:
commit 2b69a5e190197e5943d5fa3de73157ef894da7c2
Author: Herman Chen <herman.chen@rock-chips.com>
Date:   Fri Mar 19 11:32:44 2021 +0800

    [hal_h264e_vepu1]: Fix error on sei_mode

    The remain stream msb/lsb is rewrited by mistake.

    Change-Id: I99bf21852b49bcecda103e7c9bd8051364c29c32
    Signed-off-by: Herman Chen <herman.chen@rock-chips.com>
    Signed-off-by: Jeffy Chen <jeffy.chen@rock-chips.com>

closed 2021-05-17 by JeffyCN