rockchip-linux/mpp #178: 编码 稳定码率控制

Tristin-9527, 2021-01-14:
是不是有智能码率控制? CIF分辨率，编码尝试了调试CBR VBR gop qp参数，有办法让码率稳定在我想要的码率附近吗？允许上下一点波动，比如我要1024kbps，试了下没成功，很多时候都不稳定，动静态希望都稳住。

HermanChen, 2021-01-15:
用的什么测试源？还有最大最小 qp 配置是多少的？
是直接使用 mpi 接口么？

Tristin-9527, 2021-01-15:
> 用的什么测试源？还有最大最小 qp 配置是多少的？
> 是直接使用 mpi 接口么？

参照mpi_enc_test写的
芯片: RK3288
平台: Android 9.0
测试源: 底层取的yuv数据
主要参数:

mpp_enc_cfg_set_s32(m_EncCfg, "rc:bps_target", mnBitRate);
mpp_enc_cfg_set_s32(m_EncCfg, "rc:bps_max", mnBitRate * 17 / 16);
mpp_enc_cfg_set_s32(m_EncCfg, "rc:bps_min", mnBitRate * 15 / 16);

mpp_enc_cfg_set_s32(m_EncCfg, "rc:fps_in_flex", 0);
mpp_enc_cfg_set_s32(m_EncCfg, "rc:fps_in_num", 30);
mpp_enc_cfg_set_s32(m_EncCfg, "rc:fps_in_denorm", 1);
mpp_enc_cfg_set_s32(m_EncCfg, "rc:fps_out_flex", 0);
mpp_enc_cfg_set_s32(m_EncCfg, "rc:fps_out_num", 30);
mpp_enc_cfg_set_s32(m_EncCfg, "rc:fps_out_denorm", 1);
mpp_enc_cfg_set_s32(m_EncCfg, "rc:gop", 30);

mpp_enc_cfg_set_s32(m_EncCfg, "h264:qp_init", 22);
mpp_enc_cfg_set_s32(m_EncCfg, "h264:qp_max", 26);
mpp_enc_cfg_set_s32(m_EncCfg, "h264:qp_min", 12);
mpp_enc_cfg_set_s32(m_EncCfg, "h264:qp_max_i", 26);
mpp_enc_cfg_set_s32(m_EncCfg, "h264:qp_min_i", 12);

现象: CBR模式下，调QP相关的值，没法稳定在目标码率，取不到合适的ＱＰ值，帧率稳定了３０左右，码率会飙到２、３ｋｂｐｓ（比如我设置１ｋｂｐｓ），有时候两个参数又很低，挥手还有拖影

诉求: 码率和帧率能不能稳定在我设置的目标范围里边？模式不限

HermanChen, 2021-01-15:
使用 mpi_enc_test 的命令是什么样的？

HermanChen, 2021-01-15:
mpp_enc_cfg_set_s32(m_EncCfg, "h264:qp_max", 26);
mpp_enc_cfg_set_s32(m_EncCfg, "h264:qp_max_i", 26);
这两个值太小了，配置到 48

Tristin-9527, 2021-01-15:
> 使用 mpi_enc_test 的命令是什么样的？

我不是编码单个的视频文件　是取到ｙｕｖ数据后不断送去硬解的

qvoid, 2021-01-15:
> > 使用 mpi_enc_test 的命令是什么样的？
> 
> 我不是编码单个的视频文件　是取到ｙｕｖ数据后不断送去硬解的

一样可以的，把yuv单独保存成一个文件，可以直接用mpi_enc_test 进行编码，如下命令行：
```
mpi_enc_test -i input.yuv -w 320 -h 288 -t 7 -o ./test.h264
```

Tristin-9527, 2021-01-15:
> > > 使用 mpi_enc_test 的命令是什么样的？
> > 
> > 
> > 我不是编码单个的视频文件　是取到ｙｕｖ数据后不断送去硬解的
> 
> 一样可以的，把yuv单独保存成一个文件，可以直接用mpi_enc_test 进行编码，如下命令行：
> 
> ```
> mpi_enc_test -i input.yuv -w 320 -h 288 -t 7 -o ./test.h264
> ```

这个我知道呀 以文件的形式解码 我只能根据文件大小和时长算个大概码率 这样看不太出来变化

qvoid, 2021-01-15:
编码出来的码流可以发上来看看

Tristin-9527, 2021-01-18:
> 编码出来的码流可以发上来看看
附件哪里传... 文件我试过了 ffplay播放是正常的呢

qvoid, 2021-01-18:
> > 编码出来的码流可以发上来看看
> > 附件哪里传... 文件我试过了 ffplay播放是正常的呢

用百度网盘、onedrive等分享都可以

Tristin-9527, 2021-01-18:
链接: https://pan.baidu.com/s/15Ht3NrI6WkHsAc_EVkRDqQ 提取码: ecus 
链接: https://pan.baidu.com/s/1v_sqtTHUq01PIveC0mbI-w 提取码: ijpw

Tristin-9527, 2021-01-18:
> > > 编码出来的码流可以发上来看看
> > > 附件哪里传... 文件我试过了 ffplay播放是正常的呢
> 
> 用百度网盘、onedrive等分享都可以

链接: https://pan.baidu.com/s/15Ht3NrI6WkHsAc_EVkRDqQ 提取码: ecus
链接: https://pan.baidu.com/s/1v_sqtTHUq01PIveC0mbI-w 提取码: ijpw

qvoid, 2021-01-18:
mpp做过修改是吗？帧内QP 的RC被关闭了，所以整体画面质量会好些但是相应的码率会比较大些。另外qp配置的小了点，可以按照Herman的建议设置后再编码看看。

Tristin-9527, 2021-01-20:
反查了我们的代码 是mpp编码后 包变大了 我们的webrtc发包做了限制 导致丢帧  
不好意思 占用各位大佬的时间了

HermanChen, 2021-01-20:
没事，问题解决了就好了：）