rockchip-linux/mpp #180: rk3288 硬解码 以及转码耗时问题

wangjl-william, 2021-01-29:
rk3288 ubuntu18.04系统 
硬解码 1080P h264的码流，硬解码 耗时 14ms 左右 ，但是解码出来的mppframe 无法直接 用在 ffmpeg上使用
所以需要拷贝一次 然后在进行转码  转码前拷贝 耗时 29ms  ffmpeg 转码 耗时 59ms  
这样算下来 硬解 + 拷贝 + 转码 需要 100ms左右  这样的话  1秒 也就是显示 10帧， 实现不了1080P每秒 25帧 ，更不能奢望4K的。

请教大神以下问题：
1）mpp 硬解码后的frame  是否 可以直接得到  rgba /rgb/bgr等格式的frame ？
2）mpp是否能有一种方法，可以直接将 解码后的frame 直接转成 ffmpeg中的avframe ？
3）mpp是否有自己的转码接口？
4）想充分利用芯片（4K*2K硬解码）如果想输出 4K@30fps的视频，我应该怎么去做？

wangjl-william, 2021-01-29:
> ffmpeg 用的是 malloc 的 buffer，mpp 和硬件需要使用 ion / drm buffer，即 dmabuf
> MppFrame 可以通过一些修改挂到 avframe 下面也可以使用，只是非标接口

> 2. 如上，需要做非标修改
> 3. 转码接口指的是 264 -> 265 这种的转码接口么？不过 mpp 可以实现解码再编码的功能，在 mpp 的 rc test 里有
> 4. 使用 mpp 硬件解码，然后得到的 buffer 直接送显示，可以使用 rockit 这样的中间件来处理

2. 如何修改，请指导，这样的话 可以少一次拷贝，节省大量时间。
3.转码接口是 ffmpeg  264解码后nv12的数据 转码成 rgba？  你的意思 我可以 将264的帧 直接 编码成 rgba的帧 送显示？
  编码也是硬件做的是吧？大概速度有多快？
4.不好意思 HermanChen， 我也是最近才接触 mpp硬解码，你说的rockit 是什么？能够 详细说明 或者 发一个教程的连接？

HermanChen, 2021-01-29:
2. 这个只能自己去了解下 ubuntu 上的显示框架，然后做接入了
3. yuv -> rgba 只是格式转换，不是编码
4. rockit 是瑞芯微的媒体处理中间件的，在 RK 的 Android sdk 里可以看到

wangjl-william, 2021-01-29:
> 1. 这个只能自己去了解下 ubuntu 上的显示框架，然后做接入了
> 2. yuv -> rgba 只是格式转换，不是编码
> 3. rockit 是瑞芯微的媒体处理中间件的，在 RK 的 Android sdk 里可以看到
2.“转码接口指的是 264 -> 265 这种的转码接口么？不过 mpp 可以实现解码再编码的功能，在 mpp 的 rc test 里有”
----那之前说的是什么意思？
3.用的是  ubuntu 不是 Android ，应该用不了Android sdk 吧

HermanChen, 2021-01-29:
之前我说的转码指的是不同格式的视频，比如 mpeg4 的视频先解码，再编码成 264 这样，这种是一般意义上的转码。
RK 有 Android sdk 对外发布，可以参考下

wangjl-william, 2021-01-29:
> 之前我说的转码指的是不同格式的视频，比如 mpeg4 的视频先解码，再编码成 264 这样，这种是一般意义上的转码。
> RK 有 Android sdk 对外发布，可以参考下

也就是说 mpp这边没有好办法 解决 解码后 转码耗时长的问题是吧

JeffyCN, 2021-02-06:
ffmpeg出来的是drm primary(nv12), 显示端如果支持是可以直接显示。如debian sdk的mpv(drm直接送显或者dma buf走xserver的xv hack流程)

gitccl, 2021-09-17:
请问你解决了吗？

smzahraee, 2024-10-27:
where is the source code of rockit driver