rockchip-linux/mpp #181: jpeg硬解码出现IRQ STREAM ERROR问题

liyungithub, 2021-01-29:
硬件：rk1808
我现在是在对设备的jpeg硬解码情况做鲁棒性测试，把一些能触发产生异常的jpeg: unsupported coding type (0xc2)问题的jpeg放到设备测试，只要测试2~3分钟就产IRQ STREAM ERROR问题，就不能工作了，怎么避免这问题或者产生这问题之后怎么恢复正常？
[image]

HermanChen, 2021-01-29:
麻烦提供下测试源，还有出现卡住的测试源

liyungithub, 2021-01-29:
可以，测试源是2000多张图片（84M），发你邮箱？出问题不是某一张引起的，有时间的累积

liyungithub, 2021-01-29:
邮箱提供下？

FumasterLin, 2021-01-29:
Hi：
84M 邮箱附件放不下。
可以存到网盘哈。

liyungithub, 2021-01-29:
百度网盘链接（一天内有效）：链接: https://pan.baidu.com/s/1jQRsQDeEAVOcA1Rz5IMCsA 提取码: qf1w 复制这段内容后打开百度网盘手机App，操作更方便哦

HermanChen, 2021-01-30:
积累的话，需要检查是不是上层调用异常出的累积了，有没出问题时的 log？

liyungithub, 2021-01-30:
链接: https://pan.baidu.com/s/1eZ5F6VHNTx1WcqtEj1lprw 提取码: q98m 复制这段内容后打开百度网盘手机App，操作更方便哦
log有2m，也放到网盘了

liyungithub, 2021-01-30:
dmesg:
[1983727.156510] rk-vcodec ffb80000.vpu_service: resetting...
[1983727.156619] rk-vcodec ffb80000.vpu_service: reset done
[1983727.156650] rk-vcodec ffb80000.vpu_service: reset done
[1983727.184055] rk-vcodec ffb80000.vpu_service: resetting...
[1983727.184165] rk-vcodec ffb80000.vpu_service: reset done
[1983727.184195] rk-vcodec ffb80000.vpu_service: reset done
[1983727.207979] rk-vcodec ffb80000.vpu_service: resetting...
[1983727.208091] rk-vcodec ffb80000.vpu_service: reset done
[1983727.208122] rk-vcodec ffb80000.vpu_service: reset done
[1983727.232418] rk-vcodec ffb80000.vpu_service: resetting...
[1983727.232524] rk-vcodec ffb80000.vpu_service: reset done
[1983727.232554] rk-vcodec ffb80000.vpu_service: reset done
[1983727.256927] rk-vcodec ffb80000.vpu_service: resetting...
[1983727.257033] rk-vcodec ffb80000.vpu_service: reset done
[1983727.257063] rk-vcodec ffb80000.vpu_service: reset done
[1983727.275254] rk-vcodec ffb80000.vpu_service: resetting...
[1983727.275364] rk-vcodec ffb80000.vpu_service: reset done
[1983727.275395] rk-vcodec ffb80000.vpu_service: reset done
[1983727.299743] rk-vcodec ffb80000.vpu_service: resetting...
[1983727.299854] rk-vcodec ffb80000.vpu_service: reset done
[1983727.299887] rk-vcodec ffb80000.vpu_service: reset done
[1983727.325896] rk-vcodec ffb80000.vpu_service: resetting...
[1983727.326005] rk-vcodec ffb80000.vpu_service: reset done
[1983727.326035] rk-vcodec ffb80000.vpu_service: reset done
[1983727.350961] rk-vcodec ffb80000.vpu_service: resetting...
[1983727.351072] rk-vcodec ffb80000.vpu_service: reset done
[1983727.351101] rk-vcodec ffb80000.vpu_service: reset done
[1983727.390745] rk-vcodec ffb80000.vpu_service: resetting...
[1983727.390856] rk-vcodec ffb80000.vpu_service: reset done
[1983727.390886] rk-vcodec ffb80000.vpu_service: reset done

liyungithub, 2021-01-30:
补充下问题：如果发生jpeg: unsupported coding type (0xc2)的调用错误，是否还可以继续调用？如果不行应该如何进行清理？我当前就是没有做任何的特殊处理再连续调用后就出现IRQ STREAM ERROR，之后整个jpeg解码工作就不能进行了。

HermanChen, 2021-01-30:
最好做下 reset

HermanChen, 2021-01-30:
mpi 有 reset 接口

liyungithub, 2021-01-31:
增加了mpi->reset ，还是有问题

liyungithub, 2021-01-31:
[mpp_jpg.zip](https://github.com/rockchip-linux/mpp/files/5899364/mpp_jpg.zip)

liyungithub, 2021-01-31:
上传了代码你看看，是不是有些流程没处理好？

FumasterLin, 2021-02-01:
Hi：

你可以更新下mpp，我们基于同样的mpp版本来分析

Thanks

liyungithub, 2021-02-01:
好的，谢谢了

liyungithub, 2021-02-01:
下了新了mpp源码但适应不了板子啊，要怎么处理？
[image]

HermanChen, 2021-02-01:
有什么错误log？

FumasterLin, 2021-02-01:
Hi：
看一下dmesg

Thanks

Nybon, 2021-02-02:
This seems to be related to #182

liyungithub, 2021-02-02:
[image]

liyungithub, 2021-02-02:
[image]

FumasterLin, 2021-02-03:
Hi：

你这个是必现的吗？还是特定图片必现的？

我本地找了个1808的板子测试是没问题的哦。

kernel 4.4 + 最新的mpp

`[root@rk1806:/mnt/sdcard]# cat /proc/version 
Linux version 4.4.194 (wzh@SYS3) (gcc version 6.3.1 20170404 (Linaro GCC 6.3-2017.05) ) #2 SMP PREEMPT Fri Dec 11 16:12:40 CST 2020
mpp[6042]: mpp_info: mpp version: 2e5ea77c author: Johnson Ding  2021-01-05 [jpegd]: Optimize marker searching`

Thanks

liyungithub, 2021-02-04:
再问一个问题：1808板子的jpg硬解码只支持SOF0（基线离散余弦变换），其它的SOF1、SOF2、SOF3等都不支持？
[image]

liyungithub, 2021-02-04:
#2 SMP PREEMPT Fri Dec 11 16:12:40 CST 2020 mpp[6042]: mpp_info: mpp version: 2e5ea77c author: Johnson Ding 2021-01-05 [jpegd]: Optimize marker searching
我先下载你更新的jpegd_parser.c验证下。

HermanChen, 2021-02-04:
己更新，再试下看看

HermanChen, 2021-02-04:
> 
> 
> 再问一个问题：1808板子的jpg硬解码只支持SOF0（基线离散余弦变换），其它的SOF1、SOF2、SOF3等都不支持？
> [image]

是的

liyungithub, 2021-02-04:
> 己更新，再试下看看
好的