rockchip-linux/mpp #183: mpph264enc rotation=2 does nothing, rotation=1 crashes the entire device, rotation=3 produces bar artifacts

billythedummy, 2021-01-31:
**device:** RK3399

Output video from
```
gst-launch-1.0 -e v4l2src ! mpph264enc rotation=0 ! h264parse ! mp4mux ! filesink location=test.mp4
```
is the same as output video from 
```
gst-launch-1.0 -e v4l2src ! mpph264enc rotation=2 ! h264parse ! mp4mux ! filesink location=test.mp4
```

This pipeline causes the entire rk3399 device to crash (unresponsive):
```
gst-launch-1.0 -e v4l2src ! mpph264enc rotation=1 ! h264parse ! mp4mux ! filesink location=test.mp4
```

The output video from this pipeline is severely distorted with the whole video comprising vertical bars 
```
gst-launch-1.0 -e v4l2src ! mpph264enc rotation=3 ! h264parse ! mp4mux ! filesink location=test.mp4
```

For rotation=1 and rotation=3, this mpp error message was observed:
```
mpp[1413]: hal_h264e_vepu_v2: warnning: input buffer size 0x2fd000 is smaller than required size 0x384000
```

JeffyCN, 2021-05-17:
i can repro this issue on rk3399 evb(with 4.4/4.19 BSP kernel), but seems like rk356x(with 4.19 kernel) work well. might due to different IP related code in MPP.

@HermanChen

JeffyCN, 2021-05-17:
well, checking the MPP code:
mpp-release# cgrep rotation -wl
./inc/rk_venc_cmd.h
./test/mpi_enc_multi_test.c
./debian/tmp/usr/include/rockchip/rk_venc_cmd.h
./debian/librockchip-mpp-dev/usr/include/rockchip/rk_venc_cmd.h
./mpp/common/jpege_syntax.h
./mpp/hal/rkenc/h264e/hal_h264e_vepu541.c <--- it's the only h264e hal which supports rotation property
./mpp/hal/rkenc/h264e/hal_h264e_vepu541_reg.h
./mpp/hal/rkenc/h265e/hal_h265e_vepu541.c
./mpp/hal/vpu/vp8e/hal_vp8e_base.c
./mpp/hal/vpu/jpege/hal_jpege_vepu1_v2.c
./mpp/hal/vpu/jpege/hal_jpege_vepu2_v2.c
./mpp/codec/enc/jpeg/jpege_api_v2.c
./mpp/codec/enc/h264/h264e_api_v2.c
./mpp/codec/enc/vp8/vp8e_api_v2.c
./mpp/codec/enc/h265/h265e_api.c
./mpp/version.h
./mpp/vproc/inc/rga_api.h
./mpp/legacy/ppOp.cpp
./mpp/legacy/ppOp.h
./mpp/base/mpp_enc_cfg.cpp
./osal/linux/drm_mode.h

it seems like the ./mpp/codec/enc/h264/h264e_api_v2.c provide rotation property, but only rkenc support it. so only these chips(with rkvenc module) can to rotation for h264e:
mpp-release# ag HAVE_RKVENC osal/mpp_soc.cpp  -B 1
517-        ROCKCHIP_SOC_RV1108,
518:        HAVE_VDPU2 | HAVE_VEPU2 | HAVE_RKVDEC | HAVE_RKVENC,
--
529-        ROCKCHIP_SOC_RV1109,
530:        HAVE_VDPU2 | HAVE_VEPU2 | HAVE_RKVDEC | HAVE_RKVENC,
--
541-        ROCKCHIP_SOC_RV1126,
542:        HAVE_VDPU2 | HAVE_VEPU2 | HAVE_RKVDEC | HAVE_RKVENC,
--
585-        ROCKCHIP_SOC_RK3566,
586:        HAVE_VDPU2 | HAVE_VDPU2_PP | HAVE_VEPU2 | HAVE_RKVDEC | HAVE_RKVENC | HAVE_JPEG_DEC,
--
598-        ROCKCHIP_SOC_RK3568,
599:        HAVE_VDPU2 | HAVE_VDPU2_PP | HAVE_VEPU2 | HAVE_RKVDEC | HAVE_RKVENC | HAVE_JPEG_DEC,

JeffyCN, 2021-12-13:
the newest gst mpp plugin would use rga to rotate:
https://github.com/JeffyCN/rockchip_mirrors/commit/c27f61d63aeea7104afab971902f0fc1b7680aed

closed 2021-12-13 by JeffyCN