rockchip-linux/mpp #186: VP9: Sign bias is parsed by never used

ndufresne, 2021-03-08:
First, I'm not a VP9 expert, so bare with my limited understanding. We are working on mainline driver for the Linux Kernel, and while running through VP9 test vectors we found that most test into which the sign_bias is used fails. We came to mpp thinking it would be obvious and notice that this is parsed and never used when programming the HW. I totally failed to see anyway it's being used, my obvious next step would be to run the compliance with mpp, meanwhile I'm seeking for Rockchip expert to clarify how the decoder is supposed to work without this information.

We know of a fact that Hantro G2 and VC8000D do require this information in their register and seems to behave correctly for these cases. Thanks in advance.

ndufresne, 2021-03-15:
Any comment on this ?

HermanChen, 2021-03-17:
Our colleagues are checking this question.
sign_bias is used in reference frame selection and mv caculation. I think the later part could be implemented by hardware.

HermanChen, 2021-03-17:
The hardware will decode and use these flag in hardware. The driver does not need to care about it.

ndufresne, 2021-03-17:
Interesting, so the HW parses some of uncompressed header. We are bringing up mpp now (we work on mainline support), and we will run the VP9 test vector against it to see if tests result do match. Are you also validating with the VP9 test vector ?

fyi, We use Fluster, a simple runner that supports gstreamer to validate the codecs, https://github.com/fluendo/fluster (hope this can be handy for you too)

ndufresne, 2021-03-17:
Of course we should assume this is a non-issue, closing to avoid leaving this in your issue tracker, thanks for the follow up.

closed 2021-03-17 by ndufresne