rockchip-linux/mpp #188: mpi_dec_test 解码失败

WQuit, 2021-04-01:
开发板：rk3288
mpp version: 4e240f0
gcc-version:gcc-8.4
系统信息：
`root@rk3288:/home/code/mpp/build_mpp/osal/test# ./mpp_platform_test 
mpp[4037]: mpp_plat_test: kernel version: 4.4
mpp[4037]: mpp_plat_test: ioctl  version: vcodec_service
mpp[4037]: mpp_plat_test: 
mpp[4037]: mpp_plat_test: chip name: rockchip,rk3288-firefly rockchip,rk3288
mpp[4037]: mpp_plat_test: 
mpp[4037]: mpp_plat_test: chip vcodec type 00020105
mpp[4037]: mpp_plat_test: found vpu1 codec
mpp[4037]: mpp_plat_test: found rk hevc decoder
mpp[4037]: mpp_plat_test: 
mpp[4037]: mpp_plat_test: start probing decoder device name:
mpp[4037]: mpp_plat_test: H.264 decoder: /dev/vpu-service
mpp[4037]: mpp_plat_test: H.265 decoder: /dev/hevc-service
mpp[4037]: mpp_plat_test: MJPEG decoder: /dev/vpu-service
mpp[4037]: mpp_plat_test: VP9   decoder: /dev/vpu-service
mpp[4037]: mpp_plat_test: avs   decoder: /dev/vpu-service
mpp[4037]: mpp_plat_test: 
mpp[4037]: mpp_plat_test: start probing encoder device name:
mpp[4037]: mpp_plat_test: H.264 encoder: /dev/vpu-service
mpp[4037]: mpp_plat_test: H.265 encoder: /dev/vpu-service
mpp[4037]: mpp_plat_test: MJPEG encoder: /dev/vpu-service
mpp[4037]: mpp_plat_test: mpp platform test done
root@rk3288:/home/code/mpp/build_mpp/osal/test# ./mpp_env_test 
mpp[4047]: mpp_env_test: set env: test_env_debug to 256
mpp[4047]: mpp_env_test: set env: test_env_string to just for debug
mpp[4047]: mpp_env_test: start reading env:
mpp[4047]: mpp_env_test: get env: test_env_debug is 256
mpp[4047]: mpp_env_test: get env: test_env_string is just for debug
root@rk3288:/home/code/mpp/build_mpp/osal/test# ./mpp_mem_test
mpp[4060]: mpp_mem_test: calloc  success ptr 0x0x81093080
mpp[4060]: mpp_mem_test: realloc success ptr 0x0x81095558
mpp[4060]: mpp_mem_test: mpp_mem_test done
`
测试信息：
`root@rk3288:/home/code/mpp/build_mpp/test# ./mpi_dec_test -i /home/big_buck_bunny_480p.mp4 -t 8 -n -1 -w 640 -h 480
mpp[4391]: mpi_dec_utils: infinite loop decoding mode
mpp[4391]: mpi_dec_utils: cmd parse result:
mpp[4391]: mpi_dec_utils: input  file name: /home/big_buck_bunny_480p.mp4
mpp[4391]: mpi_dec_utils: output file name: 
mpp[4391]: mpi_dec_utils: config file name: 
mpp[4391]: mpi_dec_utils: width      :  640
mpp[4391]: mpi_dec_utils: height     :  480
mpp[4391]: mpi_dec_utils: type       : 8
mpp[4391]: mpi_dec_utils: debug flag : 0
mpp[4391]: mpi_dec_utils: max frames : -1
mpp[4391]: mpi_dec_test: mpi_dec_test start
mpp[4391]: mpi_dec_test: input file size 3992327
mpp[4391]: mpp_rt: NOT found ion allocator
mpp[4391]: mpp_rt: found drm allocator
mpp[4391]: mpp_info: mpp version: 4e240f0 author: Yandong Lin    2021-02-17 [m2vd_parser]: Fix the frame be output repeatly issue
mpp[4391]: mpi_dec_test: 0x804eaac0 mpi_dec_test decoder test start w 640 h 480 type 8
mpp[4391]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
mpp[4391]: mpi_dec_test: 0x804eaac0 decoded frame 0
mpp[4391]: mpi_dec_test: 0x804eaac0 input 0 pkt output 0 frm decode 0 frames
mpp[4391]: mpi_dec_test: test success max memory 0.00 MB`
`root@rk3288:/home/code/mpp/build_mpp/test# ./mpi_dec_test -i /home/big_buck_bunny_480p.mp4 -t 7 -n -1
mpp[4135]: mpi_dec_utils: infinite loop decoding mode
mpp[4135]: mpi_dec_utils: cmd parse result:
mpp[4135]: mpi_dec_utils: input  file name: /home/big_buck_bunny_480p.mp4
mpp[4135]: mpi_dec_utils: output file name: 
mpp[4135]: mpi_dec_utils: config file name: 
mpp[4135]: mpi_dec_utils: width      :    0
mpp[4135]: mpi_dec_utils: height     :    0
mpp[4135]: mpi_dec_utils: type       : 7
mpp[4135]: mpi_dec_utils: debug flag : 0
mpp[4135]: mpi_dec_utils: max frames : -1
mpp[4135]: mpi_dec_test: mpi_dec_test start
mpp[4135]: mpi_dec_test: input file size 3992327
mpp[4135]: mpp_info: mpp version: 4e240f0 author: Yandong Lin    2021-02-17 [m2vd_parser]: Fix the frame be output repeatly issue
mpp[4135]: mpi_dec_test: 0x81136500 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[4135]: mpp_rt: NOT found ion allocator
mpp[4135]: mpp_rt: found drm allocator
mpp[4135]: mpi_dec_test: 0x81136500 loop again
mpp[4135]: mpi_dec_test: 0x81136500 loop again
mpp[4135]: mpi_dec_test: 0x81136500 loop again
mpp[4135]: mpi_dec_test: 0x81136500 loop again
mpp[4135]: mpi_dec_test: 0x81136500 loop again
mpp[4135]: mpi_dec_test: 0x81136500 loop again
mpp[4135]: mpi_dec_test: 0x81136500 loop again
mpp[4135]: mpi_dec_test: 0x81136500 loop again
mpp[4135]: mpi_dec_test: 0x81136500 loop again
`
看其他帖子知道；dmesg `没又看到相应的日志输出。日志输出也打开了。命令用的是echo 0x0100 > /sys/module/rk_vcodec/parameters/debug`
想知道，解码失败的原因是什么，输出一直为0

HermanChen, 2021-04-01:
功课做得很不错：）

mpi_dec_test -i /home/big_buck_bunny_480p.mp4 -t 8 -n -1 -w 640 -h 480
mpi_dec_test 是解祼码流的，不能解 mp4 这种带封装的格式，需要把 mp4 文件里的 264 raw 数据解出来才能解
用 ffmpeg 抽一下码流
ffmpeg -i big_buck_bunny_480p.mp4 -vcodec copy -an -bsf h264_mp4toannexb big_buck_bunny_480p.h264
然后把 264 码流给解码器
还有，264 的 mpi_dec_test 参数是 -t 7，不是 -t 8，8 是 jpeg

关于封装和祼码流，还有 ffmpeg 使用，建议去雷神的博客里学习
https://blog.csdn.net/leixiaohua1020

WQuit, 2021-04-01:
> 功课做得很不错：）
>
> mpi_dec_test -i /home/big_buck_bunny_480p.mp4 -t 8 -n -1 -w 640 -h 480
> mpi_dec_test 是解祼码流的，不能解 mp4 这种带封装的格式，需要把 mp4 文件里的 264 raw 数据解出来才能解
> 用 ffmpeg 抽一下码流
> ffmpeg -i big_buck_bunny_480p.mp4 [...]

好的，随后去测试。感谢。

Anne12345, 2021-04-01:
````shell
root@rk3288:/home/cqq/mpp/test# ./mpi_dec_test -i /home/test.h264 -t 7 -o /home/tt.yuv -w 1920 -h 1080
mpp[14692]: mpi_dec_utils: cmd parse result:
mpp[14692]: mpi_dec_utils: input  file name: /home/test.h264
mpp[14692]: mpi_dec_utils: output file name: /home/tt.yuv
mpp[14692]: mpi_dec_utils: config file name: 
mpp[14692]: mpi_dec_utils: width      : 1920
mpp[14692]: mpi_dec_utils: height     : 1080
mpp[14692]: mpi_dec_utils: type       : 7
mpp[14692]: mpi_dec_utils: debug flag : 0
mpp[14692]: mpi_dec_utils: max frames : 0
mpp[14692]: mpi_dec_test: mpi_dec_test start
mpp[14692]: mpi_dec_test: input file size 8330817
mpp[14692]: mpp_info: mpp version: 4e240f0 author: Yandong Lin    2021-02-17 [m2vd_parser]: Fix the frame be output repeatly issue
mpp[14692]: mpi_dec_test: 0x80b5b660 mpi_dec_test decoder test start w 1920 h 1080 type 7
mpp[14692]: mpp_rt: NOT found ion allocator
mpp[14692]: mpp_rt: found drm allocator
mpp[14692]: mpi_dec_test: 0x80b5b660 decode_get_frame get info changed found
mpp[14692]: mpi_dec_test: 0x80b5b660 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 3760128
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 0 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 1 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 2 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 3 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 4 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 5 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 6 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 7 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 8 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 9 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 10 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 11 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 12 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 13 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 14 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 15 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 16 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 17 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 18 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 19 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 20 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 21 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 22 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 23 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 24 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 25 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 26 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 27 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 28 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 29 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 30 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 31 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 32 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 33 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 34 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 35 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 36 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 37 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 38 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 39 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 40 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 41 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 42 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 43 err 1 discard 0
mpp[14692]: vcodec_service: vcodec_service_cmd_send ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
mpp[14692]: hal_h264d_vdpu1_reg: vdpu1_h264d_start send cmd failed 14
mpp[14692]: vcodec_service: vcodec_service_cmd_poll ioctl VPU_IOC_GET_REG failed ret -1 errno 110 Connection timed out
mpp[14692]: hal_h264d_vdpu1_reg: vdpu1_h264d_wait poll cmd failed 110
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 44 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 45 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 46 err 1 discard 0
[... 178 lines cut ...]
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 221 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 222 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 223 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 224 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 225 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 226 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 227 err 1 discard 0
[... cut ...]
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 236 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 237 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 238 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 found last packet
mpp[14692]: mpi_dec_test: 0x80b5b660 decode get frame 239 err 1 discard 0
mpp[14692]: mpi_dec_test: 0x80b5b660 found last packet
mpp[14692]: mpi_dec_test: 0x80b5b660 input 2034 pkt output 241 frm decode 240 frames
mpp[14692]: mpi_dec_test: test success max memory 21.52 MB
````
为什么解码不成功，哪里出问题了？

HermanChen, 2021-04-01:
上层解码过程正常，说明码流应该是对的，可能是内核驱动的匹配问题
像楼主一样提供下环境相关的信息

Anne12345, 2021-04-01:
> 上层解码过程正常，说明码流应该是对的，可能是内核驱动的匹配问题
> 像楼主一样提供下环境相关的信息

````shell
开发板：rk3288
mpp version: 4e240f0
gcc-version:gcc (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04) 7.5.0
系统信息：
root@rk3288:/home/cqq/mpp/osal/test# ./mpp_env_test 
mpp[15204]: mpp_env_test: set env: test_env_debug to 256
mpp[15204]: mpp_env_test: set env: test_env_string to just for debug
mpp[15204]: mpp_env_test: start reading env:
mpp[15204]: mpp_env_test: get env: test_env_debug is 256
mpp[15204]: mpp_env_test: get env: test_env_string is just for debug

root@rk3288:/home/cqq/mpp/osal/test# ./mpp_platform_test 
mpp[15207]: mpp_plat_test: kernel version: 4.4
mpp[15207]: mpp_plat_test: ioctl  version: vcodec_service
mpp[15207]: mpp_plat_test: 
mpp[15207]: mpp_plat_test: chip name: rockchip,rk3288-firefly rockchip,rk3288
mpp[15207]: mpp_plat_test: 
mpp[15207]: mpp_plat_test: chip vcodec type 00020105
mpp[15207]: mpp_plat_test: found vpu1 codec
mpp[15207]: mpp_plat_test: found rk hevc decoder
mpp[15207]: mpp_plat_test: 
mpp[15207]: mpp_plat_test: start probing decoder device name:
mpp[15207]: mpp_plat_test: H.264 decoder: /dev/vpu-service
mpp[15207]: mpp_plat_test: H.265 decoder: /dev/hevc-service
mpp[15207]: mpp_plat_test: MJPEG decoder: /dev/vpu-service
mpp[15207]: mpp_plat_test: VP9   decoder: /dev/vpu-service
mpp[15207]: mpp_plat_test: avs   decoder: /dev/vpu-service
mpp[15207]: mpp_plat_test: 
mpp[15207]: mpp_plat_test: start probing encoder device name:
mpp[15207]: mpp_plat_test: H.264 encoder: /dev/vpu-service
mpp[15207]: mpp_plat_test: H.265 encoder: /dev/vpu-service
mpp[15207]: mpp_plat_test: MJPEG encoder: /dev/vpu-service
mpp[15207]: mpp_plat_test: mpp platform test done

root@rk3288:/home/cqq/mpp/osal/test# ./mpp_mem_test 
mpp[15211]: mpp_mem_test: calloc  success ptr 0x0x815af080
mpp[15211]: mpp_mem_test: realloc success ptr 0x0x815b1538
mpp[15211]: mpp_mem_test: mpp_mem_test done
````
这是我的环境相关的信息

HermanChen, 2021-04-01:
麻烦提供下 test.h264 测试源

Anne12345, 2021-04-01:
> 麻烦提供下 test.h264 测试源

[test.zip](https://github.com/rockchip-linux/mpp/files/6241759/test.zip)
麻烦您了

WQuit, 2021-04-01:
[mpp_dec_test.log](https://github.com/rockchip-linux/mpp/files/6242028/mpp_dec_test.log)
你好，我用的楼上提供的264源解码测试的。提示解码成功，但是我在windows端使用 yuvplayer播放的时候，全部是绿屏。log是实际的测试文件

HermanChen, 2021-04-01:
https://github.com/HermanChen/mpp/
试一下这里的 develop 分支有没有问题

WQuit, 2021-04-01:
> https://github.com/HermanChen/mpp/
> 试一下这里的 develop 分支有没有问题

测试 release_20171218 tag 测试。编译还是之前的问题。解码数据为0. 
内核架构：Architecture:        armv7l
[mpp_dec_test_release_20171218.log](https://github.com/rockchip-linux/mpp/files/6242478/mpp_dec_test_release_20171218.log)

WQuit, 2021-04-01:
> https://github.com/HermanChen/mpp/
> 试一下这里的 develop 分支有没有问题

这个分支测试还是同样的问题。

HermanChen, 2021-04-01:
我测试了下，Android 的 4.4 内核是正常的
firefly 是 linux 平台上的内核，由不同产品部维护，我找人看一下是不是内核驱动有差异导致的问题。

WQuit, 2021-04-01:
````shell
[86259.749885] rk_vcodec: vpu_dec task: 187 ms
[86259.783979] rk_iommu ff9a0800.iommu: Enable stall request timed out, status: 0x00020b
[86259.783993] rk_iommu ff9a0800.iommu: Page fault at 0x00000ee0 of type read
[86259.784000] rk_iommu ff9a0800.iommu: iova = 0x00000ee0: dte_index: 0x0 pte_index: 0x0 page_offset: 0xee0
[86259.784007] rk_iommu ff9a0800.iommu: mmu_dte_addr: 0x2e17a000 dte@0x2e17a000: 0x000000 valid: 0 pte@0x00000000: 0x000000 valid: 0 page@0x00000000 flags: 0x0
[86259.784017] rk_iommu ff9a0800.iommu: Page fault at 0x00000ee0 of type read
[86259.784023] rk_iommu ff9a0800.iommu: iova = 0x00000ee0: dte_index: 0x0 pte_index: 0x0 page_offset: 0xee0
[86259.784030] rk_iommu ff9a0800.iommu: mmu_dte_addr: 0x2e17a000 dte@0x2e17a000: 0x000000 valid: 0 pte@0x00000000: 0x000000 valid: 0 page@0x00000000 flags: 0x0
[86259.818881] rk_iommu ff9a0800.iommu: Disable paging request timed out, status: 0x00020b
[86259.818931] rk-vcodec ff9a0000.vpu-service: resetting...
[86259.818964] rk-vcodec ff9a0000.vpu-service: reset done
[86259.818967] rk-vcodec ff9a0000.vpu-service: reset done
[86260.241882] rk_vcodec: reg_init:1248: error: translate reg address failed, dumping regs
[86260.241891] rk-vcodec ff9a0000.vpu-service: reg[00]: 67313688
````
目前测试了 所有的release tag 分支，所有的都不可以使用。release_20170319~0428 运行崩溃。0811~1123执行成功。但是播放是绿屏。
内核信息：Linux rk3288 4.4.154 #297 SMP Tue Apr 21 15:42:34 CST 2020 armv7l armv7l armv7l GNU/Linux
cmake信息：
````shell
root@rk3288:/home/code/mpp# cmake .
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Checking for one of the modules 'pthread'
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- author: Randy Li
-- author: Wed Sep 20 15:17:01 2017 +0800
-- git version: "7a3790a author: Randy Li [WIP]:[hal]: [h264e]: generate the SPS/PPS info before encoding"
-- Install git hooks done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/code/mpp
````
下边是1107 tag 的 输出日志
[mpp_dec_test_release_20171107.log](https://github.com/rockchip-linux/mpp/files/6242701/mpp_dec_test_release_20171107.log)

WQuit, 2021-04-01:
> 我测试了下，Android 的 4.4 内核是正常的
> firefly 是 linux 平台上的内核，由不同产品部维护，我找人看一下是不是内核驱动有差异导致的问题。

好的。感谢

HermanChen, 2021-04-02:
把问题发到 firefly 论坛上去看看，我们手上没有 firefly 的固件，比较麻烦，需要核对下内核 vcodec 相关的代码

WQuit, 2021-04-02:
> firefly

好的。感谢

WQuit, 2021-04-06:
> 把问题发到 firefly 论坛上去看看，我们手上没有 firefly 的固件，比较麻烦，需要核对下内核 vcodec 相关的代码

您好，在firefly论坛并没有人回复，相关资料也很少。只有gstream调用mpp相关的。您说的核对内核源码具体需要怎么操作，我这边尝试一下

closed 2021-04-07 by WQuit

luhejie, 2022-10-15:
楼主好，请问就解决了吗，我也遇到了这个问题

EZreal-zhangxing, 2023-05-24:
同样的问题，RK3588解码失败，mpi_test 也报失败
`mpp[8567]: mpi_test: mpi_test start
mpp[8567]: mpi_test: mpi_test decoder test start
mpp[8567]: mpi: mpp_create enter ctx 0x7fcd645488 mpi 0x7fcd645490
mpp[8567]: mpp_info: mpp version: 3c5dc7fe author: sayon.chen    2020-11-23 [h265d_syntax]: Fix h265 syntax fill sps rps issue
mpp[8567]: mpi: mpp_create leave ret 0 ctx 0x5575b9b4e0 mpi 0x7fa98d8af8
mpp[8567]: mpi: mpp_init enter ctx 0x5575b9b4e0 type 0 coding 0
mpp[8567]: mpp: unable to create unsupported type 0 coding 0
mpp[8567]: mpi: mpp_init leave ret -1
mpp[8567]: mpi_test: mpp_init failed
mpp[8567]: mpi: mpp_destroy enter ctx 0x5575b9b4e0
mpp[8567]: mpi: mpp_destroy leave ret 0
mpp[8567]: mpi_test: mpi_test failed
`

HermanChen, 2023-05-24:
用的什么参数，没定义输入格式