rockchip-linux/mpp #189: RV1126使用mpi_enc_test测试失败

guankang, 2021-04-01:
使用mpi_enc_test测试h264压缩，运行结果如下：
```
[root@RV1126_RV1109:/]# mpi_enc_test -i /mnt/sdcard/image_89.yuv -o /mnt/sdcard/xxx.h264 -w 1920 -h 1080 -f 0 -t 7 -n 1 -r 30

mpp[3546]: mpi_enc_utils: cmd parse result:
mpp[3546]: mpi_enc_utils: input  file name: /mnt/sdcard/image_89.yuv
mpp[3546]: mpi_enc_utils: output file name: /mnt/sdcard/xxx.h264
mpp[3546]: mpi_enc_utils: width      : 1920
mpp[3546]: mpi_enc_utils: height     : 1080
mpp[3546]: mpi_enc_utils: format     : 0
mpp[3546]: mpi_enc_utils: type       : 7
mpp[3546]: mpi_enc_test: mpi_enc_test start
mpp[3546]: mpp_rt: NOT found ion allocator
mpp[3546]: mpp_rt: NOT found drm allocator
mpp[3546]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[3546]: mpi_enc_test: 0x338c0 mpi_enc_test encoder test start w 1920 h 1080 type 7
mpp[3546]: mpp_enc: MPP_ENC_SET_RC_CFG bps 7776000 [486000 : 8262000] fps [30:30] gop 60
mpp[3546]: h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [1920:1080] stride [1920:1080]
mpp[3546]: mpp_enc: send header for set cfg change input/format 
mpp[3546]: mpp_enc: mode cbr bps [486000:7776000:8262000] fps fix [30/1] -> fix [30/1] gop i [60] v [0]
mpp[3546]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[3546]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 5 Input/output error
mpp[3546]: hal_h264e_vepu541: hal_h264e_vepu541_wait poll cmd failed 5
mpp[3546]: mpp_enc: mpp 0x33908 mpp_enc_hal_wait:1260 failed return -1009
mpp[3546]: mpp_enc: mpp 0x33908 mpp_enc_normal:1820 failed return -1009
mpp[3546]: mpi_enc_test: 0x338c0 encoded frame 0    size 154    
mpp[3546]: mpi_enc_test: 0x338c0 encode max 1 frames
mpp[3546]: mpi_enc_test: 0x338c0 mpi_enc_test success total frame 1 bps 36960
```
kernel报错：
```
[root@RV1126_RV1109:/]# dmesg 
[  799.681575] mpp_dma_import_fd:196: dma_buf_get fd 1 failed
[  799.681586] mpp_task_attach_fd:1301: can't import dma-buf 1
[  799.681594] mpp_translate_reg_address:1352: reg[ 80]: 0x00000001 fd 1 failed
[  799.681597] mpp_task_dump_mem_region:1529: --- dump mem region ---
[  799.681607] mpp_rkvenc ffbb0000.rkvenc: no memory region mapped
[  799.681611] mpp_task_dump_reg:1552: --- dump register ---
[  799.681615] mpp_process_task:362: alloc_task failed.
[  801.758936] mpp_wait_result:664: session b44cad39 pending list is empty!
```

系统信息：
```
[root@RV1126_RV1109:/]# uname -srm
Linux 4.19.111 armv7l
```

运行mpi_test日志如下：
```
[root@RV1126_RV1109:/]# mpi_test 
mpp[1215]: mpi_test: mpi_test start
mpp[1215]: mpi_test: mpi_test decoder test start
mpp[1215]: mpi: mpp_create enter ctx 0xaefafc6c mpi 0xaefafc70
mpp[1215]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[1215]: mpi: mpp_create leave ret 0 ctx 0x28f80 mpi 0xa6f5d31c
mpp[1215]: mpi: mpp_init enter ctx 0x28f80 type 0 coding 0
mpp[1215]: mpp: unable to create dec unused for soc rv1126 unsupported
mpp[1215]: mpi: mpp_init leave ret -1
mpp[1215]: mpi_test: mpp_init failed
mpp[1215]: mpi: mpp_destroy enter ctx 0x28f80
mpp[1215]: mpi: mpp_destroy leave ret 0
mpp[1215]: mpi_test: mpi_test failed
```

请问是因为目前mpp还不支持rv1126的原因么？

Consti10, 2021-04-14:
Any updates ? Were you able to workaround the issue ?

HermanChen, 2021-04-14:
mpi_test 这个测试没有在维护中，使用 mpi_dec_test 和 mpi_enc_test 来测试吧
The mpi_test is not maintained any more. Please use mpi_dec_test and mpi_enc_test to run the test.

HermanChen, 2021-04-14:
> 使用mpi_enc_test测试h264压缩，运行结果如下：
>
> ```
> [root@RV1126_RV1109:/]# mpi_enc_test -i /mnt/sdcard/image_89.yuv -o /mnt/sdcard/xxx.h264 -w 1920 -h 1080 -f 0 -t 7 -n 1 -r 30
>
> mpp[3546]: mpi_enc_utils: cmd parse [...]

mpp[3546]: mpp_rt: NOT found ion allocator
mpp[3546]: mpp_rt: NOT found drm allocator
内核没开 drm 分配器，这样 mpp 得不到硬件用内存的分配器

Consti10, 2021-04-14:
what happens when -n is not 1 (e.g. in this run it looks as if there are some errors, but in the end it works)

HermanChen, 2021-04-15:
-n is the frame number to decode or encode.

Consti10, 2021-04-15:
I know. I am just curios if encoding more than one frame at a time works.

Consti10, 2021-04-19:
What image are you running on your rv1126 ? I can't find any references to mpp on mine, only the "linux media server" application (which has a ton of latency). 
However, I'd assume that the "linux media server" application is actually built on top of mpp, right ?

liuyuan000, 2023-01-15:
请问解决了吗