rockchip-linux/mpp #190: mpp h264编码nalu type问题

MinZhaoLin, 2021-04-06:
基于RK3399芯片进行H264硬编码，设置GOP为30的时候，每30帧中，有一帧的nalu起始为00 00 00 01 06 05，其余帧nalu起始为 00 00 00 01 21 e0 ，请问这个只有一帧的00 00 00 01 06 05，按照GOP30应该理解为IDR帧，但是IDR为00 00 00 01 65才对。按照正常的nalu type那就是属于SEI，但这样就没有了IDR帧。请问这个nalu type需要怎么解析？

HermanChen, 2021-04-07:
找个 elecard 工具看一下，或者在 ffmpeg 里加点打印解码看看
目前的编码器会在 IDR 帧前面加上 sps/pps/sei 包

MinZhaoLin, 2021-04-07:
谢谢这个工具的分享，目前已知这一个包是SEI以00 00 00 01 06 05开头，实际上包含了两个SEI和一个IDR的nalu，目前知道如何解析了。但是现在遇到一个新的问题，我用下面两种方式写下来的h264裸流通过VLC只能看到第一帧图像，后续图像不会动：
1. SPS PPS SEI SEI IDR Px29，SPS PPS SEI SEI IDR Px29，SPS PPS SEI SEI IDR Px29...
2. SPS PPS SEI SEI IDR Px29，SEI SEI IDR Px29，SEI SEI IDR Px29...

HermanChen, 2021-04-08:
可以先用 ffmpeg 解码看下是否录制的有异常，也可以录一小段异常码流放上来看下具体是什么原因。

closed 2021-05-11 by HermanChen

zhang-chuanfeng, 2023-09-08:
> 

你好  你那边怎么解决的  我也发现没找到IDR帧