rockchip-linux/mpp #193: Clarification: Used VEPU on RV1126 and RV1109

Consti10, 2021-04-18:
Hello,
I'd assume that RV1126 uses  VPU_CLIENT_VEPU2 and RV1109 uses  VPU_CLIENT_VEPU2_LITE.

Is that right ?

Consti10, 2021-04-18:
found it:
RV1126 is VEPU2 & VDPU2
https://github.com/rockchip-linux/mpp/blob/a86f586b025474996a78820f0f0e57644c28e328/osal/mpp_soc.cpp#L540

And RV1109 is also VEPU2 & VDPU2:
https://github.com/rockchip-linux/mpp/blob/a86f586b025474996a78820f0f0e57644c28e328/osal/mpp_soc.cpp#L530

Don't know why the max supported res@framerate is different for them though.

HermanChen, 2021-04-19:
Yes, you are so careful about details. That is great : ) 

RV1109 and RV1126 have jpeg only VEPU2 and VDPU2. The hardware is vdpu2_jpeg and vepu2_jpeg and the MppDecHwCap  has specify the real hardware.
But The client type only has HAVE_VEPU2 / HAVE_VDPU2 flag. This flag is for register file size and context switch and signal hardware driver.

closed 2021-04-21 by Consti10