rockchip-linux/mpp #195: mpp and rockchip "media server"

Consti10, 2021-04-21:
Hello,
My rv1126 came with an application called "media server".
Unfortunately, this application has a really high end-to-end latency. (in the range of 1s while we aim for <70ms).
Therefore, I want to build a "raspivid" - style program for rv1126.

I am wondering, 
a) is the source code of "media server" available and
b) is "media server" built on top of mpp or something else ?

HermanChen, 2021-04-22:
media server is on top of mpp
You can use mpp to build base codec layer.

Consti10, 2021-04-22:
thanks.
Is it open or closed source (media server) ?

HermanChen, 2021-05-11:
The media server layer is rkmedia. It is also open source.

Consti10, 2021-05-11:
Thanks. For those coming across the issue, here is the code:
https://github.com/rockchip-linux/rkmedia

closed 2021-05-11 by Consti10