rockchip-linux/mpp #196: [Pure external mode] can't display

chllcy, 2021-04-28:
I'm use Pure external mode from this demo,it can decode well, but the screen can't see anything,with modetest the screen display well. so is there anything wrong?

[https://github.com/rockchip-linux/mpp/issues/14](url)
[mpi_dec.zip](https://github.com/rockchip-linux/mpp/files/6389686/mpi_dec.zip)

```
./mpi_dec Tennis1080p.h264 7
^[[CCONNECTOR type: 10 id: 91
plane_id 58 crtc_id 64
mpp[11636]: mpp_info: mpp version: d9886b79 author: Caesar Wang   2020-12-08 [hal_h264d_vdpu]: add extern define for these functions
mpp[11636]: mpp_rt: NOT found ion allocator
mpp[11636]: mpp_rt: found drm allocator
mpp[11636]: mpp: deprecated block control, use timeout control instead
DISPLAY THREAD START
FRAME THREAD START
PACKET SEND 4922.387738 S 1 V 0
PACKET SEND 4922.412911 S 1 V 1
frame changed 1920(1920)x1080(1088)
PACKET SEND 4922.416219 S 2 V 1
PACKET SEND 4922.419416 S 3 V 1
PACKET SEND 4922.422710 S 4 V 1
FRAME 4922.521397 first        0 previous      0 frid 0
FRAME 4922.533485 first    12088 previous  12088 frid 1
FRAME 4922.549500 first    28102 previous  16014 frid 2
PACKET SEND 4922.553737 S 4 V 4
FRAME 4922.560797 first    39399 previous  11296 frid 3
FRAME 4922.572107 first    50709 previous  11309 frid 4
FRAME 4922.581917 first    60519 previous   9810 frid 5
FRAME 4922.588268 first    66871 previous   6351 frid 6
FRAME 4922.593863 first    72465 previous   5594 frid 7
PACKET SEND 4922.599359 S 4 V 4
FRAME 4922.599853 first    78455 previous   5990 frid 8
FRAME 4922.606243 first    84846 previous   6390 frid 9
FRAME 4922.611738 first    90341 previous   5494 frid 10
FRAME 4922.618055 first    96657 previous   6316 frid 11
FRAME 4922.624598 first   103200 previous   6542 frid 12
FRAME 4922.631038 first   109641 previous   6440 frid 13
...
```

HermanChen, 2021-05-11:
add -o option to mpi_dec_test and the output yuv can be displayed correctly.

HermanChen, 2021-05-11:
杭州的哥们直接中文吧：）
如果是用纯外部模式给 mpp 导入 buffer 的，因为导入的 buffer fd 在 mpp 内部有 dup 一次，需要看下输出的 MppBuffer 的 fd 是原始的还是层入后的新 fd，和显示框架要做下匹配，或者在 commit buffer 之后，把新旧 fd 做一个索引，用原始 fd 去送给显示