rockchip-linux/mpp #197: ubuntu 18.04 can not compile libs for aarch64 android  platform

wkfwnn, 2021-04-29:
-- cmake version 3.10.2
CMake Deprecation Warning at /usr/share/cmake-3.10/Modules/CMakeForceCompiler.cmake:69 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  build/android/android.toolchain.cmake:1132 (CMAKE_FORCE_C_COMPILER)
  build/android/CMakeFiles/3.10.2/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:30 (project)

CMake Deprecation Warning at /usr/share/cmake-3.10/Modules/CMakeForceCompiler.cmake:83 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  build/android/android.toolchain.cmake:1145 (CMAKE_FORCE_CXX_COMPILER)
  build/android/CMakeFiles/3.10.2/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:30 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
-- Configuring incomplete, errors occurred!
See also "/home/carl/work/boeos/master/BoePlayer/android/contrib/ffmpeg-arm64/mpp/build/android/CMakeFiles/CMakeOutput.log"

I guess the cmake version is not right.
Here is my cli:cmake -DCMAKE_TOOLCHAIN_FILE=android.toolchain.cmake -DANDROID_NDK=/home/carl/Android/Sdk/ndk/android-ndk-r10e -DCMAKE_BUILD_TYPE=Release -DANDROID_ABI="arm64-v8a"  ../../ && cmake --build .

wkfwnn, 2021-04-29:
I am using latest version release_20171218

HermanChen, 2021-05-11:
Please try develop branch at https://github.com/HermanChen/mpp

LeafHCode, 2023-08-28:
change to cmake 3.16.0 is ok

closed 2023-08-28 by HermanChen

LeafHCode, 2023-08-29:
直接进入mpp/build/android/aarch64修改make-Android.bash里的ANDROID_NDK，CMAKE_PROGRAM=cmake后执行bash CMAKE_PROGRAM，高版本ndk可能还需要注释-DANDROID_STL=${ANDROID_STL}