rockchip-linux/mpp #199: 使用mpi_enc_test编码fill_image()函数生成的yuv数据有残影

imxys, 2021-05-13:
以前用的是编码器库是编码参数还没有换成使用字符串配置`MppEncCfg`，`rk_mpi_cmd.h`里还有提供`MPP_ENC_SET_ALL_CFG`的版本，当时的测试程序编码效果将生成的yuv文件：
[image]
编码成了下图：
[image]
后来发现了github存在比较新的mpp库代码，测试之后虽然没有原来的边界花现象，但是存在残影：
[image]
如果把码率改为2Mbps，qp也做调整，画面则也存在边界花的现象：
[image]

调整的参数和运行记录如下
```shell
mpp[30470]: mpi_enc_utils: cmd parse result:
mpp[30470]: mpi_enc_utils: input  file name: (null)
mpp[30470]: mpi_enc_utils: output file name: /tmp/....h264
mpp[30470]: mpi_enc_utils: width      : 1920
mpp[30470]: mpi_enc_utils: height     : 1080
mpp[30470]: mpi_enc_utils: format     : 0
mpp[30470]: mpi_enc_utils: type       : 7
mpp[30470]: mpi_enc_test: mpi_enc_test start
mpp[30470]: mpp_rt: NOT found ion allocator
mpp[30470]: mpp_rt: found drm allocator
mpp[30470]: mpp_info: mpp version: 5b1156a4 author: Yandong Lin   2021-04-09 [vcodec_service]: Fix array out of bounds issue
mpp[30470]: mpi_enc_test: 0x286a37c0 mpi_enc_test encoder test start w 1920 h 1080 type 7
mpp[30470]: mpp_enc: MPP_ENC_SET_RC_CFG bps 2000000 [125000 : 2125000] fps [30:30] gop 60
mpp[30470]: h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [1920:1080] stride [1920:1080]
mpp[30470]: mpp_enc: send header for set cfg change input/format 
mpp[30470]: mpp_enc: mode vbr bps [125000:2000000:2125000] fps fix [30/1] -> fix [30/1] gop i [60] v [0]
test_mpp_enc_cfg_setup get h264:profile: 100
test_mpp_enc_cfg_setup get h264:cabac_en: 1
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 0    size 30358   qp 36
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 1    size 8792    qp 36
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 2    size 7997    qp 36
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 3    size 6847    qp 36
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 4    size 6416    qp 36
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 5    size 8924    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 6    size 8140    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 7    size 6325    qp 36
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 8    size 7411    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 9    size 6842    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 10   size 7223    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 11   size 6233    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 12   size 6908    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 13   size 6856    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 14   size 6558    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 15   size 7138    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 16   size 7061    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 17   size 7852    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 18   size 6937    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 19   size 6500    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 20   size 7296    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 21   size 7165    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 22   size 7057    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 23   size 6882    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 24   size 6377    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 25   size 7133    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 26   size 7229    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 27   size 6459    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 28   size 7286    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 29   size 7381    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 30   size 7166    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 31   size 7295    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 32   size 6825    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 33   size 6819    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 34   size 7719    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 35   size 6423    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 36   size 6801    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 37   size 7841    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 38   size 6962    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 39   size 7673    qp 34
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 40   size 6503    qp 34
[... 145 lines cut ...]
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 186  size 8113    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 187  size 6757    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 188  size 7820    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 189  size 8466    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 190  size 7188    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 191  size 8020    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 192  size 7915    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 193  size 6304    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 194  size 6564    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 195  size 6192    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 196  size 6268    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 197  size 7091    qp 34
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 198  size 6387    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 199  size 6959    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 200  size 6963    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 201  size 6601    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 202  size 7080    qp 35
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 203  size 7367    qp 34
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 204  size 7179    qp 34
mpp[30470]: mpi_enc_test: 0x286a37c0 encoded frame 205  size 7663    qp 34
^C
```

我想问的问题是，这应该是修改mpp库能解决的问题吧？我在电脑上用ffmpeg，进行x264编码，不使用B帧，码率设置2Mbps，画面主要是色块模糊没有边界花掉：
[image]

HermanChen, 2021-05-14:
这个是 3399 硬件编码器对这种色度分量步进移动的特定场景编码有一些缺陷，是芯片的正常现象，在一般场景应用下不会有这个问题。
这个问题在新的硬件编码器上已经修正

imxys, 2021-05-14:
好的

另外我尝试了一下反馈里，步进的边界没有留下马赛克，只是有一点点影子的情况
实际是，新的mpp库更改了编译mpi_enc_test的条件，从一定编译测试程序变成要根据BUILD_TEST编译
造成我在使用github上的Mpp库的时候，使用的测试程序不配套，仍在使用旧的mpi_enc_test

所以编译出来画质特别好，码流没有控制，边界只有一点影子，对应我发的第三张图；
后来我把测试程序编译出来，github的mpp库对应github的测试程序，能够控制profile和码率了，然后边界的马赛克又出现了，对应我发的第四张图。

closed 2021-05-17 by imxys