rockchip-linux/mpp #200: H264/H265 encoding latency

Consti10, 2021-05-16:
Hello,
I am wondering if it is possible to measure the encoding latency (delta between frame N in -> frame N out)
by just measuring the time it takes until
MppEncoder::process() 
returns.

Or is the following possible with rv1126 encoder hw:
put in frame 0
-> get out nothing
put in frame 1
-> get out frame 0
put in frame 2
-> get out frame 1

(e.g. encoder buffers 1 or more frames).

Because on the rpi encoder for example, the encoding process is pipelined.

HermanChen, 2021-05-17:
To show task timing setprop mpp_debug 1 or export mpp_debug=1.
The encoder put/get function is block mode only. The non-block mode need to notify when the input buffer is at end of usage.

Consti10, 2021-05-17:
Is that the same as doing:
 echo 0x100 > /sys/modules/rk_vcodec/parameters/debug
 As described here:
 https://github.com/rockchip-linux/mpp/issues/154
 ?
 After doing so, I think I get the encoding latency from dmesg:

[image]

 Can you do the same for the isp ? e.g. processing time for a frame on isp ?

HermanChen, 2021-05-18:
The isp part is not in mpp

Consti10, 2021-05-18:
But rk_vcodec is neither, right ?

HermanChen, 2021-05-19:
rk_vcodec is part of mpp.
The timing is the hardware encoding time for each encoding task.
The latency contain hardware working time and software time.

https://github.com/rockchip-linux/mpp/blob/5b1156a47bc81462ce5d7c600d5a15b338dd4d82/mpp/mpp.cpp#L453-L456
Here you can enable the total timing record when mpp encoding one frame.

Consti10, 2021-05-20:
thanks. Can you point me in the right direction regarding isp or do you not know how ?