rockchip-linux/mpp #201: Failed to set encoder config like "h264:poc_type"

imxys, 2021-05-17:
I'm just trying to set some encoder configs. While setting "h264:poc_type", mpp warns about type incorrect.
I know poc_type in MppEncH264Cfg_t is RK_U8.
How to set config like this?

Declaration:
```C++
    ENTRY(h264, poc_type,       U32, RK_U32,            MPP_ENC_H264_CFG_CHANGE_POC_TYPE,       codec.h264, poc_type) \
    ENTRY(h264, log2_max_poc_lsb,   U32, RK_U32,        MPP_ENC_H264_CFG_CHANGE_MAX_POC_LSB,    codec.h264, log2_max_poc_lsb) \
    ENTRY(h264, log2_max_frm_num,   U32, RK_U32,        MPP_ENC_H264_CFG_CHANGE_MAX_FRM_NUM,    codec.h264, log2_max_frame_num) \
```

My code:
```C++
    mpp_enc_cfg_set_u32(cfg, "h264:poc_type", 0);
    mpp_enc_cfg_set_u32(cfg, "h264:log2_max_poc_lsb", 4);
    mpp_enc_cfg_set_u32(cfg, "h264:log2_max_frame_num", 3);
```

Log:
```shell
mpp[16915]: mpp_cfg: cfg h264:poc_type expect RK_U32 input NOT RK_U32
mpp[16915]: mpp_cfg: cfg h264:log2_max_poc_lsb expect RK_U32 input NOT RK_U32
mpp[16915]: mpp_enc_cfg: mpp_enc_cfg_set_u32 failed to set h264:log2_max_frame_num to 3
```

```shell
mpp[16915]: mpp_cfg: cfg h264:poc_type expect RK_U32 input NOT RK_U32
mpp[16915]: mpi_enc_test: h264:poc_type: 0
mpp[16915]: mpp_cfg: cfg h264:log2_max_poc_lsb expect RK_U32 input NOT RK_U32
mpp[16915]: mpi_enc_test: h264:log2_max_poc_lsb: 0
mpp[16915]: mpp_cfg: cfg h264:log2_max_frm_num expect RK_U32 input NOT RK_U32
mpp[16915]: mpi_enc_test: h264:log2_max_frm_num: 0
```

FumasterLin, 2021-05-21:
Hi：

Please apply the diff and test again.

```diff
diff --git a/inc/rk_venc_cmd.h b/inc/rk_venc_cmd.h
index a2453c0e..cfe5fd09 100644
--- a/inc/rk_venc_cmd.h
+++ b/inc/rk_venc_cmd.h
@@ -606,9 +606,9 @@ typedef struct MppEncH264Cfg_t {
      * log2_max_frame_num   - used in sps
      */
     RK_U8               svc;
-    RK_U8               poc_type;
-    RK_U8               log2_max_poc_lsb;
-    RK_U8               log2_max_frame_num;
+    RK_U32              poc_type;
+    RK_U32              log2_max_poc_lsb;
+    RK_U32              log2_max_frame_num;
     RK_U32              gaps_not_allowed;
```

imxys, 2021-05-21:
it works, with no warning:
```shell
mpp[5324]: mpi_enc_test: h264:poc_type: 2
mpp[5324]: mpi_enc_test: h264:log2_max_poc_lsb: 12
mpp[5324]: mpi_enc_test: h264:log2_max_frm_num: 12
```
thanks,  there's 4 more config warns the same question and I didn't wrote here:
```shell
mpp[30756]: mpp_cfg: cfg h264:max_tid expect RK_S32 input NOT RK_S32
mpp[30756]: mpi_enc_test: h264:max_tid: 2
mpp[30756]: mpp_cfg: cfg h264:max_ltr expect RK_S32 input NOT RK_S32
mpp[30756]: mpi_enc_test: h264:max_ltr: 2
mpp[30756]: mpp_cfg: cfg h264:prefix_mode expect RK_S32 input NOT RK_S32
mpp[30756]: mpi_enc_test: h264:prefix_mode: 2
mpp[30756]: mpp_cfg: cfg h264:base_layer_pid expect RK_S32 input NOT RK_S32
mpp[30756]: mpi_enc_test: h264:base_layer_pid: 2
```

FumasterLin, 2021-05-21:
Hi:

```diff
diff --git a/inc/rk_venc_cmd.h b/inc/rk_venc_cmd.h
index 367fcf5d..9f0f6ef3 100644
--- a/inc/rk_venc_cmd.h
+++ b/inc/rk_venc_cmd.h
@@ -718,10 +718,10 @@ typedef struct MppEncH264Cfg_t {
     RK_S32              intra_refresh_arg;
 
     /* extra mode config */
-    RK_S16              max_ltr_frames;
-    RK_S16              max_tid;
-    RK_S16              prefix_mode;
-    RK_S16              base_layer_pid;
+    RK_S32              max_ltr_frames;
+    RK_S32              max_tid;
+    RK_S32              prefix_mode;
+    RK_S32              base_layer_pid;
 } MppEncH264Cfg;
 
 #define H265E_MAX_ROI_NUMBER  64
```

imxys, 2021-05-21:
Test passed.
```shell
mpp[6704]: mpi_enc_test: h264:max_tid: 0
mpp[6704]: mpi_enc_test: h264:max_ltr: 0
mpp[6704]: mpi_enc_test: h264:prefix_mode: 0
mpp[6704]: mpi_enc_test: h264:base_layer_pid: 0
mpp[6704]: mpi_enc_test: hw:aq_thrd_i: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
mpp[6704]: mpi_enc_test: hw:aq_thrd_p: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
mpp[6704]: mpi_enc_test: hw:aq_step_i: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
mpp[6704]: mpi_enc_test: hw:aq_step_p: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
```
As you can see above, I also tried to print array with `mpp_enc_cfg_get_st()` and there's no  warning.

Thanks again. I think it's ok to close this issue after these modifications commited.

Consti10, 2021-09-08:
Which chip were you using for your tests ? E.g. rv1126 or rk3368 or ... ?

FumasterLin, 2021-09-09:
The #201 problem has been resolved, your new problem will be further tracked in #236.

imxys, 2021-09-09:
> 
> 
> Which chip were you using for your tests ? E.g. rv1126 or rk3368 or ... ?

RK3399.

Consti10, 2021-09-09:
Doesn't rk3399 use poc_type=2 by default ?
Since https://github.com/rockchip-linux/mpp/blob/develop/mpp/codec/enc/h264/h264e_api_v2.c#L105

( I think rk3399 uses VPU_CLIENT_VEPU2 (poc_type=2) while rv1126 in contrast uses VPU_CLIENT_RKVENC (poc_type=0).

imxys, 2021-09-10:
> 
> 
> Doesn't rk3399 use poc_type=2 by default ?
> Since https://github.com/rockchip-linux/mpp/blob/develop/mpp/codec/enc/h264/h264e_api_v2.c#L105
> 
> ( I think rk3399 uses VPU_CLIENT_VEPU2 (poc_type=2) while rv1126 in contrast uses VPU_CLIENT_RKVENC (poc_type=0).
> 
> For reference, rk3399: https://github.com/rockchip-linux/mpp/blob/develop/osal/mpp_soc.cpp#L455
> And rv1126: https://github.com/rockchip-linux/mpp/blob/develop/osal/mpp_soc.cpp#L540
> ( watch out, rk3399 for some reason has "HAVE_RKVDEC" but not "HAVE_RKVENC")

I think you are right... These marco are about SOC's feature.
I thought I was trying to change the param to encode stream with less noise at low bitrate, and failed.