rockchip-linux/mpp #202: transform8x8 模式如何生效

imxys, 2021-05-18:
Platform: RK3399

我使用同一个yuv文件做编码测试。

1. 当我使用`ffmpeg`命令将文件编码为High Profile的文件的时候，通过Vega H264 Analyzer 软件看到文件内PPS有transform_8x8_mode_flag=true，画面I帧的宏块往往分割成8x8宏块，偶尔也看到分割成16个4x4宏块。类似下图

[image]

2. 当我使用`ffmpeg`命令将文件编码为High Profile的文件，但是使用`-8x8dct 0`关闭`transform_8x8`的时候，通过Vega H264 Analyzer 软件看到文件内PPS没有transform_8x8_mode_flag，画面I帧的宏块只会分割成16个4x4宏块。此外，即使我有指定Profile为HP，ffmpeg也编码成Main Profile的文件。类似下图：

[image]

根据1和2两点我认为RK的编码配置`h264:trans8x8`应该在宏块分割上有同样的效果。但是根据demo编码视频的分析，不管我有没有使能这个配置（设置1或0），好像所有帧内刷新类型的宏块都是被分割成4x4的。类似下图：

[image]

请问我理解错了吗？（尝试用英文提问感觉太复杂于是用了中文）

HermanChen, 2021-05-19:
8x8 模式需要配置了 high profile 才生效，默认是 baseline，看下编译出来的码流是不是 high profile 的

imxys, 2021-05-20:
> 8x8 模式需要配置了 high profile 才生效，默认是 baseline，看下编译出来的码流是不是 high profile 的

截图里有信息，确实是high profile的

HermanChen, 2021-05-21:
配置了 8x8 变换，并不是说所有块都使用 8x8 变换，这个是编码器内核根据不同模式的代价，针对不同内容块判决出不同最优的情况，开了 8x8，应该是会看有 16x16，8x8，4x4 都会存在的情况

imxys, 2021-05-21:
> 配置了 8x8 变换，并不是说所有块都使用 8x8 变换，这个是编码器内核根据不同模式的代价，针对不同内容块判决出不同最优的情况，开了 8x8，应该是会看有 16x16，8x8，4x4 都会存在的情况

按道理来讲是的，我通过电脑上ffmpeg编译验证也是这个现象（见前两张截图）；但不知为何我RK3399这边都没有见到过8x8的

Consti10, 2021-09-03:
Do you have a download link for  Vega H264 Analyzer software ?

imxys, 2021-09-03:
> 
> 
> Do you have a download link for Vega H264 Analyzer software ?

Oh, sorry to say I don't know where to download it either because I got it from my ex-colleague...
Maybe you can try the trail version at [it's official site](https://www.interrasystems.com/vega-analyzer.php),
or try opensource alternatives: [latelee/H264BSAnalyzer](https://github.com/latelee/H264BSAnalyzer), [leixiaohua1020/VideoEye](https://github.com/leixiaohua1020/VideoEye) and [leixiaohua1020/h264_analysis](https://github.com/leixiaohua1020/h264_analysis)