rockchip-linux/mpp #206: rk3288,运行mpi_dec_test失败

luocha, 2021-05-27:
firefly@firefly:~/new/mpp-release/osal/test$ ./mpp_platform_test 
mpp[27920]: mpp_plat_test: chip name: rockchip,rk3288w
mpp[27920]: mpp_plat_test: 
mpp[27920]: mpp_plat_test: chip vcodec type 00020101
mpp[27920]: mpp_plat_test: found vpu1 codec
mpp[27920]: mpp_plat_test: found rk hevc decoder
mpp[27920]: mpp_plat_test: 
mpp[27920]: mpp_plat_test: start probing decoder device name:
mpp[27920]: mpp_plat_test: H.264 decoder: /dev/vpu-service
mpp[27920]: mpp_plat_test: H.265 decoder: /dev/hevc-service
mpp[27920]: mpp_plat_test: MJPEG decoder: /dev/vpu-service
mpp[27920]: mpp_plat_test: VP9   decoder: /dev/vpu-service
mpp[27920]: mpp_plat_test: avs   decoder: /dev/vpu-service
mpp[27920]: mpp_plat_test: 
mpp[27920]: mpp_plat_test: start probing encoder device name:
mpp[27920]: mpp_plat_test: H.264 encoder: /dev/vpu-service
mpp[27920]: mpp_plat_test: H.265 encoder: /dev/vpu-service
mpp[27920]: mpp_plat_test: MJPEG encoder: /dev/vpu-service
mpp[27920]: mpp_plat_test: mpp platform test done

firefly@firefly:~/new/mpp-release/test$ ./mpi_dec_test -i ~/test/desktop.h264 
mpp[27922]: mpi_dec_test: cmd parse result:
mpp[27922]: mpi_dec_test: input  file name: /home/firefly/test/desktop.h264
mpp[27922]: mpi_dec_test: output file name: 
mpp[27922]: mpi_dec_test: config file name: 
mpp[27922]: mpi_dec_test: width      :    0
mpp[27922]: mpi_dec_test: height     :    0
mpp[27922]: mpi_dec_test: type       : 7
mpp[27922]: mpi_dec_test: debug flag : 0
mpp[27922]: mpi_dec_test: max frames : 0
mpp[27922]: mpi_dec_test: mpi_dec_test start
mpp[27922]: mpi_dec_test: input file size 1492093
mpp[27922]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[27922]: mpi_dec_test: 0x811a64a0 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[27922]: mpp_rt: NOT found ion allocator
mpp[27922]: mpp_rt: found drm allocator
mpp[27922]: mpi_dec_test: 0x811a64a0 decode_get_frame get info changed found
mpp[27922]: mpi_dec_test: 0x811a64a0 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 3760128
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 0 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 1 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 2 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 3 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 4 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 5 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 6 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 7 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 8 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 9 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 10 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 11 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 12 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 13 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 14 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 15 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 16 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 17 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 18 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 19 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 20 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 21 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 22 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 23 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 24 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 25 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 26 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 27 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 28 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 29 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 30 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 31 err 1 discard 0
[... 84 lines cut ...]
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 116 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 117 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 118 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 119 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 120 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 121 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 122 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 123 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 124 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 125 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 126 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 127 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 128 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 129 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 found last packet
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 130 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 131 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 132 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 133 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 decode get frame 134 err 1 discard 0
mpp[27922]: mpi_dec_test: 0x811a64a0 found last packet
mpp[27922]: mpi_dec_test: 0x811a64a0 input 365 pkt output 136 frm decode 135 frames
mpp[27922]: mpi_dec_test: test success max memory 14.34 MB

可以看出来，一直是 err 1，这是什么原因呢

HermanChen, 2021-05-29:
看下内核有没有什么打印，估计是内核驱动和 mpp 不匹配

luocha, 2021-05-31:
> 看下内核有没有什么打印，估计是内核驱动和 mpp 不匹配

[  733.670828] rk_iommu ff9a0800.iommu: Page fault at 0x00000ee0 of type read
[  733.677737] rk_iommu ff9a0800.iommu: iova = 0x00000ee0: dte_index: 0x0 pte_index: 0x0 page_offset: 0xee0
[  733.687235] rk_iommu ff9a0800.iommu: mmu_dte_addr: 0x2d17b000 dte@0x2d17b000: 0x000000 valid: 0 pte@0x00000000: 0x000000 valid: 0 page@0x00000000 flags: 0x0
[  733.701223] rk-vcodec ff9a0000.vpu-service: vcodec, fault addr 0x00000ee0 status 20b
[  733.708967] rk-vcodec ff9a0000.vpu-service: no memory region mapped
[  733.715237] rk-vcodec ff9a0000.vpu-service: dumping dec register set:
[  733.721681] rk-vcodec ff9a0000.vpu-service: reg[00] 67313688
[  733.727350] rk-vcodec ff9a0000.vpu-service: reg[01] 00040100
[  733.733047] rk-vcodec ff9a0000.vpu-service: reg[02] fff80510
[  733.738753] rk-vcodec ff9a0000.vpu-service: reg[03] 00081201
[  733.744462] rk-vcodec ff9a0000.vpu-service: reg[04] 3c022001
[  733.750169] rk-vcodec ff9a0000.vpu-service: reg[05] 00000000
[  733.755877] rk-vcodec ff9a0000.vpu-service: reg[06] ae025a50
[  733.761585] rk-vcodec ff9a0000.vpu-service: reg[07] 20040000
[  733.767293] rk-vcodec ff9a0000.vpu-service: reg[08] 40050000
[  733.773002] rk-vcodec ff9a0000.vpu-service: reg[09] 00084000
[  733.778709] rk-vcodec ff9a0000.vpu-service: reg[10] 128398a4
[  733.784419] rk-vcodec ff9a0000.vpu-service: reg[11] 1ee6b16a
[  733.790124] rk-vcodec ff9a0000.vpu-service: reg[12] 00000000
[  733.795831] rk-vcodec ff9a0000.vpu-service: reg[13] 00000000
[  733.801537] rk-vcodec ff9a0000.vpu-service: reg[14] 00000000
[  733.807246] rk-vcodec ff9a0000.vpu-service: reg[15] 00000000
[  733.812961] rk-vcodec ff9a0000.vpu-service: reg[16] 00000000
[  733.818669] rk-vcodec ff9a0000.vpu-service: reg[17] 00000000
[  733.824378] rk-vcodec ff9a0000.vpu-service: reg[18] 00000000
[  733.830082] rk-vcodec ff9a0000.vpu-service: reg[19] 00000000
[  733.835790] rk-vcodec ff9a0000.vpu-service: reg[20] 00000000
[  733.841499] rk-vcodec ff9a0000.vpu-service: reg[21] 00000000
[  733.847206] rk-vcodec ff9a0000.vpu-service: reg[22] 00000000
[  733.852910] rk-vcodec ff9a0000.vpu-service: reg[23] 00000000
[  733.858617] rk-vcodec ff9a0000.vpu-service: reg[24] 00000000
[  733.864324] rk-vcodec ff9a0000.vpu-service: reg[25] 00000000
[  733.870029] rk-vcodec ff9a0000.vpu-service: reg[26] 00000000
[  733.875736] rk-vcodec ff9a0000.vpu-service: reg[27] 00000000
[  733.881442] rk-vcodec ff9a0000.vpu-service: reg[28] 00000000
[  733.887150] rk-vcodec ff9a0000.vpu-service: reg[29] 00000000
[  733.892856] rk-vcodec ff9a0000.vpu-service: reg[30] 00000000
[  733.898563] rk-vcodec ff9a0000.vpu-service: reg[31] 00000000
[  733.904271] rk-vcodec ff9a0000.vpu-service: reg[32] 00000000
[  733.909978] rk-vcodec ff9a0000.vpu-service: reg[33] 00000000
[  733.915686] rk-vcodec ff9a0000.vpu-service: reg[34] 00000000
[  733.921391] rk-vcodec ff9a0000.vpu-service: reg[35] 00000000
[  733.927100] rk-vcodec ff9a0000.vpu-service: reg[36] 00000000
[  733.932808] rk-vcodec ff9a0000.vpu-service: reg[37] 00000000
[  733.938514] rk-vcodec ff9a0000.vpu-service: reg[38] 00000000
[  733.944222] rk-vcodec ff9a0000.vpu-service: reg[39] 00000000
[  733.949929] rk-vcodec ff9a0000.vpu-service: reg[40] 00000000
[  733.955638] rk-vcodec ff9a0000.vpu-service: reg[41] 0bf40000
[  733.961344] rk-vcodec ff9a0000.vpu-service: reg[42] 04208400
[  733.967053] rk-vcodec ff9a0000.vpu-service: reg[43] 0a521063
[  733.972761] rk-vcodec ff9a0000.vpu-service: reg[44] 10839cc6
[  733.978471] rk-vcodec ff9a0000.vpu-service: reg[45] 16b52929
[  733.984178] rk-vcodec ff9a0000.vpu-service: reg[46] 1ce6b58c
[  733.989885] rk-vcodec ff9a0000.vpu-service: reg[47] 062081ef
[  733.995593] rk-vcodec ff9a0000.vpu-service: reg[48] 00000000
[  734.001300] rk-vcodec ff9a0000.vpu-service: reg[49] 007fb050
[  734.007010] rk-vcodec ff9a0000.vpu-service: reg[50] fbb56f80
[  734.012715] rk-vcodec ff9a0000.vpu-service: reg[51] 00000000
[  734.018422] rk-vcodec ff9a0000.vpu-service: reg[52] 00000000
[  734.024130] rk-vcodec ff9a0000.vpu-service: reg[53] 00000000
[  734.029838] rk-vcodec ff9a0000.vpu-service: reg[54] e5da0000
[  734.035546] rk-vcodec ff9a0000.vpu-service: reg[55] 00000008
[  734.041251] rk-vcodec ff9a0000.vpu-service: reg[56] 00000000
[  734.046959] rk-vcodec ff9a0000.vpu-service: reg[57] 000000de
[  734.052666] rk-vcodec ff9a0000.vpu-service: reg[58] 00000000
[  734.058381] rk-vcodec ff9a0000.vpu-service: reg[59] 00000000
[  734.064090] rk-vcodec ff9a0000.vpu-service: reg[60] 00000000
[  734.069797] rk-vcodec ff9a0000.vpu-service: reg[61] 01010100
[... 125 lines cut ...]
[  735.925832] rk-vcodec ff9a0000.vpu-service: reg[77]: 00000000
[  735.931611] rk-vcodec ff9a0000.vpu-service: reg[78]: 00000000
[  735.937376] rk-vcodec ff9a0000.vpu-service: reg[79]: 00000000
[  735.943160] rk-vcodec ff9a0000.vpu-service: reg[80]: 00000000
[  735.948928] rk-vcodec ff9a0000.vpu-service: reg[81]: 00000000
[  735.954687] rk-vcodec ff9a0000.vpu-service: reg[82]: 00000000
[  735.960464] rk-vcodec ff9a0000.vpu-service: reg[83]: 00000000
[  735.966223] rk-vcodec ff9a0000.vpu-service: reg[84]: 00000000
[  735.971990] rk-vcodec ff9a0000.vpu-service: reg[85]: 00000000
[  735.977788] rk-vcodec ff9a0000.vpu-service: reg[86]: 00000000
[  735.983538] rk-vcodec ff9a0000.vpu-service: reg[87]: 00000000
[  735.989317] rk-vcodec ff9a0000.vpu-service: reg[88]: 00000000
[  735.995079] rk-vcodec ff9a0000.vpu-service: reg[89]: 00000000
[... cut ...]
[  736.023941] rk-vcodec ff9a0000.vpu-service: reg[94]: 00000000
[  736.029713] rk-vcodec ff9a0000.vpu-service: reg[95]: 00000000
[  736.035465] rk-vcodec ff9a0000.vpu-service: reg[96]: 00000000
[  736.041253] rk-vcodec ff9a0000.vpu-service: reg[97]: 00000000
[  736.047011] rk-vcodec ff9a0000.vpu-service: reg[98]: 00000000
[  736.052787] rk-vcodec ff9a0000.vpu-service: reg[99]: ffffffff
[  736.058566] rk-vcodec ff9a0000.vpu-service: reg[100]: fc874780
[  738.057780] rk_vcodec: vpu_service_ioctl:1816: error: pid 1106 wait 0 task done timeout

luocha, 2021-05-31:
> 看下内核有没有什么打印，估计是内核驱动和 mpp 不匹配

板子是开发板，直接烧的ubuntu18.04的系统，驱动也是自带的，mpp是源码编译的，会存在不兼容的情况吗

firefly@firefly:~$ dpkg -l |grep rockchip
ii  gstreamer1.0-rockchip1                   1.14-1firefly1                         armhf        The Gstreamer plugins for Rockchip RK3xxx platform.
ii  gstreamer1.0-rockchip1-extra             20180115-1firefly2                     armhf        The Extra Gstreamer plugins for Rockchip platform.
ii  libdrm-rockchip1:armhf                   2.4.101-2~18.04.1firefly1              armhf        Userspace interface to msm/kgsl kernel DRM services -- runtime
ii  librockchip-mpp-dev                      1.4.0-1firefly1                        armhf        Media Process Platform
ii  librockchip-mpp-static                   1.4.0-1firefly1                        armhf        Media Process Platform
ii  librockchip-mpp1                         1.4.0-1firefly1                        armhf        Media Process Platform
ii  librockchip-vpu0                         1.4.0-1firefly1                        armhf        Media Process Platform

luocha, 2021-05-31:
解决了，cmake 编译的时候加上 -DHAVE_DRM=ON就可以解码了

closed 2021-05-31 by luocha