rockchip-linux/mpp #209: 请教一下解码器缓存的问题

0x6c0x62, 2021-05-28:
目前解码器默认会缓存几帧，也就是说前几帧输入是没有输出的。我希望每解码一帧，直接等待输出当前帧的结果，有什么资料可以参考吗？

HermanChen, 2021-05-29:
有个 MPP_DEC_SET_IMMEDIATE_OUT 的 control，可以让解码器以解码序输出

0x6c0x62, 2021-05-31:
你好，我按照您给的命令，调用control，报如下的错误：

mpp[14039]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[14039]: mpp_dec: mpp_dec_control found NULL input dec (nil)
mpp[14039]: mpp: command 31000c param 0x7f9b3cc864 ret -3
mpp[14039]: mpp_log: mpi->control failed
mpp[14039]: mpi: mpp_init found invalid context (nil)
mpp[14039]: mpp_log: mpp_init failed
=====ret: 0, packet length: 61882, type: 1
mpp[14039]: mpi: mpi_decode_put_packet found invalid context (nil)
mpp[14039]: mpi: mpi_decode_get_frame found invalid context (nil)
mpp[14039]: mpp_log: decode_get_frame failed ret -2
=====ret: 0, packet length: 68197, type: 1
mpp[14039]: mpi: mpi_decode_put_packet found invalid context (nil)
mpp[14039]: mpi: mpi_decode_get_frame found invalid context (nil)
mpp[14039]: mpp_log: decode_get_frame failed ret -2
=====ret: 0, packet length: 70935, type: 1
mpp[14039]: mpi: mpi_decode_put_packet found invalid context (nil)
mpp[14039]: mpi: mpi_decode_get_frame found invalid context (nil)
mpp[14039]: mpp_log: decode_get_frame failed ret -2
=====ret: 0, packet length: 64936, type: 1
mpp[14039]: mpi: mpi_decode_put_packet found invalid context (nil)
mpp[14039]: mpi: mpi_decode_get_frame found invalid context (nil)
mpp[14039]: mpp_log: decode_get_frame failed ret -2

下面是我的测试代码
    MPP_RET ret         = MPP_OK;
    MpiCmd mpi_cmd      = MPP_CMD_BASE;
    MppParam param      = nullptr;
    RK_U32 param_value   = 1;
    ret = mpp_create(&ctx, &mpi);

    if (MPP_OK != ret) {
        mpp_err("mpp_create failed\n");
        return -1;
    }

    mpi_cmd = MPP_DEC_SET_IMMEDIATE_OUT;
    param = &param_value;
    ret = mpi->control(ctx, mpi_cmd, param);
    if (MPP_OK != ret) {
        mpp_err("mpi->control failed\n");
        return -1;
    }

    ret = mpp_init(ctx, MPP_CTX_DEC, type);
    if (MPP_OK != ret) {
        mpp_err("mpp_init failed\n");
        return -1;
    }

0x6c0x62, 2021-05-31:
我尝试将mpi->control放到mpp_init之后，没有报错。
目前基本上能满足需求
我只解码关键帧，第一帧和第二帧关键帧送进解码器没有返回，送进去第三帧关键帧，会一起输出前三帧的解码数据，从第四帧开始就每一帧输入都有对应的输出了。不过调用reset接口之后，第一帧输入没有输出，从第二帧开始，每一帧输入都有对应的输出了。
感谢！

0x6c0x62, 2021-05-31:
感谢！

closed 2021-05-31 by 0x6c0x62