rockchip-linux/mpp #211: 1080@30fps在2Mbps码率下编码有残影，4Mbps时残影不明显

imxys, 2021-05-31:
测试场景：
在背景是白墙，或是没有复杂纹路的纯色背景的场景，可能有阴影；人物在画面中在离镜头2米的位置挥手，来回走动
录制好视频，电脑上转换出yuv文件后放到RK3399上编码，

测试编码结果会发现，1080@30fps，2Mbps（低码率）的编码下，看着画面好像还比较清晰（画面运动比较少的时候）
人走动起来会留下残影，在3Mbps或者4Mbps的情况下残影的现象就会减弱

希望能调整成运动起来没有残影的视频，模糊一点也没关系。
电脑上用ffmpeg软编码yuv文件，同样是2Mbps码流，没有残影。

我想看看有没有调整运动检测或是去噪相关的参数可以使用，请问有相关代码更新或是参数配置指导吗？

HermanChen, 2021-06-01:
有没有码流发下？

imxys, 2021-06-01:
> 有没有码流发下？

链接: https://pan.baidu.com/s/1pvXsv_pgqqWNokZADho5mg 提取码: ue6v 复制这段内容后打开百度网盘手机App，操作更方便哦

zinsayon, 2021-06-01:
3399编码器可以调节空间很有限，从现象看编码器比较倾向编pskip，而ffmeg采用aq模式，在平坦的部分采用更低qp来达到编码p块或者I块，因此运动残余会好很多

imxys, 2021-06-01:
> 3399编码器可以调节空间很有限，从现象看编码器比较倾向编pskip，而ffmeg采用aq模式，在平坦的部分采用更低qp来达到编码p块或者I块，因此运动残余会好很多

这个编码倾向是硬件决定还是mpp决定的？可调节空间主要是配置QP吗？我正准备把编码的QP和IQP再降低点。

imxys, 2021-06-02:
@zinsayon 昨天回复的意思实际是“3399编码器可以调节空间很有限”，不用往这方面考虑了么？
我昨天测试了QP，QP好像就是要在31以下残影才会不明显，但码率就会超过2Mbps甚至达到3Mbps，
所以确认硬件或是mpp库都没效果？

会和视频源有关吗，1080@30fps的MJPEG转成yuv的，视频看过去人走动有点跳变不连贯的感觉，不知道是不是因为这个，rk3399编码才容易残影
我后面把ffmpeg编码出来的2Mbps视频再解出yuv让RK3399做编码，感觉残影现象好了很多
MJPEG因为100MB我就没有传，有需要我再发给你们

imxys, 2021-06-02:
@HermanChen 不好意思，RK3399有没有提供模糊处理的接口？我想如果能预先把画面处理一下再编码或许是个方法

RGA文档里说道的Dither抖色可以当作模糊来用么，RGA那边接口看得不是特别明白

HermanChen, 2021-07-07:
模糊处理这个有点麻烦，这个不在mpp能处理的范围内了，需要在 redmine 上提问题，找我们的 FAE 去解决

imxys, 2021-08-12:
> 模糊处理这个有点麻烦，这个不在mpp能处理的范围内了，需要在 redmine 上提问题，找我们的 FAE 去解决
好的，我后来想办法去处理输入源的视频解决问题了

closed 2021-08-12 by imxys