rockchip-linux/mpp #212: 请问下，怎么确认RK3399  android7.1硬件是否带mpp解码功能？

jutyy, 2021-06-02:
最近拿一块android7.1的RK3399安卓板测试mpp硬解码。
只要设置  MPP_SET_OUTPUT_BLOCK ，decode_get_frame 就会超时，解码不成功。
 是依照decode_get_frame再去读decode_get_frame 结果一直出现超时，没有成功过一次。  
想请教下：使用mpp硬件还要装什么东西吗？怎么确认硬件是否支持mpp解码？还是说所有的RK3399都支持？谢谢

jutyy, 2021-06-02:
//// 初始化
    MPP_RET ret         = MPP_OK;
    size_t file_size    = 0;

    // base flow context
    MppCtx ctx          = NULL;
    MppApi *mpi         = NULL;

    // input / output
    MppPacket packet    = NULL;
    MppFrame  frame     = NULL;

    MpiCmd mpi_cmd      = MPP_CMD_BASE;
    MppParam param      = NULL;
    RK_U32 need_split   = 1;
//    MppPollType timeout = 5;

    // paramter for resource malloc
    RK_U32 width        = 2560;
    RK_U32 height       = 1440;
    MppCodingType type  = MPP_VIDEO_CodingAVC;

    // resources
    char *buf           = NULL;
    size_t packet_size  = 8*1024;
    MppBuffer pkt_buf   = NULL;
    MppBuffer frm_buf   = NULL;

    MpiDecLoopData sdata;

    LOGD("mpp mpi_dec_test start\n");
    memset(&sdata, 0, sizeof(sdata));

//    sdata.fp_output = fopen("./tenoutput.yuv", "w+b");
//    if (NULL == sdata.fp_output) {
//        LOGE("failed to open output file %s\n", "tenoutput.yuv");
//        deInit(&packet, &frame, ctx, buf, sdata);
//    }

    LOGD("mpp mpi_dec_test decoder test start w %d h %d type %d\n", width, height, type);

    // decoder demo
    ret = mpp_create(&ctx, &mpi);
    if (MPP_OK != ret) {
        LOGE("mpp_create failed\n");
        deInit(&packet, &frame, ctx, buf, sdata);
    }

    // NOTE: decoder split mode need to be set before init
    mpi_cmd = MPP_DEC_SET_PARSER_SPLIT_MODE;
    param = &need_split;
    ret = mpi->control(ctx, mpi_cmd, param);
    if (MPP_OK != ret) {
        LOGE("mpp mpi->control failed\n");
        deInit(&packet, &frame, ctx, buf, sdata);
    }

    mpi_cmd = MPP_SET_INPUT_BLOCK;
    param = &need_split;
    ret = mpi->control(ctx, mpi_cmd, param);
    if (MPP_OK != ret) {
        LOGE("mpp mpi->control failed\n");
        deInit(&packet, &frame, ctx, buf, sdata);
    }

    mpi_cmd = MPP_SET_OUTPUT_BLOCK;
    param = &need_split;
    ret = mpi->control(ctx, mpi_cmd, param);
    if (MPP_OK != ret) {
        LOGE("mpp mpi->control failed\n");
        deInit(&packet, &frame, ctx, buf, sdata);
    }
    ret = mpp_init(ctx, MPP_CTX_DEC, type);
    if (MPP_OK != ret) {
        LOGE("mpp_init failed\n");
        deInit(&packet, &frame, ctx, buf, sdata);
    }  //// 初始化
    MPP_RET ret         = MPP_OK;
    size_t file_size    = 0;

    // base flow context
    MppCtx ctx          = NULL;
    MppApi *mpi         = NULL;

    // input / output
    MppPacket packet    = NULL;
    MppFrame  frame     = NULL;

    MpiCmd mpi_cmd      = MPP_CMD_BASE;
    MppParam param      = NULL;
    RK_U32 need_split   = 1;
//    MppPollType timeout = 5;

    // paramter for resource malloc
    RK_U32 width        = 2560;
    RK_U32 height       = 1440;
    MppCodingType type  = MPP_VIDEO_CodingAVC;

    // resources
    char *buf           = NULL;
    size_t packet_size  = 8*1024;
    MppBuffer pkt_buf   = NULL;
    MppBuffer frm_buf   = NULL;

    MpiDecLoopData sdata;

    LOGD("mpp mpi_dec_test start\n");
    memset(&sdata, 0, sizeof(sdata));

//    sdata.fp_output = fopen("./tenoutput.yuv", "w+b");
//    if (NULL == sdata.fp_output) {
//        LOGE("failed to open output file %s\n", "tenoutput.yuv");
//        deInit(&packet, &frame, ctx, buf, sdata);
//    }

    LOGD("mpp mpi_dec_test decoder test start w %d h %d type %d\n", width, height, type);

    // decoder demo
    ret = mpp_create(&ctx, &mpi);
    if (MPP_OK != ret) {
        LOGE("mpp_create failed\n");
        deInit(&packet, &frame, ctx, buf, sdata);
    }

    // NOTE: decoder split mode need to be set before init
    mpi_cmd = MPP_DEC_SET_PARSER_SPLIT_MODE;
    param = &need_split;
    ret = mpi->control(ctx, mpi_cmd, param);
    if (MPP_OK != ret) {
        LOGE("mpp mpi->control failed\n");
        deInit(&packet, &frame, ctx, buf, sdata);
    }

    mpi_cmd = MPP_SET_INPUT_BLOCK;
    param = &need_split;
    ret = mpi->control(ctx, mpi_cmd, param);
    if (MPP_OK != ret) {
        LOGE("mpp mpi->control failed\n");
        deInit(&packet, &frame, ctx, buf, sdata);
    }

    mpi_cmd = MPP_SET_OUTPUT_BLOCK;
    param = &need_split;
    ret = mpi->control(ctx, mpi_cmd, param);
    if (MPP_OK != ret) {
        LOGE("mpp mpi->control failed\n");
        deInit(&packet, &frame, ctx, buf, sdata);
    }
    ret = mpp_init(ctx, MPP_CTX_DEC, type);
    if (MPP_OK != ret) {
        LOGE("mpp_init failed\n");
        deInit(&packet, &frame, ctx, buf, sdata);
    }

初使一切都没有报错。

jutyy, 2021-06-02:
int decode_simple(MpiDecLoopData *data, AVPacket *av_packet )
{
    RK_U32 pkt_done = 0;
    RK_U32 pkt_eos  = 0;
    RK_U32 err_info = 0;
    MPP_RET ret = MPP_OK;
    MppCtx ctx  = data->ctx;
    MppApi *mpi = data->mpi;
    // char   *buf = data->buf;
    MppPacket packet = NULL;
    MppFrame  frame  = NULL;
    size_t read_size = 0;
    size_t packet_size = data->packet_size;

    mpp_log("mpp 解流开始--------------\n");
    ret = mpp_packet_init(&packet, av_packet->data, av_packet->size);
    mpp_packet_set_pts(packet, av_packet->pts);

    do {
        RK_S32 times = 5;
        // send the packet first if packet is not done
        if (!pkt_done) {
            ret = mpi->decode_put_packet(ctx, packet);
            mpp_log("mpp  decode_put_packet：%d,%d\n",ret,av_packet->size);
            if (MPP_OK == ret)
                pkt_done = 1;
        }

        // then get all available frame and release
        do {
            RK_S32 get_frm = 0;
            RK_U32 frm_eos = 0;

            try_again:
            ret = mpi->decode_get_frame(ctx, &frame);
            mpp_log("mpp  decode_get_frame：%d\n",ret);
            if (MPP_ERR_TIMEOUT == ret) {
                if (times > 0) {
                    times--;
                    msleep(2);
                    goto try_again;
                }
                mpp_err("mpp decode_get_frame failed too much time\n");
            }
            if (MPP_OK != ret) {
                mpp_err("mpp decode_get_frame failed ret %d\n", ret);
                break;
            }

            if (frame) {
                if (mpp_frame_get_info_change(frame)) {
                    RK_U32 width = mpp_frame_get_width(frame);
                    RK_U32 height = mpp_frame_get_height(frame);
                    RK_U32 hor_stride = mpp_frame_get_hor_stride(frame);
                    RK_U32 ver_stride = mpp_frame_get_ver_stride(frame);
                    RK_U32 buf_size = mpp_frame_get_buf_size(frame);

                    mpp_log("mpp decode_get_frame get info changed found\n");
                    mpp_log("mpp decoder require buffer w:h [%d:%d] stride [%d:%d] buf_size %d",
                            width, height, hor_stride, ver_stride, buf_size);

                    ret = mpp_buffer_group_get_internal(&data->frm_grp, MPP_BUFFER_TYPE_ION);
                    if (ret) {
                        mpp_err("mpp get mpp buffer group  failed ret %d\n", ret);
                        break;
                    }
                    mpi->control(ctx, MPP_DEC_SET_EXT_BUF_GROUP, data->frm_grp);

                    mpi->control(ctx, MPP_DEC_SET_INFO_CHANGE_READY, NULL);
                } else {
                    err_info = mpp_frame_get_errinfo(frame) | mpp_frame_get_discard(frame);
                    if (err_info) {
                        mpp_log("mpp decoder_get_frame get err info:%d discard:%d.\n",
                                mpp_frame_get_errinfo(frame), mpp_frame_get_discard(frame));
                    }
                    data->frame_count++;
                    mpp_log("mpp decode_get_frame get frame %d\n", data->frame_count);
                   if (data->fp_output && !err_info){
                       mpp_log("mpp 解码成功-------- %d\n", frame);
//                       cv::Mat rgbImg;
//                       YUV420SP2Mat(frame, rgbImg);
                    //    cv::imwrite("./"+std::to_string(count++)+".jpg", rgbImg);
                   
                    //    dump_mpp_frame_to_file(frame, data->fp_output);
                   }
                }
                frm_eos = mpp_frame_get_eos(frame);
                mpp_frame_deinit(&frame);

                frame = NULL;
                get_frm = 1;
            }
            else{
                mpp_log("mpp  no frame------\n");
            }

            // try get runtime frame memory usage
            if (data->frm_grp) {
                size_t usage = mpp_buffer_group_usage(data->frm_grp);
                if (usage > data->max_usage)
                    data->max_usage = usage;
            }

            // if last packet is send but last frame is not found continue
            if (pkt_eos && pkt_done && !frm_eos) {
                msleep(10);
                continue;
            }

            if (frm_eos) {
                mpp_log("mpp found last frame\n");
                break;
            }

            if (data->frame_num > 0 && data->frame_count >= data->frame_num) {
                data->eos = 1;
                break;
            }

            if (get_frm)
                continue;
            break;
        } while (1);

        if (data->frame_num > 0 && data->frame_count >= data->frame_num) {
            data->eos = 1;
            mpp_log("mpp reach max frame number %d\n", data->frame_count);
            break;
        }

        if (pkt_done)
            break;

        /*
         * why sleep here:
         * mpi->decode_put_packet will failed when packet in internal queue is
         * full,waiting the package is consumed .Usually hardware decode one
         * frame which resolution is 1080p needs 2 ms,so here we sleep 3ms
         * * is enough.
         */
        msleep(3);
    } while (1);
    mpp_packet_deinit(&packet);

    return ret;
}
decode_get_frame解码 一直打印超时-8。

HermanChen, 2021-06-03:
使用阻塞模式需要参考 mpi_dec_mt_test

jutyy, 2021-06-03:
谢谢您的回答。    如果不用MPP_SET_OUTPUT_BLOCK阻塞。
我参考的是这个 https://github.com/MUZLATAN/ffmpeg_rtsp_mpp，这个情况frame就一直为空。

jutyy, 2021-06-03:
请问下您有32位的mpp库吗？

HermanChen, 2021-06-03:
mpp 可以编译 32 位或者 64 位的
一直为空应该是解还没完成吧？

jutyy, 2021-06-03:
不是没有解完。  请问您有32位的mpp库吗

HermanChen, 2021-06-03:
我这边编译环境不一定和你的一样，编译出来的二进制文件不一定能用

jutyy, 2021-06-03:
嗯，那有编译教程吗？

jutyy, 2021-06-03:
嗯，那有编译教程吗？谢谢！

HermanChen, 2021-06-03:
编译看下 build 目录下不同环境的编译方法，需要 cmake