rockchip-linux/mpp #214: 请问，rk3288上支持yuv422的硬解码吗

luocha, 2021-07-02:
硬解码yuv420的h264是没问题的，但硬解yuv422的h264就失败了，这个是不支持吗
root@linaro-alip:/home/luoc/test/mpp-develop/test# ./mpi_dec_test -i /home/luoc/test/desktop-1920-1080.h264 
mpp[9967]: mpi_dec_utils: cmd parse result:
mpp[9967]: mpi_dec_utils: input  file name: /home/luoc/test/desktop-1920-1080.h264
mpp[9967]: mpi_dec_utils: output file name: 
mpp[9967]: mpi_dec_utils: config file name: 
mpp[9967]: mpi_dec_utils: width      :    0
mpp[9967]: mpi_dec_utils: height     :    0
mpp[9967]: mpi_dec_utils: type       : 7
mpp[9967]: mpi_dec_utils: debug flag : 0
mpp[9967]: mpi_dec_utils: max frames : 0
mpp[9967]: mpi_dec_test: mpi_dec_test start
mpp[9967]: mpi_dec_test: input file size 1362983
mpp[9967]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[9967]: mpi_dec_test: 0x80b36020 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[9967]: mpp_rt: NOT found ion allocator
mpp[9967]: mpp_rt: found drm allocator
mpp[9967]: mpi_dec_test: 0x80b36020 decode_get_frame get info changed found
mpp[9967]: mpi_dec_test: 0x80b36020 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 5222400
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 0 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 1 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 2 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 3 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 4 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 5 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 6 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 7 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 8 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 9 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 10 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 11 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 12 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 13 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 14 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 15 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 16 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 17 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 18 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 19 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 20 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 21 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 22 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 23 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 24 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 25 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 26 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 27 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 28 err 1 discard 0
mpp[9967]: mpi_dec_test: 0x80b36020 decode get frame 29 err 1 discard 0

luocha, 2021-07-06:
这里是详细的log

[BEGIN] 2021/7/6 9:58:49
root@linaro-alip:/home/luoc/test/mpp-develop/test# ./mpi_dec_test -i /home/luoc/test/desktop-1920-1080-i422.h264 -t 7
mpp[12636]: mpi_dec_utils: cmd parse result:
mpp[12636]: mpi_dec_utils: input  file name: /home/luoc/test/desktop-1920-1080-i422.h264
mpp[12636]: mpi_dec_utils: output file name: 
mpp[12636]: mpi_dec_utils: config file name: 
mpp[12636]: mpi_dec_utils: width      :    0
mpp[12636]: mpi_dec_utils: height     :    0
mpp[12636]: mpi_dec_utils: type       : 7
mpp[12636]: mpi_dec_utils: debug flag : 0
mpp[12636]: mpi_dec_utils: max frames : 0
mpp[12636]: mpi_dec_test: mpi_dec_test start
mpp[12636]: mpi_dec_test: input file size 1377643
mpp[12636]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[12636]: mpi_dec_test: 0x81063020 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[12636]: mpp_rt: NOT found ion allocator
mpp[12636]: mpp_rt: found drm allocator
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[12636]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
[... cut ...]
mpp[12636]: h264d_api: [CALLBACK] g_no=78, out_idx=1, dpberr=1, harderr=1, ref_flag=1, errinfo=1, discard=0
mpp[12636]: mpi_dec_test: 0x81063020 decode get frame 76 err 1 discard 0
mpp[12636]: mpi_dec_test: 0x81063020 decode get frame 77 err 1 discard 0
mpp[12636]: mpi_dec_test: 0x81063020 found last packet
mpp[12636]: mpi_dec_test: 0x81063020 input 337 pkt output 79 frm decode 78 frames
mpp[12636]: h264d_dpb: warning: max_dec_frame_buffering(1) is less than dpb_size(4) calculated from Profile/Level.
mpp[12636]: mpi_dec_test: test success max memory 14.94 MB

[END] 2021/7/6 9:58:57

HermanChen, 2021-07-07:
3288 的 vpu 解码器不支持 422 码流