rockchip-linux/mpp #216: 运行板子自带的mpi_dec_test能正常，运行编译源码的会报错

liuguohuan-xiezuo, 2021-07-26:
./mpi_dec_test -t 7 -i test.h264 -o output.yuv -n 10
mpp[20329]: mpi_dec_utils: cmd parse result:
mpp[20329]: mpi_dec_utils: input  file name: test.h264
mpp[20329]: mpi_dec_utils: output file name: output.yuv
mpp[20329]: mpi_dec_utils: config file name:
mpp[20329]: mpi_dec_utils: width      :    0
mpp[20329]: mpi_dec_utils: height     :    0
mpp[20329]: mpi_dec_utils: type       : 7
mpp[20329]: mpi_dec_utils: debug flag : 0
mpp[20329]: mpi_dec_utils: max frames : 10
mpp[20329]: mpi_dec_test: mpi_dec_test start
mpp[20329]: mpi_dec_test: input file size 8330818
mpp[20329]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[20329]: mpi_dec_test: 0x559c16b9c0 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[20329]: mpp_rt: NOT found ion allocator
mpp[20329]: mpp_rt: found drm allocator
mpp[20329]: mpi_dec_test: 0x559c16b9c0 decode_get_frame get info changed found
mpp[20329]: mpi_dec_test: 0x559c16b9c0 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 3760128
mpp[20329]: vcodec_service: vcodec_service_cmd_send ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
mpp[20329]: hal_h264d_vdpu_reg: vdpu2_h264d_start send cmd failed 14
mpp[20329]: vcodec_service: vcodec_service_cmd_poll ioctl VPU_IOC_GET_REG failed ret -1 errno 110 Connection timed out
mpp[20329]: hal_h264d_vdpu_reg: vdpu2_h264d_wait poll cmd failed 110
mpp[20329]: mpi_dec_test: 0x559c16b9c0 decode get frame 0 err 1 discard 0
mpp[20329]: mpi_dec_test: 0x559c16b9c0 decode get frame 1 err 1 discard 0
mpp[20329]: mpi_dec_test: 0x559c16b9c0 decode get frame 2 err 1 discard 0
mpp[20329]: mpi_dec_test: 0x559c16b9c0 decode get frame 3 err 1 discard 0
mpp[20329]: mpi_dec_test: 0x559c16b9c0 decode get frame 4 err 1 discard 0
mpp[20329]: mpi_dec_test: 0x559c16b9c0 decode get frame 5 err 1 discard 0
mpp[20329]: mpi_dec_test: 0x559c16b9c0 decode get frame 6 err 1 discard 0
mpp[20329]: mpi_dec_test: 0x559c16b9c0 decode get frame 7 err 1 discard 0
mpp[20329]: mpi_dec_test: 0x559c16b9c0 decode get frame 8 err 1 discard 0
mpp[20329]: mpi_dec_test: 0x559c16b9c0 decode get frame 9 err 1 discard 0
mpp[20329]: mpi_dec_test: 0x559c16b9c0 reach max frame number 10
mpp[20329]: mpi_dec_test: 0x559c16b9c0 input 360 pkt output 11 frm decode 10 frames
mpp[20329]: mpi_dec_test: test success max memory 14.34 MB

liuguohuan-xiezuo, 2021-07-26:
./mpp_platform_test
mpp[19186]: mpp_plat_test: kernel version: 4.4
mpp[19186]: mpp_plat_test: ioctl  version: vcodec_service
mpp[19186]: mpp_plat_test:
mpp[19186]: mpp_plat_test: chip name: rockchip,px30-evb-ddr3-v10-linux rockchip,px30
mpp[19186]: mpp_plat_test:
mpp[19186]: mpp_plat_test: chip vcodec type 0004010a
mpp[19186]: mpp_plat_test: found vpu2 codec
mpp[19186]: mpp_plat_test: found rk hevc decoder
mpp[19186]: mpp_plat_test: found vpu2 encoder
mpp[19186]: mpp_plat_test:
mpp[19186]: mpp_plat_test: start probing decoder device name:
mpp[19186]: mpp_plat_test: H.264 decoder: /dev/vpu_service
mpp[19186]: mpp_plat_test: H.265 decoder: /dev/hevc_service
mpp[19186]: mpp_plat_test: MJPEG decoder: /dev/vpu_service
mpp[19186]: mpp_plat_test: VP9   decoder: /dev/vpu_service
mpp[19186]: mpp_plat_test: avs   decoder: /dev/vpu_service
mpp[19186]: mpp_plat_test:
mpp[19186]: mpp_plat_test: start probing encoder device name:
mpp[19186]: mpp_plat_test: H.264 encoder: /dev/vpu_service
mpp[19186]: mpp_plat_test: H.265 encoder: /dev/vpu_service
mpp[19186]: mpp_plat_test: MJPEG encoder: /dev/vpu_service
mpp[19186]: mpp_plat_test: mpp platform test done

HermanChen, 2021-07-26:
vpu_service 这个内核估计是 3.10 的，比较老了，支持有问题……

liuguohuan-xiezuo, 2021-07-26:
这个有什么办法可以解决吗，升级内核？因为板子本身可以支持mpp，那如果我把他的库跟头文件，拷贝出来，也可以用吗？

liuguohuan-xiezuo, 2021-07-26:
这个有什么办法可以解决吗，升级内核？或者能回退到mpp-develop的什么版本吗？

HermanChen, 2021-07-26:
mpp[19186]: mpp_plat_test: kernel version: 4.4
mpp[19186]: mpp_plat_test: ioctl version: vcodec_service
mpp[19186]: mpp_plat_test:
mpp[19186]: mpp_plat_test: chip name: rockchip,px30-evb-ddr3-v10-linux rockchip,px30
这里看，内核虽然是 4.4 的，但是设备驱动还是 3.10 的 /dev/vpu_service，这个比较老了

升级 sdk 是比较好的选择，最新环境上的问题会少很多，
回退 mpp 需要回退到很老的版本了，不建议

liuguohuan-xiezuo, 2021-07-26:
设备驱动在哪里可以下载升级呢

FumasterLin, 2021-07-26:
你先更新下这个github上最新的mpp，再测试下看看

liuguohuan-xiezuo, 2021-07-26:
我在用的一直是最新的guthub上的mpp-develop

FumasterLin, 2021-07-26:
kmsg看一下报什么

liuguohuan-xiezuo, 2021-07-26:
[21570.560156] rk_vcodec: vpu_service_ioctl:2079: error: pid 15813 wait 0 task done timeout
[27040.138555] rk_vcodec: vpu_service_ioctl:2138: error: unknown vpu service ioctl cmd 40086c01
[27040.371119] rk_vcodec: reg_init:1487: error: translate reg address failed, dumping regs
[27040.379307] rk-vcodec vpu_combo: reg[00]: 00000000
[27040.384256] rk-vcodec vpu_combo: reg[01]: 00000000
[27040.389129] rk-vcodec vpu_combo: reg[02]: 00000000
[27040.393969] rk-vcodec vpu_combo: reg[03]: 00000000
[27040.398796] rk-vcodec vpu_combo: reg[04]: 00000000
[27040.403629] rk-vcodec vpu_combo: reg[05]: 00000000
[27040.408473] rk-vcodec vpu_combo: reg[06]: 00000000
[27040.413342] rk-vcodec vpu_combo: reg[07]: 00000000
[27040.418370] rk-vcodec vpu_combo: reg[08]: 00000000
[27040.423252] rk-vcodec vpu_combo: reg[09]: 00000000
[27040.428097] rk-vcodec vpu_combo: reg[10]: 00000000
[27040.433027] rk-vcodec vpu_combo: reg[11]: 00000000
[27040.437903] rk-vcodec vpu_combo: reg[12]: 00000000
[27040.442760] rk-vcodec vpu_combo: reg[13]: 00000000
[27040.447630] rk-vcodec vpu_combo: reg[14]: 00000000
[27040.452523] rk-vcodec vpu_combo: reg[15]: 00000000
[27040.457392] rk-vcodec vpu_combo: reg[16]: 00000000
[27040.462297] rk-vcodec vpu_combo: reg[17]: 00000000
[27040.467145] rk-vcodec vpu_combo: reg[18]: 00000000
[27040.472008] rk-vcodec vpu_combo: reg[19]: 00000000
[27040.476998] rk-vcodec vpu_combo: reg[20]: 00000000
[27040.481895] rk-vcodec vpu_combo: reg[21]: 00000000
[27040.486751] rk-vcodec vpu_combo: reg[22]: 00000000
[27040.491605] rk-vcodec vpu_combo: reg[23]: 00000000
[27040.496473] rk-vcodec vpu_combo: reg[24]: 00000000
[27040.501331] rk-vcodec vpu_combo: reg[25]: 00000000
[27040.506179] rk-vcodec vpu_combo: reg[26]: 00000000
[27040.511050] rk-vcodec vpu_combo: reg[27]: 00000000
[27040.515937] rk-vcodec vpu_combo: reg[28]: 00000000
[27040.520810] rk-vcodec vpu_combo: reg[29]: 00000000
[27040.525662] rk-vcodec vpu_combo: reg[30]: 00000000
[27040.530511] rk-vcodec vpu_combo: reg[31]: 00000000
[27040.535364] rk-vcodec vpu_combo: reg[32]: 00000000
[27040.540224] rk-vcodec vpu_combo: reg[33]: 00000000
[27040.545088] rk-vcodec vpu_combo: reg[34]: 00000000
[27040.549973] rk-vcodec vpu_combo: reg[35]: 00000000
[27040.554841] rk-vcodec vpu_combo: reg[36]: 00000000
[27040.559689] rk-vcodec vpu_combo: reg[37]: 00000000
[27040.564549] rk-vcodec vpu_combo: reg[38]: 00000000
[27040.569391] rk-vcodec vpu_combo: reg[39]: 00000000
[27040.574275] rk-vcodec vpu_combo: reg[40]: 00000000
[27040.579147] rk-vcodec vpu_combo: reg[41]: 00000000
[27040.584013] rk-vcodec vpu_combo: reg[42]: 00000000
[27040.588865] rk-vcodec vpu_combo: reg[43]: 00000000
[27040.593720] rk-vcodec vpu_combo: reg[44]: 00000000
[27040.598567] rk-vcodec vpu_combo: reg[45]: 00000000
[27040.603411] rk-vcodec vpu_combo: reg[46]: 00000000
[27040.608275] rk-vcodec vpu_combo: reg[47]: 00000000
[27040.613153] rk-vcodec vpu_combo: reg[48]: 00000000
[27040.618001] rk-vcodec vpu_combo: reg[49]: 00000000
[27040.622845] rk-vcodec vpu_combo: reg[50]: 00000000
[27040.627746] rk-vcodec vpu_combo: reg[51]: 340b0010
[27040.632621] rk-vcodec vpu_combo: reg[52]: 00100000
[27040.637484] rk-vcodec vpu_combo: reg[53]: 00000000
[27040.642349] rk-vcodec vpu_combo: reg[54]: 0000003e
[27040.647200] rk-vcodec vpu_combo: reg[55]: 00000000
[27040.652055] rk-vcodec vpu_combo: reg[56]: 003000ff
[27040.656922] rk-vcodec vpu_combo: reg[57]: 3e402471
[27040.661776] rk-vcodec vpu_combo: reg[58]: 00000000
[27040.666621] rk-vcodec vpu_combo: reg[59]: 007fb050
[27040.671491] rk-vcodec vpu_combo: reg[60]: 00000000
[27040.676370] rk-vcodec vpu_combo: reg[61]: 00000000
[27040.681253] rk-vcodec vpu_combo: reg[62]: 0bf40002
[27040.686106] rk-vcodec vpu_combo: reg[63]: 00000002
[27040.690966] rk-vcodec vpu_combo: reg[64]: 00000001
[27040.695819] rk-vcodec vpu_combo: reg[65]: 00000000
[27040.700677] rk-vcodec vpu_combo: reg[66]: 00000000
[... 68 lines cut ...]
[27041.038956] rk-vcodec vpu_combo: reg[135]: 00000000
[27041.043891] rk-vcodec vpu_combo: reg[136]: 00000000
[27041.048837] rk-vcodec vpu_combo: reg[137]: 00000000
[27041.053778] rk-vcodec vpu_combo: reg[138]: 00000000
[27041.058741] rk-vcodec vpu_combo: reg[139]: 00000000
[27041.063669] rk-vcodec vpu_combo: reg[140]: 00000000
[27041.068619] rk-vcodec vpu_combo: reg[141]: 00000000
[27041.073555] rk-vcodec vpu_combo: reg[142]: 00000000
[27041.078498] rk-vcodec vpu_combo: reg[143]: 00000000
[27041.083429] rk-vcodec vpu_combo: reg[144]: 00000000
[27041.088390] rk-vcodec vpu_combo: reg[145]: 00000000
[27041.093337] rk-vcodec vpu_combo: reg[146]: 00000000
[27041.098281] rk-vcodec vpu_combo: reg[147]: 00000000
[27041.103246] rk-vcodec vpu_combo: reg[148]: 00000000
[27041.108243] rk-vcodec vpu_combo: reg[149]: 00000000
[27041.113153] rk-vcodec vpu_combo: reg[150]: 00000000
[27041.118097] rk-vcodec vpu_combo: reg[151]: 00000000
[27041.123060] rk-vcodec vpu_combo: reg[152]: 00000000
[27041.127991] rk-vcodec vpu_combo: reg[153]: 00000000
[27041.132945] rk-vcodec vpu_combo: reg[154]: 00000000
[27041.137870] rk-vcodec vpu_combo: reg[155]: 00000000
[27041.142851] rk-vcodec vpu_combo: reg[156]: 00000000
[27041.147778] rk-vcodec vpu_combo: reg[157]: 00000000
[27041.152751] rk-vcodec vpu_combo: reg[158]: 00000000
[27043.158330] rk_vcodec: vpu_service_ioctl:2079: error: pid 18548 wait 0 task done timeout

FumasterLin, 2021-07-27:
diff --git a/build/linux/arm/make-Makefiles.bash b/build/linux/arm/make-Makefiles.bash
index 9dcbc7e8..1e891339 100755
--- a/build/linux/arm/make-Makefiles.bash
+++ b/build/linux/arm/make-Makefiles.bash
@@ -24,5 +24,6 @@ fi
 cmake -DCMAKE_BUILD_TYPE=Release \
       -DCMAKE_TOOLCHAIN_FILE=./arm.linux.cross.cmake \
       -DRKPLATFORM=ON \
+      -DHAVE_DRM=ON \
       -G "Unix Makefiles" \
       ${MPP_TOP}

FumasterLin, 2021-07-27:
合入这个修改测试下看看

```
diff --git a/build/linux/arm/make-Makefiles.bash b/build/linux/arm/make-Makefiles.bash
index 9dcbc7e8..1e891339 100755
--- a/build/linux/arm/make-Makefiles.bash
+++ b/build/linux/arm/make-Makefiles.bash
@@ -24,5 +24,6 @@ fi
 cmake -DCMAKE_BUILD_TYPE=Release \
       -DCMAKE_TOOLCHAIN_FILE=./arm.linux.cross.cmake \
       -DRKPLATFORM=ON \
+      -DHAVE_DRM=ON \
       -G "Unix Makefiles" \
       ${MPP_TOP}
```

liuguohuan-xiezuo, 2021-07-27:
我在build/linux/arm/make-Makefiles.bash里面加了-DHAVE_DRM=ON \ ，还是一样的报错

FumasterLin, 2021-07-27:
1. 加入这个修改后，
2, echo 0xffff > /sys/module/rk_vcodec/parameters/debug
3. 跑一下test，看看log和kmsg
```
diff --git a/osal/linux/os_allocator.c b/osal/linux/os_allocator.c
index b7dbfeea..e96d5751 100644
--- a/osal/linux/os_allocator.c
+++ b/osal/linux/os_allocator.c
@@ -41,6 +41,11 @@ MPP_RET os_allocator_get(os_allocator *api, MppBufferType type)
                (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_DRM)) ? allocator_drm :
 #endif
                allocator_std;
+#if HAVE_DRM
+
+        mpp_log("ion %d drm %d \n", mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_ION),
+                                mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_DRM));
+#endif
     } break;
     case MPP_BUFFER_TYPE_EXT_DMA: {
         *api = allocator_ext_dma;
```

liuguohuan-xiezuo, 2021-07-27:
这是改了的os_allocator.c文件
/*
 * Copyright 2015 Rockchip Electronics Co. LTD
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *      http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

#if defined(__gnu_linux__)
#include "mpp_log.h"
#include "mpp_runtime.h"

#include "allocator_drm.h"
#include "allocator_ext_dma.h"
#include "allocator_ion.h"
#include "allocator_std.h"

/*
 * Linux only support MPP_BUFFER_TYPE_NORMAL so far
 * we can support MPP_BUFFER_TYPE_V4L2 later
 */

MPP_RET os_allocator_get(os_allocator *api, MppBufferType type)
{
    MPP_RET ret = MPP_OK;
    switch (type) {
    case MPP_BUFFER_TYPE_NORMAL : {
        *api = allocator_std;
    } break;
    case MPP_BUFFER_TYPE_ION : {
        *api = (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_ION)) ? allocator_ion :
#if HAVE_DRM
               (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_DRM)) ? allocator_drm :
#endif
               allocator_std;
#if HAVE_DRM
    mpp_log("ion %d drm %d \n", mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_ION),
                                    mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_DRM));
#endif
    } break;
    case MPP_BUFFER_TYPE_EXT_DMA: {
        *api = allocator_ext_dma;
    } break;
    case MPP_BUFFER_TYPE_DRM : {
#if HAVE_DRM
        *api = (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_DRM)) ? allocator_drm :
#else
        * api =
#endif
               (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_ION)) ? allocator_ion :
               allocator_std;
    } break;
    default : {
        ret = MPP_NOK;
    } break;
    }
    return ret;
}

#endif

[... 193 lines cut ...]
[ 1105.635649] rk-vcodec vpu_combo: reg[140]: 00000000
[ 1105.640587] rk-vcodec vpu_combo: reg[141]: 00000000
[ 1105.645632] rk-vcodec vpu_combo: reg[142]: 00000000
[ 1105.650562] rk-vcodec vpu_combo: reg[143]: 00000000
[ 1105.655501] rk-vcodec vpu_combo: reg[144]: 00000000
[ 1105.660445] rk-vcodec vpu_combo: reg[145]: 00000000
[ 1105.665421] rk-vcodec vpu_combo: reg[146]: 00000000
[ 1105.670362] rk-vcodec vpu_combo: reg[147]: 00000000
[ 1105.675302] rk-vcodec vpu_combo: reg[148]: 00000000
[ 1105.680225] rk-vcodec vpu_combo: reg[149]: 00000000
[ 1105.685179] rk-vcodec vpu_combo: reg[150]: 00000000
[ 1105.690118] rk-vcodec vpu_combo: reg[151]: 00000000
[ 1105.695105] rk-vcodec vpu_combo: reg[152]: 00000000
[ 1105.700033] rk-vcodec vpu_combo: reg[153]: 00000000
[ 1105.704977] rk-vcodec vpu_combo: reg[154]: 00000000
[ 1105.709909] rk-vcodec vpu_combo: reg[155]: 00000000
[ 1105.714848] rk-vcodec vpu_combo: reg[156]: 00000000
[ 1105.719772] rk-vcodec vpu_combo: reg[157]: 00000000
[ 1105.724730] rk-vcodec vpu_combo: reg[158]: 00000000
[ 1105.730127] rk_vcodec: vpu_service_ioctl:1965: enter
[ 1105.730142] rk_vcodec: cmd 40086c04, VPU_IOC_SET_CLIENT_TYPE 40046c01
[ 1105.730149] rk_vcodec: pid 8527 get reg type 1
[ 1107.729809] rk_vcodec: vpu_service_ioctl:2079: error: pid 8527 wait 0 task done timeout
[ 1107.874640] rk_vcodec: vpu_service_release:2303: enter
[ 1107.874669] rk_vcodec: vpu_service_release:2338: leave

liuguohuan-xiezuo, 2021-07-27:
升级设备驱动的话，需要升级到什么版本呢

FumasterLin, 2021-07-27:
你加的这句log没打印出来，说明HAVE_DRM这个config没开。

确认下对应的make-Makefiles.bash里是否配置了-DHAVE_DRM=ON

如果已经添加了，那就把当前编译的结果 clean下，直接删除生成的文件后，再重新编译下。

#if HAVE_DRM
mpp_log("ion %d drm %d \n", mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_ION),
mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_DRM));
#endif

liuguohuan-xiezuo, 2021-07-27:
这个确实加进去了
#!/bin/bash
# Run this from within a bash shell

set +e

MPP_PWD=`pwd`
MPP_TOP=${MPP_PWD}/../../..

# toolchain detection
check_cmd(){
    "$@" >> /dev/null 2>&1
}
check_system_arm_linux_gcc(){
    check_cmd arm-linux-gcc -v
}

check_system_arm_linux_gcc
if [ $? -eq 127 ];then
    MPP_TOOLCHAIN=${MPP_TOP}/../prebuilts/toolschain/usr/bin
    export PATH=$PATH:${MPP_TOOLCHAIN}
fi

# generate Makefile
cmake -DCMAKE_BUILD_TYPE=Release \
      -DCMAKE_TOOLCHAIN_FILE=./arm.linux.cross.cmake \
      -DRKPLATFORM=ON \
      -DHAVE_DRM=ON \
      -G "Unix Makefiles" \
      ${MPP_TOP}

MPP_RET os_allocator_get(os_allocator *api, MppBufferType type)
{
    MPP_RET ret = MPP_OK;
    switch (type) {
    case MPP_BUFFER_TYPE_NORMAL : {
        *api = allocator_std;
    } break;
    case MPP_BUFFER_TYPE_ION : {
        *api = (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_ION)) ? allocator_ion :
#if HAVE_DRM
               (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_DRM)) ? allocator_drm :
#endif
               allocator_std;
#if HAVE_DRM
        mpp_log("ion %d drm %d \n", mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_ION),
        mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_DRM));
#endif
    } break;
    case MPP_BUFFER_TYPE_EXT_DMA: {
        *api = allocator_ext_dma;
    } break;
    case MPP_BUFFER_TYPE_DRM : {
#if HAVE_DRM
        *api = (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_DRM)) ? allocator_drm :
#else
        * api =
#endif
               (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_ION)) ? allocator_ion :
               allocator_std;
    } break;
    default : {
        ret = MPP_NOK;
    } break;
    }
    return ret;
}

#endif

我每次修改完都会make clean，再重新make的，你说没打印的那个日志，是应该运行mpi_dec_test的时候应该会打印出来的吗？

FumasterLin, 2021-07-27:
1、你可以试一下把build/linux/arm/下除了arm.linux.cross.cmake和arm.linux.cross.cmake两个文件，其他文件都删掉看看。
2、或者直接按如下修改

```
diff --git a/osal/linux/os_allocator.c b/osal/linux/os_allocator.c
index b7dbfeea..45064f75 100644
--- a/osal/linux/os_allocator.c
+++ b/osal/linux/os_allocator.c
@@ -37,20 +37,16 @@ MPP_RET os_allocator_get(os_allocator *api, MppBufferType type)
     } break;
     case MPP_BUFFER_TYPE_ION : {
         *api = (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_ION)) ? allocator_ion :
-#if HAVE_DRM
                (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_DRM)) ? allocator_drm :
-#endif
                allocator_std;
+        mpp_log("ion %d drm %d \n", mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_ION),
+                                mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_DRM));
     } break;
     case MPP_BUFFER_TYPE_EXT_DMA: {
         *api = allocator_ext_dma;
     } break;
     case MPP_BUFFER_TYPE_DRM : {
-#if HAVE_DRM
         *api = (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_DRM)) ? allocator_drm :
-#else
-        * api =
-#endif
                (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_ION)) ? allocator_ion :
                allocator_std;
     } break;
```

liuguohuan-xiezuo, 2021-07-27:
build/linux/arm/下本来就只有arm.linux.cross.cmake和arm.linux.cross.cmake两个文件，我刚刚试了，去掉了if HAVE_DRM，直接去执行(mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_DRM)) ? allocator_drm :的话，编译会报错undefined reference to `allocator_drm'

FumasterLin, 2021-07-27:
这个修改也同步加上

```
diff --git a/osal/CMakeLists.txt b/osal/CMakeLists.txt
index 0df9d1dc..3fe8545c 100644
--- a/osal/CMakeLists.txt
+++ b/osal/CMakeLists.txt
@@ -10,13 +10,9 @@ set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -D_REENTRANT -D_GNU_SOURCE")
 set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64")
 set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fPIC")
 
-if (HAVE_DRM)
-    add_definitions(-DHAVE_DRM)
-    set(DRM_FILES allocator/allocator_drm.c)
-    message(STATUS "compile with drm support")
-else()
-    message(STATUS "compile without drm support")
-endif()
+
+set(DRM_FILES allocator/allocator_drm.c)
+message(STATUS "compile with drm support")
 
 set(MPP_ALLOCATOR
     allocator/allocator_std.c
```

liuguohuan-xiezuo, 2021-07-28:
确实这样做解决了，没有报错了，可以使用

liuguohuan-xiezuo, 2021-08-16:
最新的版本，MPP_ENC_GET_EXTRA_INFO接口已经是不能用了是吗，我一调用，就会报错，提示我用MPP_ENC_GET_HDR_SYNC接口

HermanChen, 2021-08-17:
还可以使用，只是个提示log，但不保证将来还能支持，最好切换下接口，原接口有异步问题。

liuguohuan-xiezuo, 2021-08-17:
我把编译出来数据存264文件，用分析工具打开，看到里面的I帧跟P帧好像都有点问题，下面的264文件的格式，里面有个数据的重要级别，我用ffmpeg跟其他硬编数来的数据，I帧的数据是0x68，P帧的数据是0x61，但我用mpp编译出来的，I帧的数据是0x25，P帧的数据是0x21，请问这个可以设置他的级别吗
[image]