rockchip-linux/mpp #218: 请问，RK3328的ubuntu 20.04(内核版本5.8)上如何支持mpp解码呢？

zhangweiwahaha042, 2021-08-10:
■概要
・硬件平台
Orange Pi R1 Plus(RK3328)
・OS
ubuntu20.04
※代码地址：https://github.com/orangepi-xunlong/orangepi-build
・目的
在上述硬件平台上，我们想使用mpp的硬解码功能
・问题
(1)执行命令mpp_platform_test时，找不到decoder和encoder
(2)执行命令mpi_dec_test时发生了错误
・错误的一次解析
ubuntu20.04的linux内核版本是5.8，该内核中没有驱动vpu_service
・补充
上述硬件平台的Android9的linux内核版本是4.4，该内核中有驱动vpu_service
※驱动代码路径：kernel/drivers/video/rockchip/vcodec/

■问题一
为了在ubuntu20.04上支持mpp解码功能，我们应该怎么做比较好呢？我们现在想到的方向如下，你们觉得是否可行？
・方案①
 将Android9上内核版本4.4的vpu_service驱动移植到ubuntu20.04上内核版本5.8上
 ※如果贵公司有相关补丁提供的话，不胜感激
・方案②
 贵公司有没有其他的推荐方案

■问题二
本github的mpp代码是否只支持在kernel 4.4，不支持kernel5.8

■日志(1)
执行命令mpp_platform_test时，找不到decoder和encoder
$ ./mpp_platform_test
mpp[20607]: mpp_plat_test: kernel version: unknown
mpp[20607]: mpp_plat_test: ioctl  version: vcodec_service
mpp[20607]: mpp_plat_test:
mpp[20607]: mpp_plat_test: chip name: rockchip,rk3328
mpp[20607]: mpp_plat_test:
mpp[20607]: mpp_plat_test: chip vcodec type 0104020a
mpp[20607]: mpp_plat_test: found vpu2 codec
mpp[20607]: mpp_plat_test: found rkvdec decoder
mpp[20607]: mpp_plat_test: found vpu2 encoder
mpp[20607]: mpp_plat_test: found h265 stand-alone encoder
mpp[20607]: mpp_plat_test:
mpp[20607]: mpp_plat_test: start probing decoder device name:
mpp[20607]: mpp_plat_test: H.264 decoder: (null)    ★
mpp[20607]: mpp_plat_test: H.265 decoder: (null)    ★
mpp[20607]: mpp_plat_test: MJPEG decoder: (null)    ★
mpp[20607]: mpp_plat_test: VP9   decoder: (null)    ★
mpp[20607]: mpp_plat_test: avs   decoder: (null)    ★
mpp[20607]: mpp_plat_test:
mpp[20607]: mpp_plat_test: start probing encoder device name:
mpp[20607]: mpp_plat_test: H.264 encoder: (null)    ★
mpp[20607]: mpp_plat_test: H.265 encoder: (null)    ★
mpp[20607]: mpp_plat_test: MJPEG encoder: (null)    ★
mpp[20607]: mpp_plat_test: mpp platform test done

■日志(2)
执行命令mpi_dec_test时，报错
$./mpi_dec_test -t 7 -i input/h264.mp4 -n 10
mpp[30550]: mpi_dec_utils: cmd parse result:
mpp[30550]: mpi_dec_utils: input  file name: input/h264.mp4
mpp[30550]: mpi_dec_utils: output file name:
mpp[30550]: mpi_dec_utils: config file name:
mpp[30550]: mpi_dec_utils: width      :    0
mpp[30550]: mpi_dec_utils: height     :    0
mpp[30550]: mpi_dec_utils: type       : 7
mpp[30550]: mpi_dec_utils: debug flag : 0
mpp[30550]: mpi_dec_utils: max frames : 10
mpp[30550]: mpi_dec_test: mpi_dec_test start
mpp[30550]: mpi_dec_test: input file size 10183939
mpp[30550]: mpp_info: mpp version: 02058ce6 author: Yandong Lin   2021-07-29 [h264d_parse]: skip sp/si slice
mpp[30550]: mpi_dec_test: 0xaaaac4ebe980 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[30550]: mpp_rt: NOT found ion allocator                          ★
mpp[30550]: mpp_rt: NOT found drm allocator                          ★
mpp[30550]: vcodec_service: open vcodec_service (null) failed        ★
mpp[30550]: hal_h264d_api: mpp_dev_init failed ret: -1               ★
mpp[30550]: mpp_hal: mpp_hal_init hal h264d_rkdec init failed ret -1
mpp[30550]: mpp_hal: mpp_hal_init could not found coding type 7
mpp[30550]: mpp_dec: mpp_dec_init could not init hal
mpp[30550]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30550]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30550]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30550]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30550]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30550]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30550]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30550]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30550]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30550]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30550]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30550]: mpp: error found on mpp initialization
mpp[30550]: mpi_dec_test: 0xaaaac4ebe980 mpp_init failed
mpp[30550]: mpi_dec_test: test failed ret -1
mpi_dec_test[30550]: mpp_mem_pool: put_pool found 1 used buffer size 80
mpi_dec_test[30550]: mpp_mem_pool: put_pool found 4 used buffer size 192

HermanChen, 2021-08-10:
要使用硬件编解码器，首先需要内核驱动的支持，5.8 内核的补丁我们没有移植过
需要把 4.19 的 driver/video/rockchip/mpp 目录下的代码移植到 5.8 内核上，然后在对应 rk3328.dtsi 里加入编解码器配置
编译后如果能看到 /dev/mpp_service 设备，就说明成功了

zhangweiwahaha042, 2021-08-10:
非常感谢您的回复，给了我们很大的帮助。

》需要把 4.19 的 driver/video/rockchip/mpp 目录下的代码移植到 5.8 内核上，然后在对应 rk3328.dtsi 里加入编解码器配置
不好意思，我想再确认一下。
因为我没有找到OrangePi里版本是4.19的内核代码。在版本4.4的内核代码里，也没有找到driver/video/rockchip/mpp的目录。

你能够告诉我，在哪里可以下载到OrangePi里版本是4.19的内核代码吗？还是说，我将下面的驱动代码移植到5.8内核上就可以了呢？
https://github.com/orangepi-xunlong/linux-orangepi/tree/orange-pi-4.4-rockchip64/drivers/video/rockchip/vpu
<代码文件列表>
mpp_dev_common.c
mpp_dev_common.h
mpp_dev_h265e.c
mpp_dev_h265e.h
mpp_dev_h265e_define.h
mpp_dev_h265e_reg.h
...

HermanChen, 2021-08-10:
了解，在 4.19 上是 driver/video/rockchip/mpp，4.4 上应该是你说的对，是 driver/video/rockchip/vpu 目录

HermanChen, 2021-08-10:
需要注意下，3328 上的 H.265 编码器在最新的 mpp 里已经不支持了

zhangweiwahaha042, 2021-08-10:
非常感激您的回复。
我们这边参考您的意见做下尝试。

zhangweiwahaha042, 2021-08-25:
非常感谢您的帮助，我们使用下面的方法，现在可以使用mpp功能了，所以关闭本issue。
由于kernel 5.8上不支持ion的驱动，导致最终还是无法使用mpp功能，所以我们修改了方案，直接将kernel 4.4.xxx移植到了板子上。
※使用mpp功能时，当然也出现了新的问题，还在解决中

closed 2021-08-25 by zhangweiwahaha042

HermanChen, 2021-08-25:
好的，感谢反馈，5.8 可以使用 drm，mpp 也支持 drm 分配器，编译时加上 HAVE_DRM 就好