rockchip-linux/mpp #219: JPG decoding latency

bakwc, 2021-08-11:
Hi! I have a huge JPG decoding latency.
When decoding 150 KB, 2560x1440 jpeg file I get 63ms latency.
For 1MB jpeg file I get 150ms latency.
Is normal? I expect 1-10ms latency from hardware accelerated decoder, is it possible?

```
./mpi_dec_test -i img.jpg -t 8 -w 2560 -h 1440 -o result.bin
vpu_api: dlopen vpu lib failed
[info] jpg file loaded, size: 127406
mpi_dec_test start
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpi_dec_test decoder test start w 2560 h 1440 type 8
mpi: mpp version: 1ffa5af1 author: JMCC Fix CPU detection
jpg cpy: 1ms
jpg enqueue: 1ms
mpp_dec: mpp_dec_advanced_thread slot size 7372800 is not equal to buffer size 14745600
mpp_dec: Assertion slot_size == buffer_size failed at mpp_dec_advanced_thread:834
jpg poll: 63ms
jpg dequeue: 63ms
jpg task: 63ms
Decode time: 63ms
```

HermanChen, 2021-08-12:
The hardware has performance limit on huffman decoding. So when stream is larger then the decoding time is longer.
You can check the hardware timing:
echo 0x100 > /sys/modules/rk_vcodec/parameters/mpp_dev_debug

bakwc, 2021-08-12:
I run it: `echo 0x100 > /sys/module/mpp_dev_common/parameters/mpp_dev_debug`
Nothing in logs or dmesg. (I'm using old mpp version).

HermanChen, 2021-08-13:
the directory under /sys/modules may be vpu_service or rk_vcodec, please ls and try