rockchip-linux/mpp #221: 请问rk3568的板子可以解码 h264的 422格式的视频数据吗？播放解码后输出的文件 为什么是花屏？

wangjl-william, 2021-08-16:
写文件的命令：mpi_dec_test -i desktop-1920-1080-i422.h264 -t 7 -o 422.yuv，播放yuv数据花屏。
以下是mpi_dec_test 执行的详细信息：

linaro@linaro-alip:~/Hello_Triangle$ mpi_dec_test -d 1  -i desktop-1920-1080-i422.h264 -t 7  -o 422.yuv  -n 10
mpp[5376]: mpi_dec_test: cmd parse result:
mpp[5376]: mpi_dec_test: input  file name: desktop-1920-1080-i422.h264
mpp[5376]: mpi_dec_test: output file name: 422.yuv
mpp[5376]: mpi_dec_test: config file name: 
mpp[5376]: mpi_dec_test: width      :    0
mpp[5376]: mpi_dec_test: height     :    0
mpp[5376]: mpi_dec_test: type       : 7
mpp[5376]: mpi_dec_test: debug flag : 1
mpp[5376]: mpi_dec_test: max frames : 10
mpp[5376]: mpi_dec_test: mpi_dec_test start
mpp[5376]: mpi_dec_test: input file size 1362983
mpp[5376]: mpi: mpp_create enter ctx 0x7ff8985ad0 mpi 0x7ff8985ad8
mpp[5376]: mpp_info: mpp version: 41c9c809 author: sayon.chen    2020-12-24 [mpp2str]: Add rc_gop_mode 2 str
mpp[5376]: mpi: mpp_create leave ret 0 ctx 0x5597178740 mpi 0x7fa62c7c08
mpp[5376]: mpi_dec_test: 0x5597178740 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[5376]: mpi: mpi_control enter ctx 0x5597178740 cmd 310005 parm 0x7ff8985ac8
mpp[5376]: mpi: mpi_control leave ret 0
mpp[5376]: mpi: mpp_init enter ctx 0x5597178740 type 0 coding 7
mpp[5376]: mpp_rt: NOT found ion allocator
mpp[5376]: mpp_rt: found drm allocator
mpp[5376]: mpi: mpp_init leave ret 0
mpp[5376]: mpi: mpi_decode_put_packet enter ctx 0x5597178740 packet 0x55971786a0
mpp[5376]: mpi: mpi_decode_put_packet leave ret 0
mpp[5376]: mpi: mpi_decode_get_frame enter ctx 0x5597178740 frame 0x7ff8985b08
mpp[5376]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[5376]: mpi: mpi_decode_get_frame leave ret 0
mpp[5376]: mpi: mpi_decode_put_packet enter ctx 0x5597178740 packet 0x55971786a0
mpp[5376]: mpi: mpi_decode_put_packet leave ret 0
mpp[5376]: mpi: mpi_decode_get_frame enter ctx 0x5597178740 frame 0x7ff8985b08
mpp[5376]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[5376]: mpi: mpi_decode_get_frame leave ret 0
mpp[5376]: mpi: mpi_decode_put_packet enter ctx 0x5597178740 packet 0x55971786a0
mpp[5376]: mpi: mpi_decode_put_packet leave ret 0
mpp[5376]: mpi: mpi_decode_get_frame enter ctx 0x5597178740 frame 0x7ff8985b08
mpp[5376]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[5376]: mpi: mpi_decode_get_frame leave ret 0
mpp[5376]: mpi: mpi_decode_put_packet enter ctx 0x5597178740 packet 0x55971786a0
mpp[5376]: mpi: mpi_decode_put_packet leave ret 0
mpp[5376]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[5376]: mpi: mpi_decode_get_frame enter ctx 0x5597178740 frame 0x7ff8985b08
mpp[5376]: mpi: mpi_decode_get_frame leave ret 0
mpp[5376]: mpi: mpi_decode_put_packet enter ctx 0x5597178740 packet 0x55971786a0
mpp[5376]: mpi: mpi_decode_put_packet leave ret 0
mpp[5376]: mpi: mpi_decode_get_frame enter ctx 0x5597178740 frame 0x7ff8985b08
mpp[5376]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[5376]: mpi: mpi_decode_get_frame leave ret 0
mpp[5376]: mpi: mpi_decode_put_packet enter ctx 0x5597178740 packet 0x55971786a0
mpp[5376]: mpi: mpi_decode_put_packet leave ret 0
mpp[5376]: mpi: mpi_decode_get_frame enter ctx 0x5597178740 frame 0x7ff8985b08
mpp[5376]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[5376]: mpi: mpi_decode_get_frame leave ret 0
mpp[5376]: mpi: mpi_decode_put_packet enter ctx 0x5597178740 packet 0x55971786a0
mpp[5376]: mpi: mpi_decode_put_packet leave ret 0
mpp[5376]: mpi: mpi_decode_get_frame enter ctx 0x5597178740 frame 0x7ff8985b08
mpp[5376]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[5376]: mpi: mpi_decode_get_frame leave ret 0
mpp[5376]: mpi: mpi_decode_put_packet enter ctx 0x5597178740 packet 0x55971786a0
mpp[5376]: mpi: mpi_decode_put_packet leave ret 0
mpp[5376]: mpi: mpi_decode_get_frame enter ctx 0x5597178740 frame 0x7ff8985b08
mpp[5376]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[5376]: mpi: mpi_decode_get_frame leave ret 0
mpp[5376]: mpi: mpi_decode_put_packet enter ctx 0x5597178740 packet 0x55971786a0
mpp[5376]: mpi: mpi_decode_put_packet leave ret 0
mpp[5376]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=0
mpp[5376]: mpi: mpi_decode_get_frame enter ctx 0x5597178740 frame 0x7ff8985b08
mpp[5376]: mpi: mpi_decode_get_frame leave ret 0
[... 522 lines cut ...]
mpp[5376]: h264d_init: [DPB_REF_ERR] slot_idx=0, dpb_err[0]=0
mpp[5376]: h264d_init: list0 dpb: cur_err_flag=0, pps_refs=1, over_flag=0, num_ref_l0=1
mpp[5376]: h264d_parse: SliceSTATE_InitPicture
mpp[5376]: h264d_parse: SliceSTATE_GetSliceData
mpp[5376]: h264d_parse: SliceSTATE_RegisterOneFrame
mpp[5376]: h264d_dpb: [DPB_size] p_Dpb->used_size=1
mpp[5376]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=1098, pkt_no=10
mpp[5376]: h264d_api: [pkt_in_timeUs] is_avcC=0, in_pts=0, pkt_eos=0, len=4096, pkt_no=10
mpp[5376]: h264d_api: [CALLBACK] g_no=10, out_idx=1, dpberr=0, harderr=0, ref_flag=1, errinfo=0, discard=0
mpp[5376]: mpi: mpi_decode_get_frame enter ctx 0x5597178740 frame 0x7ff8985b08
mpp[5376]: mpi: mpi_decode_get_frame leave ret 0
mpp[5376]: mpi_dec_test: 0x5597178740 decode get frame 8
mpp[5376]: mpi: mpi_decode_get_frame enter ctx 0x5597178740 frame 0x7ff8985b08
mpp[5376]: mpi: mpi_decode_get_frame leave ret 0
mpp[5376]: mpi_dec_test: 0x5597178740 decode get frame 9
mpp[5376]: mpi_dec_test: 0x5597178740 reach max frame number 10
mpp[5376]: mpi: mpi_control enter ctx 0x5597178740 cmd 310101 parm 0x7ff8985b08
mpp[5376]: mpi: mpi_control leave ret 0
mpp[5376]: mpi_dec_test: 0x5597178740 input 68 pkt output 11 frm decode 10 frames
mpp[5376]: mpi: mpi_reset enter ctx 0x5597178740
mpp[5376]: h264d_dpb: warning: max_dec_frame_buffering(1) is less than dpb_size(4) calculated from Profile/Level.
mpp[5376]: mpi: mpi_reset leave ret 0
mpp[5376]: mpi: mpp_destroy enter ctx 0x5597178740
mpp[5376]: mpi: mpp_destroy leave ret 0
mpp[5376]: mpi_dec_test: test success max memory 24.90 MB

HermanChen, 2021-08-16:
麻烦上传下码流文件，可能 422 格式写文件有问题，也有可能是 YUV 工具有问题

wangjl-william, 2021-08-17:
> 麻烦上传下码流文件，可能 422 格式写文件有问题，也有可能是 YUV 工具有问题

[422.zip](https://github.com/rockchip-linux/mpp/files/6996599/422.zip)
yuv播放工具 以及 使用系统中的工具  mpi_dec_test 解码后写的文件。

wangjl-william, 2021-08-18:
@HermanChen 帮忙看了吗，谢谢。

HermanChen, 2021-08-30:
麻烦测试下新的版本是否已经解决了这个问题~多谢先