rockchip-linux/mpp #222: decode_get_frame returns null for the last frame

bakwc, 2021-08-16:
When decoding a single h264 frame file - no frames decoded.
When decoding a file with 2 h264 frames - only first frame is decoded.
The last frame is always missing, `decode_get_frame` returns zero for last frame.

bakwc, 2021-08-16:
Here is a frame that I try to decode:
https://drive.google.com/file/d/1Q3VCnBPoHp1Z5x1kplvF5lyllEtfiV14/view?usp=sharing
Is it possible to decode a single frame? With ffmpeg it decodes without problem, but I get empty file when decode with `mpi_dec_test`

FumasterLin, 2021-08-17:
My local test is normal . 
Please tell us the version of your mpp repo？

bakwc, 2021-08-17:
I use `release_20171218` (`NanoPI M4`, with `rk3399`), repo https://github.com/teacupx/mpp, branch `20171218-armbian`.

FumasterLin, 2021-08-17:
The version is too old, you can update newest develop branch from repo: https://github.com/rockchip-linux/mpp, and test again.

bakwc, 2021-08-17:
The fresh version doesn't work with armbian kernel, `mpp_platform_test` shows null:

```
sudo ./mybuild/osal/test/mpp_platform_test
mpp_plat_test: chip name: (null)
mpp_plat_test:
mpp_plat_test: chip vcodec type 00000000
mpp_plat_test:
mpp_plat_test: start probing decoder device name:
mpp_plat_test: H.264 decoder: (null)
mpp_plat_test: H.265 decoder: (null)
mpp_plat_test: MJPEG decoder: (null)
mpp_plat_test: VP9   decoder: (null)
mpp_plat_test: avs   decoder: (null)
mpp_plat_test:
mpp_plat_test: start probing encoder device name:
mpp_plat_test: H.264 encoder: (null)
mpp_plat_test: H.265 encoder: (null)
mpp_plat_test: MJPEG encoder: (null)
mpp_plat_test: mpp platform test done
```
Should I build fresh kernel too? Should I use branch `develop-4.4` from here: https://github.com/rockchip-linux/kernel ?
Does it matches https://github.com/rockchip-linux/mpp `develop` branch?

FumasterLin, 2021-08-17:
The log you posted doesn't seem to be the latest mpp result.
You can double check again.

bakwc, 2021-08-17:
Checked on fresh version. `mpp_platform_test` looks good:

```
fippo@nanopim4:~/koding/mpp_base$ ./mybuild/osal/test/mpp_platform_test
mpp[6609]: mpp_plat_test: kernel version: 4.4
mpp[6609]: mpp_plat_test: ioctl  version: vcodec_service
mpp[6609]: mpp_plat_test:
mpp[6609]: mpp_plat_test: chip name: friendlyelec,nanopi-m4 rockchip,rk3399
mpp[6609]: mpp_plat_test:
mpp[6609]: mpp_plat_test: chip vcodec type 0004020a
mpp[6609]: mpp_plat_test: found vpu2 codec
mpp[6609]: mpp_plat_test: found rkvdec decoder
mpp[6609]: mpp_plat_test: found vpu2 encoder
mpp[6609]: mpp_plat_test:
mpp[6609]: mpp_plat_test: start probing decoder device name:
mpp[6609]: mpp_plat_test: H.264 decoder: /dev/rkvdec
mpp[6609]: mpp_plat_test: H.265 decoder: /dev/rkvdec
mpp[6609]: mpp_plat_test: MJPEG decoder: /dev/vpu_service
mpp[6609]: mpp_plat_test: VP9   decoder: /dev/rkvdec
mpp[6609]: mpp_plat_test: avs   decoder: /dev/vpu_service
mpp[6609]: mpp_plat_test:
mpp[6609]: mpp_plat_test: start probing encoder device name:
mpp[6609]: mpp_plat_test: H.264 encoder: /dev/vpu_service
mpp[6609]: mpp_plat_test: H.265 encoder: /dev/vpu_service
mpp[6609]: mpp_plat_test: MJPEG encoder: /dev/vpu_service
mpp[6609]: mpp_plat_test: mpp platform test done
```

Decoding not working:

```
fippo@nanopim4:~/koding/mpp_base$ ./mybuild/test/mpi_dec_test  -i frame_0.264 -w 1920 -h 1080 -t 7 -o dum.yuv
mpp[6639]: mpi_dec_utils: cmd parse result:
mpp[6639]: mpi_dec_utils: input  file name: frame_0.264
mpp[6639]: mpi_dec_utils: output file name: dum.yuv
mpp[6639]: mpi_dec_utils: config file name:
mpp[6639]: mpi_dec_utils: width      : 1920
mpp[6639]: mpi_dec_utils: height     : 1080
mpp[6639]: mpi_dec_utils: type       : 7
mpp[6639]: mpi_dec_utils: debug flag : 0
mpp[6639]: mpi_dec_utils: max frames : 0
mpp[6639]: mpi_dec_test: mpi_dec_test start
mpp[6639]: mpi_dec_test: input file size 64066
mpp[6639]: mpp_info: mpp version: ac4c6718 author: Johnson Ding  2021-08-12 [vepu_common]: Support RGBA little endian format
mpp[6639]: mpi_dec_test: 0x5582a3cbc0 mpi_dec_test decoder test start w 1920 h 1080 type 7
mpp[6639]: mpp_rt: NOT found ion allocator
mpp[6639]: mpp_rt: found drm allocator
mpp[6639]: mpi_dec_test: 0x5582a3cbc0 found last packet
mpp[6639]: mpi_dec_test: 0x5582a3cbc0 decode_get_frame get info changed found
mpp[6639]: mpi_dec_test: 0x5582a3cbc0 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 4177920
mpp[6639]: vcodec_service: vcodec_service_cmd_send ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
mpp[6639]: hal_h264d_rkv_reg: rkv_h264d_start send cmd failed 14
mpp[6639]: vcodec_service: vcodec_service_cmd_poll ioctl VPU_IOC_GET_REG failed ret -1 errno 110 Connection timed out
mpp[6639]: hal_h264d_rkv_reg: rkv_h264d_wait poll cmd failed 110
mpp[6639]: mpi_dec_test: 0x5582a3cbc0 decode get frame 0 err 1 discard 0
mpp[6639]: mpi_dec_test: 0x5582a3cbc0 found last packet
mpp[6639]: mpi_dec_test: 0x5582a3cbc0 input 16 pkt output 2 frm decode 1 frames
mpp[6639]: mpi_dec_test: test success max memory 3.98 MB
```
`dum.yuv` is empty.

FumasterLin, 2021-08-17:
Please apply the diff below

```
diff --git a/build/linux/arm/make-Makefiles.bash b/build/linux/arm/make-Makefiles.bash
index 9dcbc7e8..1e891339 100755
--- a/build/linux/arm/make-Makefiles.bash
+++ b/build/linux/arm/make-Makefiles.bash
@@ -24,5 +24,6 @@ fi
 cmake -DCMAKE_BUILD_TYPE=Release \
       -DCMAKE_TOOLCHAIN_FILE=./arm.linux.cross.cmake \
       -DRKPLATFORM=ON \
+      -DHAVE_DRM=ON \
       -G "Unix Makefiles" \
       ${MPP_TOP}
```

bakwc, 2021-08-17:
Thank you! Fresh version decodes good.

closed 2021-08-17 by bakwc